espressif/esp-idf #6239: esp_http_client disable_auto_redirect working as intended?

jgcjoec, 2020-12-10:
## Environment

- Development Kit: Hiletgo ESP32 dev board
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.1
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

I have a program that uses esp_http_client to make an HTTP POST request to a specific URL. One of the possible responses to that HTTP POST request is a HTTP 302 (redirect) with a "Location" HTTP header. Interestingly, ESP-IDF appears to respect redirects, which is pretty neat, but undesirable in this particular use-case where I wanted to just obtain the new URL supplied by the server, via the "Location" HTTP header.

I attempted to disable auto redirects by setting disable_auto_redirect to true, but when I did, I started to see odd behavior - HTTP requests whose response code was 200 started failing.

I looked at the esp_http_client code, in function esp_http_check_response() and it appears that any time disable_auto_redirect is true, the function will set an error.  Note that the value in disable_auto_redirect is checked in the if() block below, before the status code is checked.

`
static esp_err_t esp_http_check_response(esp_http_client_handle_t client)
{
    if (client->redirect_counter >= client->max_redirection_count || client->disable_auto_redirect) {
        ESP_LOGE(TAG, "Error, reach max_redirection_count count=%d", client->redirect_counter);
        return ESP_ERR_HTTP_MAX_REDIRECT;
    }
    switch (client->response->status_code) {
        case HttpStatus_MovedPermanently:
        case HttpStatus_Found:
        case HttpStatus_TemporaryRedirect:
            esp_http_client_set_redirection(client);
            client->redirect_counter ++;
            client->process_again = 1;
            break;
        case HttpStatus_Unauthorized:
            esp_http_client_add_auth(client);
    }
    return ESP_OK;
}
`

It is entirely possible that I am missing something here, but that seems undesirable. The documentation on this option is fairly vague, stating simply "Disable HTTP automatic redirects" with no further information. I did not see any examples that use this option that may shed some additional light on its intended use.

If I've missed something, please set me straight :)

### Expected Behavior
If disable_auto_redirect is set to true, the HTTP request should only fail if the HTTP response is, in fact, a redirect.

### Actual Behavior
It seems that HTTP 200 responses also fail

Alvin1Zhang, 2020-12-11:
Thanks for reporting and letting us know, we will look into.

shubhamkulkarni97, 2020-12-11:
Hi @jgcjoec, we have already fixed this issue on master branch. Fix will be updated on GitHub once all internal checks are passed.

Please find attached patch to fix issue on release/v4.1 branch and let me know if it works for you.
I'll make sure fix is backported to release branches.

[0001-esp_http_client-Skip-check-for-redirection-counter-i.patch.zip](https://github.com/espressif/esp-idf/files/5677146/0001-esp_http_client-Skip-check-for-redirection-counter-i.patch.zip)

Thanks,
Shubham

jgcjoec, 2020-12-11:
Thanks, @shubhamkulkarni97!

Alvin1Zhang, 2020-12-14:
@jgcjoec Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/aceb38c326a5450b1aa7e1b4202bc4648e1829fd, feel free to reopen if it happens. Thanks.

closed 2020-12-14 by Alvin1Zhang