espressif/esp-idf #6242: WiFi Station Disconnects from SoftAP

gecko242, 2020-12-10:
## Environment

- WROOM32U on custom PCB.
- IDF version v4.3-dev-1720-g494a124d9
- Build System:        idf.py
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: VSCode, not using esp extension

## Problem Description
ESP32 set up as a soft AP + Client. When a station connects, it often disconnects shortly ~(3-5s) after. Once connected, it remains connected OK, and does not disconnect. This appears to be an issue for the first station to connect to the network, any subsequent devices connect fine first time.

This seems to happen regardless of OS, tested on both IOS and Android. 

I suspect I may have a configuration setting wrong, but as the wifi blob is closed, I cant easily debug. Any advice on the below log would be great :) 

## Debug Logs

```
D (43600) wifi:ap recv assoc/reassoc request
I (43610) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<11,0>, prof:1
D (43610) wifi:bss=0x3ffeb9b8, resp=16 join
I (43610) wifi:station: fe:00:4c:cc:23:3d join, AID=1, bgn, 20
D (43610) wifi:join success, ap send assoc response
D (08:04:13.937) event: running post WIFI_EVENT:14 with handler 0x400e9acc and context 0x3ffbfcd8 on loop 0x3ffbf794
0x400e9acc: network_event_handler at c:\esp32\projects\safeguard\build/../main/network.c:175

I (08:04:13.947) SafeGuard.network: station fe:00:4c:cc:23:3d join, AID=1
W (43710) wifi:<ba-add>idx:4 (ifx:1, fe:00:4c:cc:23:3d), tid:0, ssn:0, winSize:64
D (08:04:14.072) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbfd58
D (08:04:14.127) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbfd58
I (08:04:14.127) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
D (08:04:14.132) event: running post IP_EVENT:2 with handler 0x400e9acc and context 0x3ffbfd28 on loop 0x3ffbf794
0x400e9acc: network_event_handler at c:\esp32\projects\safeguard\build/../main/network.c:175

I (08:04:14.408) SafeGuard.network: Replying to DNS request for www.google.com from 192.168.4.2
I (08:04:14.412) SafeGuard.network: Replying to DNS request for connectivitycheck.gstatic.com from 192.168.4.2

// I have removed some unrelated HTTP logs here, serving a webpage through a captive portal. 
// There are no other clues as to why the device may deauth. 

D (49560) wifi:ap recv deauth/disassoc from sta fe:00:4c:cc:23:3d, reason:4
I (49560) wifi:station: fe:00:4c:cc:23:3d leave, AID = 1, bss_flags is 134243, bss:0x3ffeb9b8
I (49560) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<11,0>, prof:1
W (49570) wifi:<ba-del>idx
D (08:04:19.892) event: running post WIFI_EVENT:15 with handler 0x400e9acc and context 0x3ffbfcd8 on loop 0x3ffbf794
0x400e9acc: network_event_handler at c:\esp32\projects\safeguard\build/../main/network.c:175

D (49580) wifi:ap recv deauth/disassoc from sta ff:ff:ff:ff:ff:ff, reason:7
I (08:04:19.903) SafeGuard.network: station fe:00:4c:cc:23:3d leave, AID=1
```

YouDONG-ESP, 2020-12-11:
@gecko242 Thanks for reporting, could you show me a code to reproduce this issue?

gecko242, 2020-12-11:
Okay, a little more info:

This deauth seems to only happen whilst I am serving pages from flash using the HTTPD component. Occasionally, I get the following error:

```
D (382588) wifi:ap recv deauth/disassoc from sta fe:00:4c:cc:23:3d, reason:4
I (382588) wifi:station: fe:00:4c:cc:23:3d leave, AID = 1, bss_flags is 134243, bss:0x3ffec588
I (382588) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<11,0>, prof:1
W (382598) wifi:<ba-del>idx
D (382598) wifi:no bcast bss, send deauth: reason not authed(6)
D (382608) wifi:no bcast bss, send deauth: reason not authed(6)
D (382608) wifi:no bcast bss, send deauth: reason not authed(6)
D (382618) wifi:no bcast bss, send deauth: reason not authed(6)
D (382618) wifi:no bcast bss, send deauth: reason not authed(6)
D (382628) wifi:no bcast bss, send deauth: reason not authed(6)
D (382638) wifi:no bcast bss, send deauth: reason not authed(6)
D (382638) wifi:no bcast bss, send deauth: reason not authed(6)
D (382648) wifi:no bcast bss, send deauth: reason not authed(6)
D (382648) wifi:no bcast bss, send deauth: reason not authed(6)
D (382658) wifi:no bcast bss, send deauth: reason not authed(6)
D (382658) wifi:no bcast bss, send deauth: reason not authed(6)
D (382668) wifi:no bcast bss, send deauth: reason not authed(6)
D (382678) wifi:no bcast bss, send deauth: reason not authed(6)
D (382688) wifi:no bcast bss, send deauth: reason not authed(6)
D (382698) wifi:no bcast bss, send deauth: reason not authed(6)
D (382708) wifi:no bcast bss, send deauth: reason not authed(6)
D (07:20:39.070) event: running post WIFI_EVENT:15 with handler 0x400e9ac4 and context 0x3ffbfcc8 on loop 0x3ffbf784
0x400e9ac4: network_event_handler at c:\esp32\projects\safeguard\build/../main/network.c:175

```

I wonder what the "reason not authed" is all about? Possibly some sort of memory corruption? I have the heap corruption set to comprehensive, and that is not throwing any errors.

YouDONG-ESP, 2020-12-22:
@gecko242 Hi, `D (382658) wifi:no bcast bss, send deauth: reason not authed(6)` It shows that the sta send a data frame during authentication step. Our softap hasn't got any connection. Could you provide a packet capture for debugging? 
You could follow this guide:
The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

gecko242, 2020-12-23:
Hi @YouDONG-ESP, 

The only issue is that I cannot recreate this with 2 devices on the network, and Windows does not exhibit the same disconnection problem. 

I.E, if I connect my windows laptop to the soft-AP, I then cannot recreate the disconnection problem on my Android phone. 

Do you know of a way to obtain a packet capture on Android?

Many Thanks, and have a great Christmas!
Sam

YouDONG-ESP, 2020-12-24:
Merry Christmas Sam @gecko242 ! 

I've never capture packet with phone, we always use wireshark/omnipeak with a network adapter . I took a look at google play, there are quiet a lot andoid sniffer apps, some of them requires root access. Pick one can capture wifi packet should do. (Maybe this one  _Sniffer Wicap_ ?)

gecko242, 2021-01-04:
@YouDONG-ESP Thankyou, and Happy New Year :) 

I will take a look and report back as to how I get on.

gecko242, 2021-01-05:
Hi @YouDONG-ESP 
Packet capture attached. Let me know if you can make anything of it! 

[capture.zip](https://github.com/espressif/esp-idf/files/5769487/capture.zip)

Cheers,
Sam

YouDONG-ESP, 2021-01-05:
Hi @gecko242 the packets you sent is actually wired packet transmit by ethernet. What i need is wireless 80211 packet, Could you try again to get it? Or a easier way, you could give me a code to reproduce this issue and I can do it in my side.

gecko242, 2021-01-05:
That's strange, that must just be down to the packet capture app I was using. Let me see if I can find something that will grab wireless packets. 
The code is quite large, so stripping it down such that it will run without my hardware may prove problematic.

gecko242, 2021-01-09:
Hi @YouDONG-ESP 

I have finally got to the bottom of this, and I am just updating and closing the issue. It turns out this was down to an issue with an uninitialised array index, leading to some memory corruption, clearly in an area that broke the wifi! 

Thanks for you help on this :)

closed 2021-01-09 by gecko242