espressif/esp-idf #6243: Time-of-Flight ESP32S2 initiate without connecting first

seijikun, 2020-12-10:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER
- IDF version: feature/ftm_support
- Build System: CMake | idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
The current ftm-shell example project requires the second ESP to connect to the first one's access-point before being able to act as ftm initiator. Is this also possible without having to connect first?
We also tried this with a Google Pixel2 XL and the official `Wifi RTT Scan` app. The access point created by the first ESP32S2 is shown as FTM-capable, though when trying to scan, it does not answer.
As soon as we choose the created access-point, the ESP's console shows the following error:
```
E (53165) wifi:Unknown Element Id - 0xdd!
```
This is logged everytime we try to scan, using the `Wifi RTT Scan` app.

### Expected Behavior
The expected behavior is for FTM to also work without connecting to the access-point first. 

### Steps to reproduce

1. Build official FTM example from esp-idf
2. Flash onto a ESP32S2
3. Start access point with `ap` command
4. Use `Wifi RTT Scan` app, and choose the new access point

### Code to reproduce this issue
The official ftm example in esp-idf.

georgcampana, 2020-12-10:
I agree. Common scenario is too use multiple peers in order to decimate the measure error.
Ideally the cfg structure should take a target bssid (not ssdid)  that is used as FTM responder.

Alvin1Zhang, 2020-12-11:
Thanks for reporting and letting us know, we will look into.

sagb2015, 2020-12-11:
@seijikun @georgcampana - We understand the requirement and it is already work under progress. Our first target is to get the feedback over measurement. We will update this thread soon.

FrankDeinzer, 2020-12-11:
Just to let you know: I'm also eagerly awaiting the completion of this very feature.

aleixtejadagimenez, 2020-12-15:
Following up on this issue, the same 'Unknown Element Id - 0xdd!' happens when using the ESP32  in station mode against a [compulab wild ](https://fit-iot.com/web/products/wild/) acting as FTM responder.

The requests issued by the ESP appear to be valid (as reported by the built-in tool on the wild), here's the output of the tool:

FTM responder stats:
	Success num 131
	Partial success num 7
	Failed num 3
	ASAP success num 3
	Non ASAP num 138
	Total duration 9172372
	Unknown triggers num 0

dtony, 2021-01-06:
I faced more or less the same problem using a Xiaomi Mi9T Pro as initiator with "WifiRttScan App" and SoftAP example with ftm enabled. Here are the logs from my esp :
```E (741326) wifi:Unknown Element Id - 0x26!
E (741326) wifi:Unknown Element Id - 0x26!
E (741326) wifi:Unknown Element Id - 0xdd!
I (741336) wifi:Responding to FTM initiator b2:33:e6:d2:34:14 with 8 FTM frames every 4000 uSec```