espressif/esp-idf #6244: ESP32 WIFI LORA

JenniferGladysPereiraCavalcante, 2020-12-10:
Hi, I have this message showing up.
I use an ESP32 WIFI LORA with a GY-GPS6MV2 module
I use the IDE and when compiling it asks to press the RST
when I open the serial monitor, this message appears and stabilizes, it does nothing else.

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

YouDONG-ESP, 2020-12-11:
Hi @JenniferGladysPereiraCavalcante, ESP32 WIFI LORA can only use with another ESP32, It won't work with others type of chips

Alvin1Zhang, 2020-12-16:
Thanks for reporting.

closed 2020-12-16 by Alvin1Zhang