espressif/esp-idf #6245: ipc1 task crash

sebirdman, 2020-12-10:
## Environment

- IDF version: 4.0.2
- Build System:  Cmake
- Compiler version: esp-2020r3

## Problem Description

I've received a coredump from one of my devices in the field that is attached. kinda strange as it doesn't seem to point to anything particularly wrong as far as i can tell

### Expected Behavior

not crashing

### Actual Behavior

crashing

### Steps to reproduce

Unknown, just seeing this on some devices in the field 

## Debug Logs

```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffb9a3c, name: 'ipc1', GDB name: 'process 1073453628'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x4008515d
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x40081710       0x40081710 <esp_crosscore_int_send_yield+12>
lbeg           0x0      0
lend           0x0      0
lcount         0x0      0
sar            0x19     25
ps             0x60920  395552
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8009725d       -2146864547
a1             0x3ffb7ef0       1073446640
[... 361 lines cut ...]
.coredump.tasks.data 0x3ffd1840 0x24c RW 
.coredump.tasks.data 0x3ffd7068 0x170 RW 
.coredump.tasks.data 0x3ffd65e0 0xa80 RW 
.coredump.tasks.data 0x3ffb74cc 0x170 RW 
.coredump.tasks.data 0x3ffb72e0 0x1e4 RW 
.coredump.tasks.data 0x3ffd919c 0x170 RW 
.coredump.tasks.data 0x3ffd8f20 0x274 RW 
.coredump.rom.text 0x40000000 0x170 R X
.coredump.rom.text 0x40000180 0x8 R X
.coredump.rom.text 0x400001c0 0x8 R X
.coredump.rom.text 0x40000200 0x8 R X
.coredump.rom.text 0x40000240 0x8 R X
.coredump.rom.text 0x40000280 0xc R X
.coredump.rom.text 0x400002c0 0x4 R X
.coredump.rom.text 0x40000300 0x8 R X
.coredump.rom.text 0x40000340 0x18 R X
.coredump.rom.text 0x400003c0 0x8 R X
.coredump.rom.text 0x40000400 0x160 R X
.coredump.rom.text 0x40000560 0xd10c R X
.coredump.rom.text 0x40010000 0x54ef4 R X

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

Alvin1Zhang, 2020-12-11:
Thanks for reporting and letting us know, and sorry for the inconvenience, we will look into.

xiewenxiang, 2021-06-09:
@sebirdman 

Could you please provide the example code and reproduce steps? And the commit of ESP-IDF.

closed 2021-06-24 by sebirdman

Coopdis, 2023-09-03:
I've just run into this same issue.  This also happens with devices in the field, and it happens intermittently.  I've got a few of devices in the field.  Every time a dumps core, it sends back the core dump on reboot.  It seems that *every time* my devices dump core it is because of an issue that fits this description, which is to say...

`===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffafcd0, name: 'ipc1', GDB name: 'process 1073413328'

================== CURRENT THREAD REGISTERS ===================`

`==================== CURRENT THREAD STACK =====================
#0  0x40008150 in ?? ()
#1  0x40081bc2 in print_str (str=0x3f405568 \"\\r\
\\r\
Backtrace:\", panic=false) at /home/bill/esp/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:61
#2  0x40081c38 in esp_backtrace_print_from_frame (depth=100, frame=0x3ffb15d0, panic=false) at /home/bill/esp/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:76
#3  0x40081d88 in esp_backtrace_print (depth=100) at /home/bill/esp/esp-idf/components/esp_system/port/arch/xtensa/debug_helpers.c:110
#4  0x40084174 in esp_crosscore_isr (arg=0x3ffb94e4 <reason+4>) at /home/bill/esp/esp-idf/components/esp_system/crosscore_int.c:92
#5  0x40082ffc in _xt_lowint1 () at /home/bill/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111
#6  0x40084873 in spi_flash_op_block_func (arg=0x0) at /home/bill/esp/esp-idf/components/spi_flash/cache_utils.c:126
#7  0x40085b50 in ipc_task (arg=<optimized out>) at /home/bill/esp/esp-idf/components/esp_ipc/src/esp_ipc.c:77

`

Since it is an intermittent issue detected across a fleet of devices, I'm not able to give a clear reproduction at this time.  But I can say that this is likely to happen 2-10 time per day on any given device.

Interestingly, there does not appear to be a single other thread, or single piece of code on a thread that is causing this issue. After analyzing 10 core dumps of this general description, there is no other thread that is the same spot twice, unless it is paused waiting for it's turn, or waiting for an event.

But the ipc1 crash is always the same.

I am running on origin/release/v4.4 @ commit 755ce1077df771ee872871ea03f5ce747ff99bfa

MedAouadhi, 2024-11-27:
@Coopdis I am also facing a similar issue, can I ask what is the status of that bug and did you manage to find the root cause ?

KonstantinKondrashov, 2024-11-28:
Hi @MedAouadhi!
It seems the original issue is related to an assert in the Bluetooth API (r_assert_param). Please open a new issue with more details. Note that ESP-IDF v4.4 is no longer supported; upgrading to a newer version is recommended as the issue might already be resolved.

If you can not update it and think that it is related to IPC you might try to disable the 
CONFIG_ESP_IPC_USES_CALLERS_PRIORITY option or try to apply the fix related to IPC recently https://github.com/espressif/esp-idf/commit/3df89ebe. It was backported till v5.0.