espressif/esp-idf #6246: ESP32-S2 rejecting GPIO light sleep.

tannewt, 2020-12-11:
## Environment

- Development Kit:     Adafruit MagTag
- Module or chip used: ESP32-S2-WROVER
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-1198-gd06744f5e
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I'm setting up GPIO interrupts to wake from light sleep in CircuitPython. However, esp_light_sleep_start returns immediately with `1`. From my code digging, this implies the RTC is rejecting the sleep request. I've been careful to enable the interrupt and wake after pulling the pin but it makes no difference.

### Expected Behavior
esp_light_sleep_start should block until I press the button attached to pin 11. The button connects to ground so I'm trying to pull up as well.

### Actual Behavior

esp_light_sleep_start return immediately with 1 and the wake reason isn't gpio. (I think the reject reason is though.)

### Steps to reproduce

Code that sets the gpio wake settings: https://github.com/tannewt/circuitpython/blob/pin_alarm/ports/esp32s2/common-hal/alarm/pin/PinAlarm.c#L253-L308

Code that enters the light sleep: https://github.com/tannewt/circuitpython/blob/pin_alarm/ports/esp32s2/common-hal/alarm/__init__.c#L143-L156

### Code to reproduce this issue
```python
import alarm
import alarm.pin
import alarm.time
import time
import neopixel
import board
import digitalio

if alarm.wake_alarm:
    print("awake", alarm.wake_alarm, alarm.wake_alarm.pin)
else:
    print("no wake up alarm")

enable = digitalio.DigitalInOut(board.NEOPIXEL_POWER)
enable.switch_to_output(False)

# Sleep for 5 so USB can start.
time.sleep(5)

pixels = neopixel.NeoPixel(board.NEOPIXEL, 4)
pixels.fill(0x0f0000)

time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic()+10)
print("light time sleep")
a = alarm.light_sleep_until_alarms(time_alarm)
print(a)

pixels.fill(0x000f00)
pin_alarm = alarm.pin.PinAlarm(pin=board.D11, value=False, pull=True)

print("light pin sleep")
a = alarm.light_sleep_until_alarms(pin_alarm)
print(a, a.pin)

time.sleep(1)

print("sleeping")

# We don't turn off the neopixels explicitly.
# We expect code after the VM to do it.

alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
```

## Debug Logs

```
I (1686) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (1686) pin alarm: reset
I (1766) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1766) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (1766) pin alarm: reset
I (2226) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (7226) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (7226) alarm: real light sleep
I (7226) alarm: reject config 0
W (7246) alarm: light sleep result 0       <<<<<<<<----- this is the first light sleep that waits on time
I (7246) alarm: low int 00000000
I (7246) alarm: light sleep done
I (7246) alarm: raw wake up cause 8
I (7246) alarm: timealarm wake up
I (7256) alarm: wake up cause 4
W (7256) pin alarm: reset
I (7266) pin alarm: light sleep gpio wakeup
I (7266) pin alarm: 11
I (7266) pin alarm: gpio13 interrupt on 11
I (7276) pin alarm: gpio interrupt on low level
I (7276) pin alarm: gpio interrupt with pull
I (7286) pin alarm: waiting for pull
I (7286) pin alarm: took 0 cycles
I (7296) alarm: real light sleep
I (7296) alarm: reject config 0
W (7316) alarm: light sleep result 1   <<<<<<<--- this is where it reject the sleep on gpio
I (7316) alarm: low int 00000000
E (7316) alarm: reject cause 4
I (7316) alarm: light sleep done
I (7316) alarm: raw wake up cause 8
I (7326) alarm: idf wake up
I (7326) alarm: wake up cause 4
W (7326) pin alarm: reset
I (7396) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
```

Alvin1Zhang, 2020-12-11:
Thanks for reporting and letting us know, we will look into.

igrr, 2020-12-11:
Confirmed with @tannewt, this seems to be fixed with https://github.com/espressif/esp-idf/commit/5ad2ec79fd2cdb1c923c0232802beb1291fe1df4.

closed 2020-12-11 by igrr