espressif/esp-idf #6249: Ethernet: RMII loses packets

lexxvir, 2020-12-11:
## Environment

- Custom Hardware
- Module or chip used:  ESP32-WROVER-B
- IDF version v3.3.4
- Build System:  Make
- Compiler version  1.22.0-80-g6c4433a
- Operating System:     Linux

## Problem Description

We use ESP32 in our custom hardware. We work using Ethernet chip [KSZ8863](https://www.microchip.com/wwwproducts/en/KSZ8863) connected to ESP32 by RMII interface.

Most of time all work fine. But sometimes, after device is powered up, ESP32 falls into the state when it loses packets.
In that state ESP32 loses TCP packets as well as ARP packets.

Device falls down into that state and stayes there until power down.
Sometimes reboot by the external pin helps but most of time it does not help.

I enable debug logs of LWIP and also read MIB counters/registers of KSZ8863.
According that info I figured out that loses are occurs in the RMII interface.

### Expected Behavior
That's normal behavior when packet is not lost:
```
tcpip_thread: PACKET 0x3ffd4054
ethernet_input: dest:ff:ff:ff:ff:ff:ff, src:c4:12:f5:d4:90:4f, type:806
etharp_update_arp_entry: 10.42.0.1 - c4:12:f5:d4:90:4f
etharp_find_entry: found empty entry 0
etharp_find_entry: selecting empty entry 0
etharp_update_arp_entry: updating stable entry 0
etharp_input: incoming ARP request
etharp_raw: sending raw ARP packet.
ethernet_output: sending packet 0x3ffdca64
etharp_timer
etharp_timer
I (27494) eth: Port1 RxLoPriorityByte: 64
I (27509) eth: Port1 RxBroadcast: 1
I (27509) eth: Port1 Rx64Octets: 1
I (27519) eth: Port1 TxLoPriorityByte: 64
I (27529) eth: Port1 TxUnicastPkts: 1

I (27539) eth: Port3 RxLoPriorityByte: 64
I (27554) eth: Port3 RxUnicast: 1
I (27554) eth: Port3 Rx64Octets: 1
I (27564) eth: Port3 TxLoPriorityByte: 64
I (27569) eth: Port3 TxBroadcastPkts: 1
```

Port3 transmits packet (`TxBroadcastPkts`) to RMII and receives reply (`RxUnicast`) from RMII.

### Actual Behavior
That's behavior when packet is lost in RMII:

```
I (148705) lwip: arp stats_proto { xmit: 5, recv: 3, fw: 0, drop: 0, chkerr: 0, lenerr: 0, memerr: 0, rterr: 0, proterr: 0, opterr: 0, err: 0, cachehit: 18 }
etharp_timer
tcpip_thread: CALLBACK 0x3ffd4054
etharp_timer

>>>>>>>>
Here I send one ARP request by `arping -c 1`
As you can see by 'eth Port*' messages KSZ8863 successfully got packet and forward it from
Port1 (external Ethernet) to Port 3 (RMII interface to ESP32).
But LWIP does not detects/logs any packet.
>>>>>>>>

etharp_timer
etharp_timer
etharp_timer
etharp_timer
I (153835) eth: Port1 RxLoPriorityByte: 64
I (153850) eth: Port1 RxBroadcast: 1
I (153855) eth: Port1 Rx64Octets: 1
I (153905) eth: Port3 TxLoPriorityByte: 64
I (153910) eth: Port3 TxBroadcastPkts: 1
I (153925) lwip: arp stats_proto { xmit: 5, recv: 3, fw: 0, drop: 0, chkerr: 0, lenerr: 0, memerr: 0, rterr: 0, proterr: 0, opterr: 0, err: 0, cachehit: 18 }
```

Port3 transmits packet (`TxBroadcastPkts`) to RMII but gets nothing.

### Steps to reproduce

Just power down and then power up device until it falls into that state.

- [sdkconfig.zip](https://github.com/espressif/esp-idf/files/5677429/sdkconfig.zip)

lexxvir, 2020-12-11:
Additional info:
- KSZ8863RLL generates a 50 MHz reference clock for the RMII interface
- We use 10Mbit half-duplex mode on all ports

suda-morris, 2020-12-14:
I remember someone also met packet loss with RMII due to signal integrity issue

lexxvir, 2020-12-15:
Reproduce the issue on IDF v4.2.

suda-morris, 2020-12-16:
Could you delay some time after you reset PHY chip by checking id some PHY register can return correct value? I guess PHY is going into a wrong state after power on.

lexxvir, 2020-12-16:
Hi @suda-morris,

I've added 50 ms delay after reset the PHY. After that I:
 - read Chip ID0 and Chip ID1/Start Switch registers (both looks OK)
 - configure switch to force 10Mbit and half-duplex mode (both actions done successfully)

> I guess PHY is going into a wrong state after power on.

Yes, it is possible. But, as I mentioned early, I read the MIBs counters in the loop and the switch reports the right numbers of the transmitted packet/bytes.

suda-morris, 2020-12-16:
OK, I wonder after you configured the switch PHY chip into 10M+HD, does [this line](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth.c#L136) got triggered? The logic in esp-idf ethernet driver is, we depend on auto-negotiation to decide which duplex and speed to be set on MAC side.

lexxvir, 2020-12-16:
I do no negotiation at all in my PHY component.
It is the code:

```c
static esp_err_t ksz8863_negotiate(esp_eth_phy_t *phy)
{
    phy_ksz8863_t *ksz8863 = __containerof(phy, phy_ksz8863_t, parent);
    esp_eth_mediator_t *eth = ksz8863->eth;

    eth_link_t link = ETH_LINK_UP;
    eth_speed_t speed = ETH_SPEED_10M;
    eth_duplex_t duplex = ETH_DUPLEX_HALF;
    PHY_CHECK(eth->on_state_changed(eth, ETH_STATE_SPEED, (void *)speed) == ESP_OK, "send speed event failed", err);
    PHY_CHECK(eth->on_state_changed(eth, ETH_STATE_DUPLEX, (void *)duplex) == ESP_OK, "send duplex event failed", err);
    if (ksz8863->link_status != link) {
        PHY_CHECK(eth->on_state_changed(eth, ETH_STATE_LINK, (void *)link) == ESP_OK, "send link event failed", err);
        ksz8863->link_status = link;
    }

    return ESP_OK;
}
```

So I just call `on_state_changed` with appropriate settings (bitrate, duplex mode).
There are no error log messages in the console.

suda-morris, 2020-12-16:
The link status should read from PHY register instead of hard coded, or the state might be different between MAC and PHY.  Could you add some delay each time link status got updated?

lexxvir, 2020-12-16:
KSZ8863 is the switch with three ports: RMII (Port3) and two regular Ethernet ports. If I understand correctly in that code we provide link status of RMII port. That port is always UP due to it's nature (PCB lines). So it's why I hard coded link status here.

Nevertheless, I've tried to add some delay before and after `eth->on_state_changed(eth, ETH_STATE_LINK..` line.
The issue does not vanish.

suda-morris, 2020-12-16:
Thanks for your explanation. So have you configured some register of KSZ8863 to use internal RMII clock, not only just configured by strap GPIO? I suspect the RMII clock at ESP32 side has a big phase deviation from KSZ8863

lexxvir, 2020-12-21:
Hi @suda-morris !

Your words about RMII clock force me to re-evaluate how KSZ8863 provides clock for the ESP32. After some investigation I found that KSZ has two modes how it outputs the clock: internally and externally. I've switched to internally mode and it seems that fixed the issue! I guess that external mode also should work normally but may be there is an issue in our PCB.

So, I'll consult with our HW developer and wait for feedback from our customers.
Just for now I close this one.

Thank you for help @suda-morris !

closed 2020-12-21 by lexxvir