espressif/esp-idf #6250: Error "Re-enable cpu cache" while using PSRAM in WROVER kit

ansee7221, 2020-12-11:
## Environment
- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    //v4.3-dev-1521-g0289d1c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // crosstool-NG esp-2020r3
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
I set up the WROVER kit to use PSRAM and it reboots with a "Re-enable cpu cache" error.

//Detailed problem description goes here.

### Expected Behavior
### Actual Behavior
### Steps to reproduce

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
	if (nvs_flash_init() != ESP_OK) {
		print_error("nvs_flash_init failed");
	}
	//gpio_reset_pin(BLINK_GPIO);
	//gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);

	pthread_cond_init(&cv, NULL);

	print_macro_info();
	initialise_wifi();
	initialize_console();

	// Register commands	   
	esp_console_register_help_command();
	register_ctt();

#ifdef EXTERNAL_SPRAM_USE

	// Create the task without using any dynamic memory allocation.
	xHandle = xTaskCreateStatic(	  
	server_main,	 // Function that implements the task.		
	"server_main", // Text name for the task.	   
	STACK_SIZE,	  // Stack size in bytes, not words.	  
	NULL, 		 // Parameter passed into the task. 	 
	5, 			// Priority at which the task is created.	   
	xStack,		  // Array to use as the task's stack.		
	&xTaskBuffer); // Variable to hold the task's data structure.

	if (xHandle == NULL) {
		print_error("task create failed (server_main)");
	} 

	xHandleConsole = xTaskCreateStatic(
	console_task, "console", 
	STACK_SIZE_CONSOLE, 
	NULL, 
	configMAX_PRIORITIES - 2, 
[... 308 lines cut ...]
Successfully installed PKI certificate
Successfully installed intermediate CA certificate
**Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).**

Core  0 register dump:
PC      : 0x400860d3  PS      : 0x00060034  A0      : 0x80085ff0  A1      : 0x3ffb0670
A2      : 0x40085d20  A3      : 0x3ffb1c18  A4      : 0x3f801ce4  A5      : 0x40085d64
A6      : 0x3ffb197c  A7      : 0x3ffbbe58  A8      : 0x800860c5  A9      : 0x3ffb0650
A10     : 0xbad00bad  A11     : 0x3ffb0670  A12     : 0x3ffb0678  A13     : 0x00060021
A14     : 0x000000fe  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x400860d0:0x3ffb0670 0x40085fed:0x3ffb06a0 0x40086471:0x3ffb06c0 0x40084231:0x3ffb06e0 0x400848ab:0x3ffbd240 0x40085e9c:0x3ffbd260

ELF file SHA256: 003d9eb7016a05a0
```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible
- [ ] sdkconfig file (attach the sdkconfig file from your project folder) 
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-12-11:
Thanks for reporting and letting us know, we will look into.

YouDONG-ESP, 2020-12-14:
@ansee7221 Could you provide a complete code to reproduce this issue?

YouDONG-ESP, 2020-12-16:
@ansee7221 We had found the cause, CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY option will put one of our wifi variable into psram. Could you replace $IDF_PATH/esp-idf/components/esp_wifi/lib by this and see if this issue fixed? Thanks.

[lib.zip](https://github.com/espressif/esp-idf/files/5700879/lib.zip)

ansee7221, 2020-12-18:
@YouDONG-ESP , It has been resolved after updating with the file you sent. Thank you very much.

Alvin1Zhang, 2020-12-18:
Thanks for reporting and updates, will reopen until the fix is available, the issue will close automatically once the fix is on GitHub. Thanks.

YouDONG-ESP, 2021-01-12:
Hi @ansee7221  it has been fixed in commit e040168a9f5697fdc68e567ecc7314cdbe61ac5b

closed 2021-01-12 by Alvin1Zhang