espressif/esp-idf #6251: pyparsing>=2.0.3,<2.4.0

Reiner1210, 2020-12-11:
Hello,

build fails with pyparsing>=2.0.3,<2.4.0

I have Linux Mint 20
and pyparsing 2.4.6-1 is installed

Alvin1Zhang, 2020-12-11:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

dobairoland, 2020-12-11:
@Reiner1210 There is a compatibility issue with the latest pyparsing versions. The requirement `pyparsing>=2.0.3,<2.4.0` was set accordingly.

However, it should not affect you. You can have pyparsing 2.4.6 installed on your computer (for other applications) but when you run `install.sh` and `. export.sh` you switch to a new Python environment where the IDF-compatible version can be installed and used.

If this isn't helping you then please give more information as suggested before me.

Alvin1Zhang, 2020-12-18:
Thanks for reporting, will close now, feel free to reopen with more details, thanks.

closed 2020-12-18 by Alvin1Zhang