espressif/esp-idf #6253: Undefined reference to app_main freertos.a

GunashekarG, 2020-12-11:
OS: Ubuntu Mate
IDFV: v4.3-dev-1901-g178b122c1

Example: tusb_serail_device

While compiling I'm getting the error as shown in below screenshot

[image]

sdkconfig file has been attached below.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5679063/sdkconfig.txt)

igrr, 2020-12-11:
Hi @GunashekarG, ESP32-S2 chip is only supported by the CMake build system. It looks like you are using the legacy GNU Make based build system to compile this example. Assuming you want to get this example running on the ESP32-S2 and not on the ESP32 (which doesn't have a USB peripheral), you will need to switch to CMake. Please follow https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32s2/get-started/index.html to get started. CMake specific instructions start at step 8.

Alvin1Zhang, 2020-12-14:
@GunashekarG Thanks for reporting and please help follow the notes by @igrr , feel free to reopen, thanks.

GunashekarG, 2020-12-14:
Thanks for the response guys. Will definitely try the lead. Thanks again.

GunashekarG, 2020-12-14:
OS: Linux(Ubuntu Mate)
ESP-IDF Vesrion: v4.2
Chip used: ESP32-S2
Exapmle: Hello_world

While trying to build the in built example hello world code by selecting the target as esp32s2, I am getting the following error.

[image]

@igrr can you please help me out with this issue?

Alvin1Zhang, 2020-12-29:
Thanks for reporting, sorry that we do not support make build system for ESP32-S2, will close the ticket, thanks.

closed 2020-12-29 by Alvin1Zhang