espressif/esp-idf #6256: i2c driver crash with invalid clock rate/flags due to I2C_CHECK() in critical section.

lsuchocki, 2020-12-11:
In components/driver/i2c.c, on or about line 667:
`i2c_param_config()` checks the `i2c_conf->clk_flags` and `i2c_conf->master.clk_speed` values with 
`I2C_CHECK(i2c_get_clk_src(i2c_conf) != I2C_SCLK_MAX, I2C_CLK_FLAG_ERR_STR, ESP_ERR_INVALID_ARG);`

Only this is performed after interrupts have been disabled with `I2C_ENTER_CRITICAL()`, such that if they are invalid, the resulting `printf` in the macro will lead to a crash.

This check should be placed earlier at the beginning of the function with the rest of the checks with something like:
`if (i2c_conf->mode != I2C_MODE_SLAVE)
I2C_CHECK(i2c_get_clk_src(i2c_conf) != I2C_SCLK_MAX, I2C_CLK_FLAG_ERR_STR, ESP_ERR_INVALID_ARG);`

Alvin1Zhang, 2020-12-14:
@lsuchocki Thanks for reporting, and we already fixed the issue. The fix is now under sync into GitHub, will update once the fix is available. Thanks.

Alvin1Zhang, 2020-12-14:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/a3b59aeeb5fe78d768667636f135eaf1470fd750, feel free to reopen if the issue still happens. Thanks.

closed 2020-12-14 by Alvin1Zhang

brunohorta82, 2021-02-15:
Hello I am using the ESP32-S2 and I have this error:;
I Using platformIO and Wire.begin();

i2c: i2c_param_config(644): i2c clock choice is invalid, please check flag and frequency
[  1199][E][esp32-hal-i2c.c:1828] i2cInit(): i2c_param_config failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.