espressif/esp-idf #6257: static ip set in wifi- station mode in esp32

kendo55, 2020-12-13:
Hi 
like in issue #5266 describe, the static station address don't work.
i modified the station_example_main.c ( ESP-IDF v4.1-beta1)
like this:

```
static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
        s_retry_num = 0;
 
        tcpip_adapter_ip_info_t ip_info;
        ip_info.ip.addr = ipaddr_addr("192.168.1.166");
        ip_info.gw.addr = ipaddr_addr("192.168.1.4");
        ip_info.netmask.addr = ipaddr_addr("255.255.255.0");
        tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &ip_info);

        ESP_ERROR_CHECK(tcpip_adapter_dhcpc_start(TCPIP_ADAPTER_IF_STA));

        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
    }
```

felmue, 2020-12-14:
HI @kendo55 

check out the solution presented [here](https://www.esp32.com/viewtopic.php?t=14689).

Thanks
Felix

david-cermak, 2020-12-17:
@kendo55 
It is recommended to use esp-netif API on `v4.1` and later.
Please update the station example in `void wifi_init_sta(void)` to
``` c
    esp_netif_t *sta = esp_netif_create_default_wifi_sta();
    esp_netif_dhcpc_stop(sta);
    esp_netif_ip_info_t ip_info;
    ip_info.ip.addr = ipaddr_addr("192.168.10.133");
    ip_info.gw.addr = ipaddr_addr("192.168.10.1");
    ip_info.netmask.addr = ipaddr_addr("255.255.255.0");
    esp_netif_set_ip_info(sta, &ip_info);
```
(instead of this `esp_netif_create_default_wifi_sta()` [line](https://github.com/espressif/esp-idf/blob/release/v4.1/examples/wifi/getting_started/station/main/station_example_main.c#L73) )

If for some reason the legacy api needs to be used, just paste the below
``` c
    tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);
    tcpip_adapter_ip_info_t ip_info;
    ip_info.ip.addr = ipaddr_addr("192.168.10.133");
    ip_info.gw.addr = ipaddr_addr("192.168.10.1");
    ip_info.netmask.addr = ipaddr_addr("255.255.255.0");
    tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &ip_info);

```
after the  `esp_netif_create_default_wifi_sta()` line. Plus need to enable the tcpip-adapter compatibility lager in 
```
(Top) → Component config → ESP NETIF Adapter
[*] Enable backward compatible tcpip_adapter interface
```

kendo55, 2020-12-17:
@david-cermak thanks and 
sorry @felmue for the late feedback. Your link solved the "problem".
I wanted to test the definition of the DNA and give feedback here.
like this:
```
  //** STATIC IP ******************************************
    // https://www.esp32.com/viewtopic.php?t=14689
    //esp_netif_create_default_wifi_sta();
    esp_netif_t *my_sta = esp_netif_create_default_wifi_sta();
    esp_netif_dhcpc_stop(my_sta);

     esp_netif_ip_info_t ip_info;

     IP4_ADDR(&ip_info.ip, 192, 168, 1, 165);
     IP4_ADDR(&ip_info.gw, 192, 168, 1, 4);
     IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);
     // typedef struct {
     //    esp_ip4_addr_t ip;      /**< Interface IPV4 address */
     //    esp_ip4_addr_t netmask; /**< Interface IPV4 netmask */
     //    esp_ip4_addr_t gw;      /**< Interface IPV4 gateway address */
     // } esp_netif_ip_info_t;

     // 	* @param[in]  esp_netif Handle to esp-netif instance
     //  * @param[in] ip_info IP information to set on the specified interface
     esp_netif_set_ip_info(my_sta, &ip_info);

     esp_netif_dns_info_t	dns_info;
     IP4_ADDR(&dns_info.ip.u_addr.ip4, 192, 168, 1, 4);

     esp_netif_set_dns_info(my_sta, ESP_NETIF_DNS_MAIN, &dns_info);
     // esp_err_t esp_netif_set_dns_info(esp_netif_t *esp_netif, esp_netif_dns_type_t type, esp_netif_dns_info_t *dns);
     /*
       * @param[in]  esp_netif Handle to esp-netif instance
       * @param[in]  type Type of DNS Server to set: ESP_NETIF_DNS_MAIN, ESP_NETIF_DNS_BACKUP, ESP_NETIF_DNS_FALLBACK
       * @param[in]  dns  DNS Server address to set
       *
       * @return
       *      - ESP_OK on success
       *      - ESP_ERR_ESP_NETIF_INVALID_PARAMS invalid params
       */
    //** END STATIC IP ******************************************
```

closed 2020-12-17 by kendo55