espressif/esp-idf #6259: Wrong descrption of Freertos-smp changed in latest documentation.

lorddoskias, 2020-12-13:
In the [latest](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html#critical-sections-disabling-interrupts) documentation guide  one of the changes related to locking states that all critical section related macros really boil down to calling `vtaskentercritical`. This creates the wrong impression that calling `vtaskentercritical` is the same as calling `taskENTER_CRITICAL`. In fact eps32's related changes are really in `vPortEnterCritical`

Alvin1Zhang, 2020-12-14:
Thanks for reporting, we will look into.

projectgus, 2020-12-14:
Hi @lorddoskias,

Thanks for letting us know. I think the functionality has been updated without updating the docs to match, and I notice a comment in portmacro.h is the same. Will fix.

closed 2021-02-10 by projectgus