espressif/esp-idf #6261: Got a lot of warning: "LWIP_HOOK_FILENAME" redefined

AxelLin, 2020-12-14:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3-dev-2136-gb0150615dff5
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I have a customized ip4_route_src_hook_local implementation for my project (to override the ip4_route_src_hook() in lwip)

It was working fine before pull esp-idf update today.
After pull the latest esp-idf, now I got a lot of build warnings.

Generated /home/axel/esp/esp-idf-dev/apps/myproj/build/bootloader/bootloader.bin
[341/1064] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/ping/esp_ping.c.obj
In file included from /home/axel/esp/esp-idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /home/axel/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:40,
                 from /home/axel/esp/esp-idf/components/lwip/apps/ping/esp_ping.c:18:
/home/axel/esp/esp-idf/components/lwip/port/esp32/include/lwipopts.h:770: warning: "LWIP_HOOK_FILENAME" redefined
 #define LWIP_HOOK_FILENAME              "lwip_default_hooks.h"

<command-line>: note: this is the location of the previous definition
[344/1064] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/ping/ping.c.obj
In file included from /home/axel/esp/esp-idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /home/axel/esp/esp-idf/components/lwip/apps/ping/ping.c:41:
/home/axel/esp/esp-idf/components/lwip/port/esp32/include/lwipopts.h:770: warning: "LWIP_HOOK_FILENAME" redefined
 #define LWIP_HOOK_FILENAME              "lwip_default_hooks.h"

### Expected Behavior
The existing project should work fine without "LWIP_HOOK_FILENAME" redefined warning.
### Actual Behavior
See above messages.
Looks like now it does not allow user to add a customized hook.
### Steps to reproduce

Just build the project.

### Code to reproduce this issue
Add below in CMakeLists.txt to build:
target_compile_definitions(${lwip} PRIVATE "-DLWIP_HOOK_FILENAME=\"ip4_route_local.h\"")
target_compile_options(${lwip} PRIVATE "-I${PROJECT_DIR}/main/lwip_local/include")
target_sources(${lwip} PRIVATE "${PROJECT_DIR}/main/lwip_local/ip4_route_local.c")

Alvin1Zhang, 2020-12-14:
Thanks for reporting and letting us know, we will look into.

KaeLL, 2020-12-15:
You should see the shit ton of macro redefinition warnings spilled if you disable LwIP assert checks on menuconfig. I'm running v4.2, not sure if it has already been fixed on master, I bet it hasn't.

AxelLin, 2020-12-15:
@KaeLL 
I think we are talking about different thing, it's nothing to do with LwIP assert checks or not.
The redefine warning is introduced by cf60ec0ffc7e057f9ef66e4b6fc2abe3339b29e0 in master tree.

KaeLL, 2020-12-15:
@AxelLin 
I meant the same thing happens (but worse) when you disable the LwIP assert checks, thought it's with the `LWIP_NOASSERT` macro instead of `LWIP_HOOK_FILENAME`.

david-cermak, 2020-12-16:
@AxelLin Thanks for the report, I haven't thought of user defined hooks.
We are going to fix it with additional `ESP-IDF`'s hook in the `"lwip_default_hooks.h"` file
[0001-lwip-Add-IDF-hook-filename-to-customize-lwip-hooks.patch.txt](https://github.com/espressif/esp-idf/files/5701695/0001-lwip-Add-IDF-hook-filename-to-customize-lwip-hooks.patch.txt)

which I think makes sense as it allows additional user defined hooks to be added to the IDF standard (default) hooks, such as `lwip_hook_tcp_isn()` without redefining them.

AxelLin, 2020-12-16:
Hi @david-cermak 
Still got the same build warning with above patch.

BTW, I don't find ESP_IDF_LWIP_HOOK_FILENAME in esp-idf, is there something missed?
LWIP_HOOK_FILENAME is defined in components/lwip/port/esp32/include/lwipopts.h

david-cermak, 2020-12-16:
@AxelLin The ideas is that it's the `ESP_IDF_LWIP_HOOK_FILENAME` macro now which has to defined
``` cmake
target_compile_definitions(${lwip} PRIVATE "-DESP_IDF_LWIP_HOOK_FILENAME =\"ip4_route_local.h\"")
```

AxelLin, 2020-12-16:
> @AxelLin The ideas is that it's the `ESP_IDF_LWIP_HOOK_FILENAME` macro now which has to be defined
> 
> ```cmake
> target_compile_definitions(${lwip} PRIVATE "-DESP_IDF_LWIP_HOOK_FILENAME =\"ip4_route_local.h\"")
> ```

OK, that works.
Thanks .

AxelLin, 2020-12-16:
I want to make my project to be built with different IDF versions.

For esp-idf-v4.3.x, now I need to use
target_compile_definitions(${lwip} PRIVATE "-DESP_IDF_LWIP_HOOK_FILENAME =\"ip4_route_local.h\"")
For older branches, I need to use
target_compile_definitions(${lwip} PRIVATE "-DLWIP_HOOK_FILENAME =\"ip4_route_local.h\"")

What is the recommend way to check esp-idf version in CMakeLists.txt?

david-cermak, 2020-12-16:
You can use the below variables from [the build/project vars](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#build-project-variables)
`IDF_VERSION_MAJOR`, `IDF_VERSION_MINOR`, `IDF_VERSION_PATCH`

``` cmake
if (${IDF_VERSION_MAJOR} STREQUAL "4" AND ${IDF_VERSION_MINOR} STREQUAL "3")
    message("version is 4.3")
    target_compile_definitions(${lwip} PRIVATE "-DESP_IDF_LWIP_HOOK_FILENAME =\"ip4_route_local.h\"")
elseif (${IDF_VERSION_MAJOR} STREQUAL "4" AND ${IDF_VERSION_MINOR} STREQUAL "2")
    message("version is 4.2")
    target_compile_definitions(${lwip} PRIVATE "-DLWIP_HOOK_FILENAME =\"ip4_route_local.h\"")
endif()
```

closed 2021-01-15 by espressif-bot