espressif/esp-idf #6262: ERROR: Could not find a version that satisfies the requirement bidict>=0.21.0

GunashekarG, 2020-12-14:
OS: Ubuntu Mate
Version: v4.2

While installing the ESP-IDF for ESPS2, I am finding an error as shown below.
[image]

Requirements file is also attached below.

[requirements.txt](https://github.com/espressif/esp-idf/files/5686780/requirements.txt)

My current pip version is: pip 20.3.1 (python 2.7)

Can anyone please solve this issue?

linweifu, 2020-12-14:
Me too, waiting for a solution

dobairoland, 2020-12-15:
Hi @GunashekarG & @fetullah-usr. We already fixed this but the change haven't made it into Github yet. The following checksum will become a clickable link when it does: dc98d0aeee8b4766d9b312c6065b503073c42105

Until then the workaround is to add manually the following line into the `requirements.txt`:
`python-socketio<5; python_version<="2.7"`

We are sorry for the inconvenience.

dobairoland, 2020-12-15:
The long term recommendation is to switch to Python 3: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#setting-up-python-3-as-default-for-ubuntu-and-debian

That will also fix the above issue and avoid future Python 2 incompatibilities.

GunashekarG, 2020-12-15:
It's working now after updating to Python3. Thank you guys. Cheers.

dobairoland, 2020-12-15:
Thank you for the feedback. I keep this open for others to see it until the fix will be available.

samiralavi, 2020-12-17:
Another easy solution could be adding "python3" explicitly in the install.sh script.

ellipsistechnology, 2020-12-30:
This worked for me also - thanks for the help.
One note (probably only an issue on Mac) if you are having issues convincing it to run with Python3 (as is common on recent version of Mac OS) you can force the `tools/idf_tools.py` script to use Python3 by changing its first line to `#!/usr/bin/env python3` and then run `./install.sh`.

dobairoland, 2021-01-05:
The fix is available in master now: https://github.com/espressif/esp-idf/commit/dc98d0aeee8b4766d9b312c6065b503073c42105

closed 2021-01-05 by dobairoland

MohamedZaheri, 2021-01-27:
@dobairoland , the fix is merged to master but not to the V4.2 release branch, is it? Because I still have same issue using the V4.2 branch.

jtchitty, 2021-01-27:
> The long term recommendation is to switch to Python 3: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#setting-up-python-3-as-default-for-ubuntu-and-debian
> 
> That will also fix the above issue and avoid future Python 2 incompatibilities.

This is very likely going to cause problems for other things on the system. It has for us, several times. It breaks other things in very subtle and hard-to-debug ways. In fact, this is the one major complaint I have with the ESP-IDF Getting Started for Ubuntu and Debian.

The long-term plan _should_ be to update the ESP-IDF to not require a specific system-wide default. ;)

dobairoland, 2021-01-27:
@MohamedZaheri The fix is in v4.2 as well: https://github.com/espressif/esp-idf/commit/9ff5f94233d5e8b5a124ab9b58b913f049e47597

dobairoland, 2021-01-27:
> The long-term plan should be to update the ESP-IDF to not require a specific system-wide default. ;)

@jtchitty Yes, that is on the table.

9ae8sdf76, 2021-02-22:
I just downloaded - as of today - ESP-IDF from your website and ran into this problem. Did this fix make its way into the official release? 

After switching the repository to master, I of course had no issues with installation. Thank you.

dobairoland, 2021-02-23:
@9ae8sdf76 Unfortunately, the fix haven't made into v4.2. It will be included in v4.2.1. 

It is already available in the branch https://github.com/espressif/esp-idf/tree/release/v4.2. You can already use that branch. Or you can manually apply the patch to your v4.2 copy.

We are sorry for the inconvenience.