espressif/esp-idf #6263: Error while building the hello_world in ESP32-S2

GunashekarG, 2020-12-14:
OS: Linux(Ubuntu Mate)
ESP-IDF Vesrion: v4.2
Chip used: ESP32-S2
Exapmle: Hello_world

While trying to build the in built example hello world code by selecting the target as esp32s2, I am getting the following error.

[image]

Can anyone please solve this issu?

Kiznoh, 2020-12-14:
Hi @GunashekarG, 

As posted here: https://github.com/espressif/esp-idf/issues/6253#issuecomment-743175431, you'll need to use the CMake build system.

Follow this guide to get started building applications for the ESP32: https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32s2/get-started/index.html

Alvin1Zhang, 2020-12-29:
Thanks for reporting, sorry that we do not support make build system for ESP32-S2, will close the ticket, thanks.

closed 2020-12-29 by Alvin1Zhang