espressif/esp-idf #6264: [ESP32-S2] xTaskCreateStatic broken in master

chegewara, 2020-12-14:
I know `xTaskCreateStatic` with SPIRAM hasbeen working few weeks ago on S2, but now it is broken. I am guessing it is because in S2 menuconfig `SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` is missing. 

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <stdio.h>

#define STACK_SIZE 200
StaticTask_t xTaskBuffer;
StackType_t *xStack;

void test_task(void*p)
{
    while(1)
    {
        vTaskDelay(100);
    }
}

void app_main(void)
{
  xStack = (uint8_t*)heap_caps_calloc(1, 10*1024, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT);
  xTaskCreateStatic(test_task, "test", 10*1024, NULL, 6, xStack, &xTaskBuffer);
}
```

```
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025131
0x40025131: esp_restart_noos_dig at /home/chegewara/esp/esp-idf/components/esp_system/system_api.c:62 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x1810
load:0x4004c000,len:0x9d8
load:0x40050000,len:0x2e08
entry 0x4004c1ec
I (48) boot: ESP-IDF v4.3-dev-2136-gb0150615d 2nd stage bootloader
I (48) boot: compile time 15:20:21
I (48) boot: chip revision: 0
I (52) boot.esp32s2: SPI Speed      : 80MHz
I (56) boot.esp32s2: SPI Mode       : DIO
I (61) boot.esp32s2: SPI Flash Size : 2MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=05f84h ( 24452) map
I (122) esp_image: segment 1: paddr=00015fac vaddr=3ffbea10 size=0262ch (  9772) load
I (128) esp_image: segment 2: paddr=000185e0 vaddr=40024000 size=00404h (  1028) load
I (134) esp_image: segment 3: paddr=000189ec vaddr=40024404 size=0762ch ( 30252) load
I (151) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=11424h ( 70692) map
I (166) esp_image: segment 5: paddr=0003144c vaddr=4002ba30 size=02fd4h ( 12244) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (188) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (188) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (192) spiram: Found 16MBit SPI RAM device
I (197) spiram: SPI RAM mode: sram 80m
I (201) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (208) cpu_start: Pro cpu up.
I (431) spiram: SPI SRAM memory test OK
I (431) spiram: Instructions copied and mapped to SPIRAM
I (448) spiram: Read only data copied and mapped to SPIRAM
I (509) cpu_start: Pro cpu start user code
I (509) cpu_start: cpu freq: 160000000
I (509) cpu_start: Application information:
I (511) cpu_start: Project name:     main
I (516) cpu_start: App version:      v4.3-dev-2136-gb0150615d
I (522) cpu_start: Compile time:     Dec 14 2020 15:20:19
I (529) cpu_start: ELF file SHA256:  df5507cc40cec4f2...
I (535) cpu_start: ESP-IDF:          v4.3-dev-2136-gb0150615d
I (541) heap_init: Initializing. RAM available for dynamic allocation:
I (548) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (554) heap_init: At 3FFC1958 len 0003A6A8 (233 KiB): DRAM
I (561) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (567) spiram: Adding pool of 1856K of external SPI memory to heap allocator
I (575) spi_flash: detected chip: generic
I (579) spi_flash: flash io: dio
W (583) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (597) cpu_start: Starting scheduler on PRO CPU.
/home/chegewara/esp/esp-idf/components/freertos/tasks.c:641 (xTaskCreateStaticPinnedToCore)- assert failed!

abort() was called at PC 0x4002973a on core 0
0x4002973a: xTaskCreateStaticPinnedToCore at /home/chegewara/esp/esp-idf/components/freertos/tasks.c:641 (discriminator 3)

Backtrace:0x400286f3:0x3ffc2af0 0x40028f89:0x3ffc2b10 0x4002da16:0x3ffc2b30 0x4002973a:0x3ffc2ba0 0x40084d6e:0x3ffc2be0 0x40091202:0x3ffc2c10 0x4002a685:0x3ffc2c30
0x400286f3: panic_abort at /home/chegewara/esp/esp-idf/components/esp_system/panic.c:367

0x40028f89: esp_system_abort at /home/chegewara/esp/esp-idf/components/esp_system/system_api.c:108

0x4002da16: abort at /home/chegewara/esp/esp-idf/components/newlib/abort.c:46

0x4002973a: xTaskCreateStaticPinnedToCore at /home/chegewara/esp/esp-idf/components/freertos/tasks.c:641 (discriminator 3)

0x40084d6e: xTaskCreateStatic at /home/chegewara/esp/esp-idf/examples/get-started/sample_project/build/../../../../components/freertos/include/freertos/task.h:556
 (inlined by) app_main at /home/chegewara/esp/esp-idf/examples/get-started/sample_project/build/../main/main.c:20

0x40091202: main_task at /home/chegewara/esp/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)

0x4002a685: vPortTaskWrapper at /home/chegewara/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: df5507cc40cec4f2
```

Its not that important right now, but would be nice to have it supported again.
Thanks

Alvin1Zhang, 2020-12-15:
Thanks for pointing out, we will look into.

projectgus, 2021-01-04:
Hi @chegewara,

Thanks for pointing this out. There was a window in time between commits a3c90bf5 and 199b2cb when the check for `portVALID_STACK_MEM(pxStackBuffer)` was removed from `xTaskCreateStaticPinnedToCore()` on all targets.

This was a regression that was "fixed", but the side effect was that for a while it was possible to create stacks in PSRAM on ESP32-S2 (or on ESP32 when the "SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY" was not set)!

The root cause here is that we haven't actually added support for placing stack in PSRAM on ESP32-S2. We will look into this, it should be possible to support once we've added tests for it.

Thanks again.

elpamyelhsa, 2022-05-29:
It there an ETA for a release of ESP-IDF with SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY support on EPS32S2?

esp-wzh, 2022-05-29:
Hi, @elpamyelhsa , I'm about to start working on this feature and expect to release a supported version in a few weeks.

AHU-HUI, 2022-06-06:
Hi, @elpamyelhsa @chegewara. It is found that this feature has been supported for esp32s2 and esp32s3 after this MR: https://github.com/espressif/esp-idf/commit/934a3951b8a66133bf1e0ac7fe733a8ea45ec9a1
You can use this feature on the latest master, or apply the changes to the version you are currently using.

esp-wzh, 2022-06-06:
Hi, [@elpamyelhsa](https://github.com/elpamyelhsa) [@chegewara](https://github.com/chegewara). It is found that this feature has been supported for esp32s2 and esp32s3 after this MR: https://github.com/espressif/esp-idf/commit/934a3951b8a66133bf1e0ac7fe733a8ea45ec9a1
You can use this feature on the latest master, or apply the changes to the version you are currently using.

esp-wzh, 2022-10-20:
@marchingband Yes, ESP32 is the first chip to support this feature, you can enable it in 
`(Top) → Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config → Allow external memory as an argument to xTaskCreateStatic`

marchingband, 2022-10-20:
My apologies I thought this was an ESP32-Arduino thread.

Alvin1Zhang, 2022-11-16:
Thanks for reporting, feel free to reopen.

closed 2022-11-16 by Alvin1Zhang

lanistor, 2023-09-29:
> xTaskCreateStatic

Why i cannot found this configuration in menuconfig? Using ESP32

lanistor, 2023-09-29:
> @marchingband Yes, ESP32 is the first chip to support this feature, you can enable it in `(Top) → Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config → Allow external memory as an argument to xTaskCreateStatic`

@esp-wzh Why i cannot found this configuration in menuconfig? Using ESP32

Dazza0, 2023-09-29:
@lanistor Which `ESP-IDF` tag are you on? The `SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` exists on `v4.4.6`

lanistor, 2023-09-30:
> @lanistor Which `ESP-IDF` tag are you on? The `SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` exists on `v4.4.6`

Got it, i'm using v4.4.6, thanks.