espressif/esp-idf #6265: example_connect() repeatedly disconnecting

bsalha1, 2020-12-14:
## Environment

- Development Kit: none
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-1901-g178b122c1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Eclipse with ESP-IDF plugin
- Power Supply: USB

## Problem Description

I'm using the example program under the protocol category "examples\protocols\sockets\tcp_server" and the example_connect() function which is configured to connect to my 2.4 GHz Wi-Fi network repeatedly disconnects from the network dozens of times before successfully connecting and initializing the tcp-server. Even after intializing the tcp-server, the device continues to disconnect from the Wi-Fi network.

### Expected Behavior
Connect to Wi-Fi network -> Initialize TCP server

### Actual Behavior
Failed to connect to Wi-Fi network several times -> Connec to Wi-Fi network -> Initialize TCP Server -> Failed to connect to Wi-Fi

### Steps to reproduce
Just run the "examples\protocols\sockets\tcp_server" example

### Code to reproduce this issue

```cpp
#include <string.h>
#include <sys/param.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "protocol_examples_common.h"

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include <lwip/netdb.h>

void app_main(void)
{
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(example_connect());
}
```

## Debug Logs

```
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x15e6c ( 89708) map
I (134) esp_image: segment 1: paddr=0x00025e94 vaddr=0x3ffc5300 size=0x039b4 ( 14772) load
I (138) esp_image: segment 2: paddr=0x00029850 vaddr=0x40022000 size=0x00404 (  1028) load
I (141) esp_image: segment 3: paddr=0x00029c5c vaddr=0x40022404 size=0x063bc ( 25532) load
I (156) esp_image: segment 4: paddr=0x00030020 vaddr=0x40080020 size=0x6afb0 (438192) map
I (250) esp_image: segment 5: paddr=0x0009afd8 vaddr=0x400287c0 size=0x0cb3c ( 52028) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (286) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
[... 658 lines cut ...]
I (823853) example_connect: Wi-Fi disconnected, trying to reconnect...
I (825913) example_connect: Wi-Fi disconnected, trying to reconnect...
I (826033) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (826033) wifi:state: init -> auth (b0)
I (826033) wifi:state: auth -> assoc (0)
I (826043) wifi:state: assoc -> run (10)
W (826053) wifi:<ba-add>idx:0 (ifx:0, 20:c0:47:56:5f:26), tid:0, ssn:0, winSize:64
I (828213) wifi:connected with salha_24, aid = 8, channel 1, BW20, bssid = 20:c0:47:56:5f:26
I (828213) wifi:security: WPA2-PSK, phy: bgn, rssi: -87
I (828213) wifi:pm start, type: 1

I (828263) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (829433) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:7edf:a1ff:fe08:df34, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (875623) esp_netif_handlers: example_connect: sta ip: 192.168.1.254, mask: 255.255.255.0, gw: 192.168.1.1
I (875623) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.254
W (1095323) wifi:<ba-del>idx
W (1098203) wifi:<ba-add>idx:0 (ifx:0, 20:c0:47:56:5f:26), tid:0, ssn:13, winSize:64
W (1270053) wifi:<ba-del>idx
W (1288663) wifi:<ba-add>idx:0 (ifx:0, 20:c0:47:56:5f:26), tid:0, ssn:20, winSize:64
W (1393943) wifi:<ba-del>idx

```

## Other items if possible
[sdkconfig_and_elf.zip](https://github.com/espressif/esp-idf/files/5690310/sdkconfig_and_elf.zip)

Alvin1Zhang, 2020-12-15:
Thanks for reporting and letting us know, we will look into.

espxiehang, 2020-12-17:
Hi  @bsalha1 , From the log, we can see that the RSSI of AP is very low.

Two suggestions:

1. Check whether the antenna of station(ESP32) and AP( salha_24 ) is normal.

2. Put station and AP closer together.

I (828213) wifi:connected with salha_24, aid = 8, channel 1, BW20, bssid = 20:c0:47:56:5f:26
I (828213) wifi:security: WPA2-PSK, phy: bgn, **rssi: -87**

bsalha1, 2020-12-17:
Yes, thank you. Moving the station closer to the AP did work. Is there any way I can improve the strength of the esp32 antenna, because I have to move to about 10 feet away from my router in order for it to connect? I know there is a LR mode but it's incompatible with my router.

bsalha1, 2020-12-17:
I actually attached a better antenna, and it worked. Sorry I'm a bit of a noob at this. Thank you!

closed 2020-12-17 by bsalha1