espressif/esp-idf #6267: esp_wifi_sta_get_ap_info does not appear to fill wifi_country_t (e.g. country.cc) details

BennyE, 2020-12-14:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-S2 / featherS2 (UnexpectedMaker)]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-2136-gb0150615d
- Build System:         [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

While I was working on adding the "country code" to CircuitPython (via this: https://github.com/BennyE/circuitpython/commit/210c0ec89c88b50089067f74c2f0893fdd8613b4) I noticed that the function worked fine while esp_wifi_scan_get_ap_records() would fill wifi_ap_record_t, but not if esp_wifi_sta_get_ap_info would do the same. I read https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_sta_get_ap_infoP16wifi_ap_record_t and there was no note that the country would be missing.

### Expected Behavior

esp_wifi_sta_get_ap_info should return wifi_ap_record_t including country (wifi_country_t) with correct values.

### Actual Behavior

esp_wifi_sta_get_ap_info should returns wifi_ap_record_t where country is empty.

### Steps to reproduce

To ensure that I didn't do something wrong in CircuitPython, I have simply done the same in your espressif/esp-idf example code. See below

### Code to reproduce this issue

```cpp
// esp-idf -> examples -> getting_started -> station -> code
// with the following additions
#include "esp_wifi_types.h"

// .. inserted after this ..
//    } else {
//        ESP_LOGE(TAG, "UNEXPECTED EVENT");
//    }
    wifi_ap_record_t apinfo;
    memset(&apinfo, 0, sizeof(apinfo));
    esp_wifi_sta_get_ap_info(&apinfo);
    ESP_LOGI(TAG, "RSSI: %d", apinfo.rssi);
    ESP_LOGI(TAG, "Country (CC): %s", (char *)apinfo.country.cc); // otherwise compiler wasn't happy
    ESP_LOGI(TAG, "Channel: %d", apinfo.primary);
    ESP_LOGI(TAG, "SSID: %s", apinfo.ssid);

```

## Debug Logs

Please note that Country (CC): is "" instead of the expected value "DE". 

```
I (659) phy: phy_version: 603, 72dfd77, Jul 7 2020, 19:57:05, 0, 2
I (659) wifi:mode : sta (7c:df:a1:03:b7:a0)
I (659) wifi:enable tsf
I (659) wifi station: wifiinitsta finished.
I (779) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (779) wifi:state: init -> auth (b0)
I (809) wifi:state: auth -> assoc (0)
I (819) wifi:state: assoc -> run (10)
I (3029) wifi:connected with Stellar-DSPSK, aid = 1, channel 1, BW20, bssid = dc:08:56:36:06:a0
I (3029) wifi:security: WPA2-PSK, phy: bgn, rssi: -44
I (3029) wifi:pm start, type: 1
I (3049) wifi:AP's beacon interval = 102400 us, DTIM period = 2
W (4439) wifi:<ba-add>idx:0 (ifx:0, dc:08:56:36:06:a0), tid:6, ssn:2, winSize:64
W (5409) wifi:<ba-add>idx:1 (ifx:0, dc:08:56:36:06:a0), tid:0, ssn:0, winSize:64
I (5429) espnetifhandlers: sta ip: 192.168.11.149, mask: 255.255.255.0, gw: 192.168.11.254
I (5429) wifi station: got ip:192.168.11.149
I (5429) wifi station: connected to ap SSID:Stellar-DSPSK password:testing12345
I (5439) wifi station: RSSI: -44
I (5439) wifi station: Country (CC):
I (5449) wifi station: Channel: 1
I (5449) wifi station: SSID: Stellar-DSPSK
```

Alvin1Zhang, 2020-12-15:
Thanks for the detailed report and letting us know, we will look into.

Alvin1Zhang, 2020-12-15:
Thanks for reporting, we have a fix under internal reviewing, the issue will close automatically when the fix is available on GitHub. Thanks.

closed 2021-01-05 by espressif-bot