espressif/esp-idf #6268: New Component - ImportError: No module named click

Oleg-Perevyshin, 2020-12-15:
Hello!
Mac OS + VSCode (ESP-IDF plugin)
I want to create a new component and write in the command line "idf.py -C components create-component wifi".
But I am getting ImportError: No module named click.

brianignacio5, 2020-12-15:
Try to create a terminal using `Open ESP-IDF Terminal` command, which loads the extension settings into the terminal. If you just open a terminal, our extension settings are not included in the environment variables.

Oleg-Perevyshin, 2020-12-15:
```
poe@MBP-Oleg SOI-WiFi-General % printenv PATH        
/Users/poe/esp/esp-idf/tools:/Library/Frameworks/Python.framework/Versions/3.9/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Users/poe/esp/esp-idf/tools:/Library/Frameworks/Python.framework/Versions/3.9/bin
poe@MBP-Oleg SOI-WiFi-General % printenv IDF_PATH                         
/Users/poe/esp/esp-idf
poe@MBP-Oleg SOI-WiFi-General % idf.py -C components create-component wifi
Traceback (most recent call last):
  File "/Users/poe/esp/esp-idf/tools/idf.py", line 45, in <module>
    from idf_py_actions.tools import (executable_exists, idf_version, merge_action_lists, realpath)  # noqa: E402
  File "/Users/poe/esp/esp-idf/tools/idf_py_actions/tools.py", line 1, in <module>
    import click
ImportError: No module named click
poe@MBP-Oleg SOI-WiFi-General % 
```
Paths to esp-idf are set globally.

brianignacio5, 2020-12-16:
the thing is, you are calling idf.py without specifying python interpreter. When you installed ESP-IDF and executed `$IDF_PATH/install.sh` a virtual environment python is created where all ESP-IDF dependencies are created. Are you using this python as your global python interpreter ? Based on your PATH, I assume you are using system level python which doesn't have the required python packages.

Ways to fix
A) Run `$IDF_PATH/export.sh` (assuming you already did $IDF_PATH/install.sh to install ESP-IDF tools and virtual environment)  on your terminal then run your `idf.py` command. (You can check changes in env variables)
B) Configure the extension using `ESP-IDF: Configure ESP-IDF extension` and launch a terminal with `ESP-IDF: Open ESP-IDF Terminal` (this basically run export.sh before you are able to type any terminal command)

closed 2020-12-16 by Oleg-Perevyshin

mianqi2016, 2021-09-20:
I installed both esp-idf-v4.3 and esp-idf-v3.3.5 on Ubuntu 20.04, esp-idf-v4.3 used well, I can ran "idf.py menuconfig", while, since I want to use esp-adf, so I need to use esp-idf-v3.3.5, I have ran "./install.sh" and ". ./export.sh" under esp-idf-v3.3.5, but, when I ran "idf.py menuconfig", it was prompted:"ImportError: No module named click", I found in the **prompt(https://ibb.co/Ht4MQDT),** it searched tolls in v4.3 directory, how can I change it to v3.3.5?

epneo-banuprakash, 2022-05-18:
> 

you just run ./install.sh and run . . /export.sh ,its worked for me and make sure the **which idf.py** and **printenv IDF_PATH** these two commands should be give output, and check python should be have.

[image]

and i am trying to do Automation with Jenkins, how we doing like compiling the code manually, so do u have any idea how to do this, because i am getting error idf.py not found in the jenkins console output