espressif/esp-idf #6269: MQTT publish outside of mqtt handler takes up to couple of seconds (ESP32-S2)

adokitkat, 2020-12-15:
## Environment

- Development Kit:    **ESP32-S2 Saola-1**
- Kit version:    **v1.2**
- Module or chip used:   **ESP32-S2-WROVER**
- IDF version:
   **ESP-MDF with ESP-IDF v4.2-dev-2084-g98d5b5dfd-dirty**
   **ESP-IDF v4.3-dev-2136-gb0150615d**
- Build System:    **idf.py**
- Compiler version:
    **xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0**
    **xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0**
- Operating System:     **Linux**
- Using an IDE?:   **No**
- Power Supply:   **USB** and **external 5V**

## Problem Description

The time of MQTT publish via `esp_mqtt_client_publish()` outside MQTT event handler varies between instant to a couple of seconds with weird repeating message in `idf.py monitor`. The repeating message is variation of 
```
I (7592040) phy: pll_cap_ext 10
```

### Expected Behavior

Instant or almost instant mqtt message publish without weird messages appearing on `idf.py monitor`.

### Steps to reproduce

1. Build, flash & monitor the program
2. Make interrupt on GPIO pin 17
3. Watch the idf.py monitor

### Code to reproduce this issue

[GIST](https://gist.github.com/adokitkat/9d4db77e046fd9489d34e376c173e034)

## Debug Logs

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�x�x��x����x����������x�x��ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x17f0
load:0x4004c000,len:0x9d8
load:0x40050000,len:0x2e08
entry 0x4004c1ec
I (46) boot: ESP-IDF v4.3-dev-2136-gb0150615d 2nd stage bootloader
I (46) boot: compile time 06:38:21
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1e874h (125044) map
I (141) esp_image: segment 1: paddr=0002e89c vaddr=3ffc8860 size=0177ch (  6012) load
I (143) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=88878h (559224) map
I (263) esp_image: segment 3: paddr=000b88a0 vaddr=3ffc9fdc size=02314h (  8980) load
I (266) esp_image: segment 4: paddr=000babbc vaddr=40024000 size=00404h (  1028) load
I (269) esp_image: segment 5: paddr=000bafc8 vaddr=40024404 size=14450h ( 83024) load
I (311) boot: Loaded app from partition at offset 0x10000
I (311) boot: Disabling RNG early entropy source...
I (323) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (323) cpu_start: Pro cpu up.
I (376) cpu_start: Pro cpu start user code
I (376) cpu_start: cpu freq: 160000000
I (377) cpu_start: Application information:
I (379) cpu_start: Project name:     imp
I (384) cpu_start: App version:      1
I (388) cpu_start: Compile time:     Dec 15 2020 06:38:13
I (394) cpu_start: ELF file SHA256:  ee539ffa585b5eea...
I (400) cpu_start: ESP-IDF:          v4.3-dev-2136-gb0150615d
I (407) heap_init: Initializing. RAM available for dynamic allocation:
I (414) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (420) heap_init: At 3FFD0058 len 0002BFA8 (175 KiB): DRAM
I (426) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (433) spi_flash: detected chip: generic
I (437) spi_flash: flash io: dio
W (441) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (458) cpu_start: Starting scheduler on PRO CPU.
I (460) IMP Motion sensor: [APP] Startup..
I (460) IMP Motion sensor: [APP] Free memory: 186212 bytes
I (470) IMP Motion sensor: [APP] IDF version: v4.3-dev-2136-gb0150615d
I (490) gpio: GPIO[17]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:1 
I (500) wifi:wifi driver task: 3ffd6fe0, prio:23, stack:6656, core=0
I (500) system_api: Base MAC address is not set
I (500) system_api: read default base MAC address from EFUSE
I (520) wifi:wifi firmware version: e1e57aa
I (520) wifi:wifi certification version: v7.0
I (520) wifi:config NVS flash: enabled
I (520) wifi:config nano formating: disabled
I (520) wifi:Init data frame dynamic rx buffer num: 32
I (530) wifi:Init management frame dynamic rx buffer num: 32
I (530) wifi:Init management short buffer num: 32
I (540) wifi:Init dynamic tx buffer num: 32
I (540) wifi:Init static rx buffer size: 1600
I (550) wifi:Init static rx buffer num: 10
I (550) wifi:Init dynamic rx buffer num: 32
I (550) wifi_init: rx ba win: 6
I (560) wifi_init: tcpip mbox: 32
I (560) wifi_init: udp mbox: 6
I (560) wifi_init: tcp mbox: 6
I (570) wifi_init: tcp tx win: 5744
I (570) wifi_init: tcp rx win: 5744
I (580) wifi_init: tcp mss: 1440
I (580) wifi_init: WiFi IRAM OP enabled
I (590) wifi_init: WiFi RX IRAM OP enabled
I (590) example_connect: Connecting to WIFI...
I (710) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
I (710) wifi:mode : sta (7c:df:a1:05:fc:d2)
I (710) wifi:enable tsf
I (710) example_connect: Waiting for IP(s)
I (840) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2180) wifi:state: init -> auth (b0)
I (2200) wifi:state: auth -> assoc (0)
I (2200) wifi:state: assoc -> run (10)
I (2210) wifi:connected with WIFI, aid = 3, channel 1, BW20, bssid = 18:e8:29:fb:0d:4e
I (2210) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (2210) wifi:pm start, type: 1

W (2220) wifi:<ba-add>idx:0 (ifx:0, 18:e8:29:fb:0d:4e), tid:0, ssn:0, winSize:64
I (2280) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (3190) esp_netif_handlers: example_connect: sta ip: 192.168.0.154, mask: 255.255.255.0, gw: 192.168.0.1
I (3190) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.0.154
I (3480) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:7edf:a1ff:fe05:fcd2, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (3480) example_connect: Connected to example_connect: sta
I (3490) example_connect: - IPv4 address: 192.168.0.154
I (3490) example_connect: - IPv6 address: fe80:0000:0000:0000:7edf:a1ff:fe05:fcd2, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (3510) Motion sensor: Other event id:7
I (4440) Motion sensor: MQTT_EVENT_CONNECTED
I (4440) Motion sensor: sent publish successful, msg_id=23729
I (4450) Motion sensor: sent subscribe successful, msg_id=21965
I (4470) Motion sensor: MQTT_EVENT_PUBLISHED, msg_id=23729
I (4510) Motion sensor: MQTT_EVENT_SUBSCRIBED, msg_id=21965
I (13510) Motion sensor: [GPIO] sent publish successful, msg_id=58815
GPIO[17] intr, val: 1
I (13510) phy: pll_cap_ext 10
I (13520) phy: pll_cap_ext 10
I (13530) phy: pll_cap_ext 10
I (13540) phy: pll_cap_ext 10
I (13550) phy: pll_cap_ext 10
I (13560) phy: pll_cap_ext 10
I (13560) phy: pll_cap_ext 10
I (13570) phy: pll_cap_ext 10
I (13570) phy: pll_cap_ext 10

... // many times repeats

I (15250) phy: pll_cap_ext 10
I (15250) phy: pll_cap_ext 10
I (15260) phy: pll_cap_ext 10
I (15260) phy: pll_cap_ext 10
I (15260) phy: pll_cap_ext 10
I (15320) Motion sensor: MQTT_EVENT_PUBLISHED, msg_id=58815
I (19510) Motion sensor: [GPIO] sent publish successful, msg_id=8655
```

Alvin1Zhang, 2020-12-15:
Thanks for the detailed report and letting us know, we will look into.

david-cermak, 2020-12-15:
@adokitkat This seems to be a duplicate of https://github.com/espressif/esp-idf/issues/6017, but were not able to reproduce the issue.
Does it happen also with any other board? From the log you've posted it looks like your log-level is set to the default *info*, correct? Could you please share your `sdkconfig`, too? Thanks.

adokitkat, 2020-12-15:
@david-cermak  
Unfortunately I don't have any other boards with GPIO pins.
I have updated the GIST file, it might have contained something wrong (I've edited the file beforehand) so now I have just pasted all of the program there. Now it should be fine, so please try again?
Log level may be DEBUG, it is set in sdkconfig.ci ...

[sdkconfig & sdkconfig.ci (zip)](https://github.com/espressif/esp-idf/files/5695599/sdkconfig_sdkconfigci.zip)

sinuswave, 2020-12-22:
I got the same problem on my custom board with ESP32-S2-WROVER, when using function adc1_get_raw(ADC_CHANNEL_4). However it has been running fine until I updated to the latest master branch b0150615dff529662772a60dcb57d5b559f480e2.

Muragavino, 2020-12-22:
Same as sinuswave, I have a custom board with ESP32-S2-Wrover-I, using ADC1 with function adc1_get_raw(ADC(1)_CHANNEL_3) while being connected via WiFi STA. My code reads the ADC, but green messages "phy: pll_cap_ext" start pouring out. My impression is the CPU has too much to handle and stalls, but without blocking entierly, - but I well might be wrong. 
The similar issue #6017 suggests the log verbosity being the problem, but I could not solve the issue by lowering the log verbosity below INFO (=3).
Would be nice to get a fix on this problem, since for me work in progress is blocked by it. Maybe the ESP32-S2 is not yet mature for new developments? For the moment I'll switch back to the good old ESP32-Wroover, where my firmware runs error free.

colorflux, 2020-12-22:
I came across the same issue using the same development board and chipset as @adokitkat  (ESP32-S2 Saola-1 and ESP32-S2-WROVER) while attempting to send some voltage readings over MQTT/TCP using the master branch of ESP-IDF.

There were thousands of lines of  `phy: pll_cap_ext 10` pouring onto the monitor which quickly resulted in packet loss and the board disconnecting from it's configured network.

I was able to resolve this issue by rolling back to:
```
idf.py --version
ESP-IDF v4.2-47-g2532ddd9f-dirty
```

kerwincui, 2021-01-12:
Same as sinuswave,There is no problem using ADC1 to collect data directly. If WiFi is used at the same time, a lot of "phy:pll_cap_ext" will be printed through the serial port.Problem code：
`adc_reading += adc1_get_raw((adc1_channel_t)ADC_CHANNEL_0);`
1. version:  ESP-IDF v4.2-47-g2532ddd9f
2. borad:  ESP32-S2-Wrover
3. sdkconfig: CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y; CONFIG_BOOTLOADER_LOG_LEVEL=3

Has anyone solved it？

adokitkat, 2021-02-27:
With the newest IDF version (ESP-IDF v4.4-dev-4-g73db14240) and ESP32S2-Saola-1 it stopped working completely, on interrupt it just reboots and doesn't even send the MQTT message in the end - so the problem got worse.

My program works perfectly fine on ESP32 DevkitC v4.

Log (rebooting again and again):
```
ELF file SHA256: d85ab76f5ef57a0e

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40024060
0x40024060: esp_restart_noos at /home/ado/esp/esp-idf/components/esp32s2/system_api_esp32s2.c:109 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17f4
load:0x4004c000,len:0x9d8
load:0x40050000,len:0x2ebc
entry 0x4004c1ec
I (46) boot: ESP-IDF v4.4-dev-4-g73db14240 2nd stage bootloader
I (46) boot: compile time 21:41:53
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1eaech (125676) map
I (139) esp_image: segment 1: paddr=0002eb14 vaddr=3ffc65b0 size=01504h (  5380) load
I (140) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=8b5b0h (570800) map
I (253) esp_image: segment 3: paddr=000bb5d8 vaddr=3ffc7ab4 size=02804h ( 10244) load
I (255) esp_image: segment 4: paddr=000bdde4 vaddr=40022000 size=145a8h ( 83368) load
I (290) boot: Loaded app from partition at offset 0x10000
I (290) boot: Disabling RNG early entropy source...
I (301) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (301) cpu_start: Pro cpu up.
I (355) cpu_start: Pro cpu start user code
I (355) cpu_start: cpu freq: 160000000
I (355) cpu_start: Application information:
I (358) cpu_start: Project name:     imp
I (363) cpu_start: App version:      1
I (367) cpu_start: Compile time:     Feb 27 2021 21:41:46
I (373) cpu_start: ELF file SHA256:  d85ab76f5ef57a0e...
I (379) cpu_start: ESP-IDF:          v4.4-dev-4-g73db14240
I (386) heap_init: Initializing. RAM available for dynamic allocation:
I (393) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (399) heap_init: At 3FFCEB48 len 0002D4B8 (181 KiB): DRAM
I (405) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (412) spi_flash: detected chip: generic
I (416) spi_flash: flash io: dio
W (420) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (437) cpu_start: Starting scheduler on PRO CPU.
I (439) IMP Motion sensor: [APP] Startup..
I (439) IMP Motion sensor: [APP] Free memory: 187600 bytes
I (449) IMP Motion sensor: [APP] IDF version: v4.4-dev-4-g73db14240
I (459) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:1 
I (479) wifi:wifi driver task: 3ffd6c18, prio:23, stack:6656, core=0
I (479) system_api: Base MAC address is not set
I (479) system_api: read default base MAC address from EFUSE
I (489) wifi:wifi firmware version: 6b2834e
I (489) wifi:wifi certification version: v7.0
I (489) wifi:config NVS flash: enabled
I (499) wifi:config nano formating: disabled
[... 78 lines cut ...]
0x400ea673: set_channel_rfpll_freq at ??:?

0x400ea6e4: chip_v7_set_chan at ??:?

0x400eb6ad: noise_init at ??:?

0x400ec17a: bb_init at ??:?

0x400ec22b: register_chipv7_phy at ??:?

0x40093fc0: esp_phy_load_cal_and_init at /home/ado/esp/esp-idf/components/esp_wifi/src/phy_init.c:596

0x40093fe9: esp_phy_enable at /home/ado/esp/esp-idf/components/esp_wifi/src/phy_init.c:215

0x400d3737: wifi_hw_start at ??:?

0x400d3a46: wifi_start_process at ??:?

0x400d0b3d: ieee80211_ioctl_process at ??:?

0x400329e0: ppTask at ??:?

0x4002d0c9: vPortTaskWrapper at /home/ado/esp/esp-idf/components/freertos/port/xtensa/port.c:168

```

zhangyanjiaoesp, 2021-03-01:
@Muragavino  @ghost @kerwincui Please update your IDF version to test again. According to @sinuswave  and  #6017, they have solved their problem when they updated their IDF version.

zhangyanjiaoesp, 2021-03-01:
@adokitkat We will try to reproduce your problem with your example code and ESP-IDF v4.4-dev-4-g73db14240.

Did your config is still same with `sdkconfig & sdkconfig.ci (zip)` ?

leeebo, 2021-03-01:
@zhangyanjiaoesp The same error message when I try to build [esp-drone](https://github.com/espressif/esp-drone) project with the current master branch, no MQTT used there, only Wi-Fi AP with UDP socket.

```
target: esp32s2
esp-idf version: ESP-IDF v4.4-dev-4-g73db14240-dirty

abort() was called at PC 0x4002450b on core 0
0x4002450b: lock_acquire_generic at /home/libo/esp/esp-idf/components/newlib/locks.c:138

Backtrace:0x4002b236:0x3ffe4d60 0x4002baed:0x3ffe4d80 0x400330fe:0x3ffe4da0 0x4002450b:0x3ffe4e10 0x40024639:0x3ffe4e40 0x400fa33e:0x3ffe4e60 0x400fd39d:0x3ffe5170 0x40106343:0x3ffe51a0 0x40031b96:0x3ffe51d0 0x400ec2d1:0x3ffe5220 0x400ec301:0x3ffe52a0 0x40027137:0x3ffe52f0 0x40027211:0x3ffe5350 0x40028c78:0x3ffe5370 0x400d88bf:0x3ffe5390 0x40034ad5:0x3ffe53d0
0x4002b236: panic_abort at /home/libo/esp/esp-idf/components/esp_system/panic.c:367

0x4002baed: esp_system_abort at /home/libo/esp/esp-idf/components/esp_system/system_api.c:112

0x400330fe: abort at /home/libo/esp/esp-idf/components/newlib/abort.c:46

0x4002450b: lock_acquire_generic at /home/libo/esp/esp-idf/components/newlib/locks.c:138

0x40024639: _lock_acquire_recursive at /home/libo/esp/esp-idf/components/newlib/locks.c:166

0x400fa33e: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x400fd39d: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x40106343: esp_log_writev at /home/libo/esp/esp-idf/components/log/log.c:189

0x40031b96: esp_log_write at /home/libo/esp/esp-idf/components/log/log.c:199

0x400ec2d1: lib_printf at /home/libo/esp/esp-idf/components/esp_wifi/src/lib_printf.c:47 (discriminator 13)

0x400ec301: phy_printf at /home/libo/esp/esp-idf/components/esp_wifi/src/lib_printf.c:57

0x40027137: pll_vol_cal at ??:?

0x40027211: wifi_track_pll_cap at ??:?

0x40028c78: tx_pwctrl_background at ??:?

0x400d88bf: lmacProcessTxComplete at ??:?

0x40034ad5: ppTask at ??:?

ELF file SHA256: 7a435284893679c9

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40023f32
0x40023f32: esp_restart_noos at /home/libo/esp/esp-idf/components/esp32s2/system_api_esp32s2.c:109 (discriminator 1)
```

adokitkat, 2021-03-01:
> @adokitkat We will try to reproduce your problem with your example code and ESP-IDF v4.4-dev-4-g73db14240.
> 
> Did your config is still same with `sdkconfig & sdkconfig.ci (zip)` ?

Yes it should be the same.

rskaliotis, 2021-03-22:
I'm having this problem on v4.2 and I've tracked it down to [e7435a7](https://github.com/espressif/esp-idf/commit/2532ddd9f447f6fab02bc2d1654534a7621e033a). At this commit I see the "pll_cap_ext" messages and have wifi issues, immediately before at 10d40bd everything works normally.

zhangyanjiaoesp, 2021-03-23:
@adokitkat @rskaliotis The root cause has been found for the crash issue, will merge it as soon as possible.

closed 2021-04-01 by espressif-bot

PIYUSHGIRI2021, 2021-06-07:
hello @Alvin1Zhang and every one ...i compile pppos_client example in esp idf before some day i copiled it will work fine but not its not work,please let me know how to resolve below error..

I (24525) pppos_example: Modem PPP Started
C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x40088345 on core 0
0x40088345: xQueueGenericReceive at C:/Users/Frank/Desktop/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

Backtrace:0x40084f7f:0x3ffb7ab0 0x4008561d:0x3ffb7ad0 0x4008928e:0x3ffb7af0 0x40088345:0x3ffb7b60 0x400f04a5:0x3ffb7ba0 0x400f07bc:0x3ffb7bc0 0x400deb80:0x3ffb7be0 0x400dec0c:0x3ffb7c00 0x400df989:0x3ffb7c20 0x400f4c5c:0x3ffb7c50 0x400f4e9c:0x3ffb7c80 0x400f55c5:0x3ffb7ca0 0x400f34f2:0x3ffb7cc0 0x400f3569:0x3ffb7ce0 0x400f3575:0x3ffb7d00 0x400f36a5:0x3ffb7d20 0x400eed19:0x3ffb7d40 0x400eed4e:0x3ffb7d60 0x400ee0a1:0x3ffb7d80 0x400eea08:0x3ffb7db0 0x400eeb5d:0x3ffb7dd0 0x400dd2d1:0x3ffb7df0 0x400dd350:0x3ffb7e10 0x40085625:0x3ffb7e40
0x40084f7f: panic_abort at C:/Users/Frank/Desktop/esp-idf/components/esp_system/panic.c:330

0x4008561d: esp_system_abort at C:/Users/Frank/Desktop/esp-idf/components/esp_system/system_api.c:106

0x4008928e: abort at C:/Users/Frank/Desktop/esp-idf/components/newlib/abort.c:46

AxelLin, 2021-06-07:
@PIYUSHGIRI2021 
You should create a new issue.

yshivam909, 2022-07-24:
> @adokitkat @rskaliotis The root cause has been found for the crash issue, will merge it as soon as possible.

Is this issue solved ? . Kindly let us know the updated version .