espressif/esp-idf #6272: Not receiving correct ir code using esp-idf\examples\peripherals\rmt\ir_protocols code

DipaliDh, 2020-12-15:
Hi,
      I am new to esp-idf and esp32-s2. I am using esp-idf\examples\peripherals\rmt\ir_protocols.  I am trying to receive an IR code using the NEC protocol on pin 37.  I have commented tx functions as I only wanted to receive from remote. I collected all values and then observed its MSB and LSB. I compared it with the reference remote code array.

For example, if button 1 is pressed, I received, 0x2859. While comparing it with a remote array, I found that if I inverse this value, 4-bit MSB is matching in each 1 byte and LSB is less or more by 1 in both bytes. This happens for random button values in the remote array. i.e. 0x2895(button 1 pressed) --> 0x5982(inverse), remote code: 0x5a81(this is for button 6 in remote array)

I have also got 1 value as it is but the button number pressed is not matching. 
Please guide me if there are any settings or configuration I am missing.

felmue, 2020-12-15:
Hello @DipaliDh 

from what I can tell by default RX pin is GPIO19. Did you use `menuconfig` to switch RX to pin GPIO37?

When you say you do not receive the correct IR code. What is the IR code you receive and what IR code would you expect?

Thanks
Felix

DipaliDh, 2020-12-15:
@Felix Thanks for your reply...

Yes, I have configured RX pin for GPIO37 and also selected nec protocol in menuconfig.... 

The code I want to receive when Button1 is pressed is, 0xCFD12E and what I received on RX pin is, 0x748B

felmue, 2020-12-15:
Hello @Dipali

your setup is fine. The output is just in a different format. Got me too a long time ago until a good friend pointed out to me that the received data actually is what I was looking for. Here is how it goes:
```
received:   7    4    8    B
binary:  0111 0100 1000 1011
1 <=> 0: 1000 1011 0111 0100
flipped: 0001 1101 1110 0010
hex:        1    D    E    2
swapped:    D    1    2    E
```
I guess the parser only allows for 16 bits therefor the 0xCF at the beginning is lost.

Cheers
Felix

suda-morris, 2020-12-16:
According to @felmue 's nice analysis, maybe you can try to enable the `IR_TOOLS_FLAGS_INVERSE` flag in ir_parser_config

DipaliDh, 2020-12-16:
@felmue it is the nice explaination...thanks for your efforts & @suda-morris thanks....i will try by setting this flag

felmue, 2020-12-16:
Hi guys

@suda-morris: interesting, I didn't notice that flag.

@DipaliDh: I am curious what you'll find. When I set this flag my remote isn't recognised anymore at all.

My guess is that the flag is meant for electrically inverted inputs.

Thanks
Felix

DipaliDh, 2020-12-16:
Hi,
@suda-morris & @felmue, I set this flag. Now, the code is not getting parsed anymore... 

Thanks
Dipali

suda-morris, 2020-12-16:
@felmue Ah sorry, yes, this flag is for electronically inverted signals. Also please note, IR protocols have a lot of variations, our example can't parse all of them. We suggest our customer to get a spec of your remote controller then added the parser for it.

felmue, 2020-12-16:
Hello @suda-morris 

no worries. Oh yes, I know, there are many variations in IR protocols. A long time ago I've developed an IR Blaster plugin for a music streaming device called 'Squeezebox'. I've learned and adopted quite a bit from the [lirc](https://www.lirc.org) project.

Cheers
Felix

closed 2021-08-05 by Alvin1Zhang