espressif/esp-idf #6273: Error cause: wifi:token mismatch, expect=3

ZweiEuro, 2020-12-15:
## Environment

- Development Kit:     ESP32 Devkit V1 -Doit (I think, it's soldered i can't look at it anymore) [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html#esp32-wroom-32)
- Kit version -
- Module or chip used:  ESP32-WROOM-32

- IDF version (run ``git describe --tags`` to find it): framework-espidf 3.40001.200521 (4.0.1)  (PIO)
- Build System:         platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:   Linux

- Using an IDE?:  Vscode platformio
- Power Supply:         USB

# Poblem:

What does
```
W (52008) wifi:token mismatch, expect=3
W (52008) wifi:token mismatch, expect=3
W (59308) wifi:state wrong txa_flags=9
```
mean, where does it come from, and how can I debug this? 
I have the weird issue that i can't connect to the esp (running an example tcp server) from my phone (simple tcp socket app) (in the same wifi as the esp)
but i can connect from my computer (lan) on the same router and a simple python script, which is also the one from your example.
the connection established by the python script also takes much longer to establish than a usual tcp connection.

YouDONG-ESP, 2020-12-17:
@ZweiEuro  This log shows that esp try to initialize a block Ack mechanism and the adddba response frame recieved by esp is not correct. It's seems like a problem from the sender.

YouDONG-ESP, 2020-12-17:
You can do a packet capture to verify it

ZweiEuro, 2020-12-22:
How would I do that? I'd need to relay all traffic from the router /hub to my laptop in order to wireshark that, am i correct ?

negativekelvin, 2020-12-22:
> You could follow this guide:
> The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
> https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
> Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

YouDONG-ESP, 2020-12-23:
@ZweiEuro Could you provide your sender's model?

ZweiEuro, 2020-12-23:
@YouDONG-ESP Model of the Router ? 
A1 Hybrid box
[image]

YouDONG-ESP, 2020-12-28:
@ZweiEuro Dose this issue only happen with this router? If yes, we will buy one to check the root cause.

ZweiEuro, 2020-12-29:
It doesn't seem to happen on my mobile phone hotspot (with which I've been developing thus far since i can't use my router)
Galaxy S7 (jailbroken) bit i.g. the hotspot feature wasn't touched on from my break (i just got rid of bloatware).

I've also tried it with an IPhone from a friend and that worked fine as well. As i recall it was one of the newer versions.
(To be exact: <asked him, waiting for answer>)

YouDONG-ESP, 2021-01-21:
@ZweiEuro  We can't get this router, considering this issue won't affect the use and it's not seems to be our bug, we will probably close it. 
Could you provide a wifi packet capture to let us verify? 

The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

ZweiEuro, 2021-01-29:
Okay I will try to provide this data in order to debug.

I would appreciate if (even if it's the routers fault) there can be some fix created to still make the chip function properly
@YouDONG-ESP

ZweiEuro, 2021-01-31:
I just fried my esp32, i will comment as soon as i have a new one and can do the test

ZweiEuro, 2021-03-08:
The error was caused by the routers sim card internal firewall.
When i called my ISP to turn it off it worked. They couldn't tell me why either

closed 2021-03-08 by ZweiEuro