espressif/esp-idf #6274: esp-as-lib with CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=y doesn't build

meros, 2020-12-15:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version WroverKit: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
v4.3-dev-2136-gb0150615d
- Build System:        cmake (idf-as-lib)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Using idf-as-lib build method, enabling CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=y causes an undefined reference to `esp_gdbstub_panic_handler' during linkage of esp_system

### Expected Behavior

esp_system should link with esp_gdbstub when this flag is set (manually updating esp_system to include this dep 'fixes' the problem)

### Actual Behavior

esp_system doesn't link with esp_gdbstub causing undefined reference

### Steps to reproduce

1. step1
clone esp-idf
2. build examples/build_system/cmake/idf_as_lib according to README.md (this should work)
3. change ESP System settings to use CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=y
4. build again, this will fail with the undefined reference described above

## Debug Logs

```
l  -u pthread_include_pthread_cond_impl  -u pthread_include_pthread_local_storage_impl  -u vfs_include_syscalls_impl  -L /Users/meros/git/above/ahhbe-fw-esp32/external/esp-idf/components/esp_wifi/lib/esp32 && :
/Volumes/git/above/ahhbe-fw-esp32/external/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_system/libesp_system.a(panic.c.obj):(.literal.esp_panic_handler+0x30): undefined reference to `esp_gdbstub_panic_handler'
/Volumes/git/above/ahhbe-fw-esp32/external/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_system/libesp_system.a(panic.c.obj): in function `esp_panic_handler':
/Users/meros/git/above/ahhbe-fw-esp32/external/esp-idf/components/esp_system/panic.c:287: undefined reference to `esp_gdbstub_panic_handler'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5695589/sdkconfig.zip)

Alvin1Zhang, 2020-12-16:
Thanks for the report and letting us know, we will look into.

renzbagaporo, 2020-12-17:
Hi @meros, incidentally, there is a pending MR that solves this. This should be in the public repo in the coming days.

closed 2021-01-05 by espressif-bot