espressif/esp-idf #6276: espnow example cannot set channel 0

nopnop2002, 2020-12-16:
## Environment
```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.3-dev-1901-g178b122c1-dirty
chip model is 1, chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
package is ESP32D0WDQ5
```

## Problem Description
Call esp_now_add_peer() to add the device to the paired device list before you send data to this device.    
The range of the channel of paired devices is from 0 to 14. If the channel is set to 0, data will be sent on the current channel.   
But current channel range is 1 to 13.   

```
    config ESPNOW_CHANNEL
        int "Channel"
        default 1
        range 1 13
        help
            The channel on which sending and receiving ESPNOW data.
```

```
    memset(peer, 0, sizeof(esp_now_peer_info_t));
    peer->channel = CONFIG_ESPNOW_CHANNEL;
    peer->ifidx = ESPNOW_WIFI_IF;
    peer->encrypt = false;
    memcpy(peer->peer_addr, s_example_broadcast_mac, ESP_NOW_ETH_ALEN);
    ESP_ERROR_CHECK( esp_now_add_peer(peer) );

```

Alvin1Zhang, 2020-12-16:
Thanks for reporting, we will look into.

YouDONG-ESP, 2020-12-17:
@nopnop2002 Thanks for reporting, a repaire patch will be released.

closed 2021-01-05 by espressif-bot