espressif/esp-idf #6277: Correct mode switching (AP, STA, NULL)

Oleg-Perevyshin, 2020-12-16:
Hello everyone!
I want to learn how to switch modes correctly.
The final goal is STA mode with automatic maintenance of connection and reconnection.
But if the password is incorrect or there is no Access Point, then the ESP should switch to APSTA mode and periodically search for the installed network.

I split the problem into parts.
But the switch doesn't happen.
The documentation describes everything in a very complicated and confusing way.
Help me figure out how to do it right.

```
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

esp_err_t wifi_init() {
  if(esp_netif_init() != ESP_OK) {
    return ESP_FAIL;
  }
  if (initSTA() == ESP_OK) {
    return ESP_OK;
  } else if (initAP() == ESP_OK) {
    return ESP_OK;
  } else {
    return ESP_FAIL;
  }
}
```
```
/* Event STA Handler */
static void eventSTAHandler(void * arg, esp_event_base_t event_base, int32_t event_id, void * event_data) {
  static uint8_t retryNum = 0;
  if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
    esp_wifi_connect();
  } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
    if (retryNum < 5) {
      esp_wifi_connect();
      retryNum++;
    } else {
      xEventGroupSetBits(wifiEventGroup, WIFI_FAIL_BIT);
    }
  } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
    ip_event_got_ip_t * event = (ip_event_got_ip_t *)event_data;
    retryNum = 0;
    xEventGroupSetBits(wifiEventGroup, WIFI_CONNECTED_BIT);
  }
}

// Init STA Mode
esp_err_t initSTA() {

	esp_wifi_stop();
	esp_wifi_deinit();

  wifiEventGroup = xEventGroupCreate();

  if(esp_event_loop_create_default() != ESP_OK) {
    return ESP_FAIL;
  }
  esp_netif_create_default_wifi_sta();

  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  if(esp_wifi_init(&cfg) != ESP_OK) {
    return ESP_FAIL;
  }

  esp_event_handler_instance_t instance_any_id;
  if(esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &eventSTAHandler, NULL, &instance_any_id) != ESP_OK) {
    return ESP_FAIL;
  }
  esp_event_handler_instance_t instance_got_ip;
  if(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &eventSTAHandler, NULL, &instance_got_ip) != ESP_OK) {
    return ESP_FAIL;
  }

  wifi_config_t wifiSTAconfig = {
    .sta = {
      .ssid = CONFIG_WIFI_SSID_STA,
      .password = CONFIG_WIFI_PASSWORD_STA,
      .pmf_cfg = {
        .capable = true,
        .required = false
      },
    },
  };

  if(esp_wifi_set_mode(WIFI_MODE_STA) != ESP_OK) {
    return ESP_FAIL;
  }
  if(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifiSTAconfig) != ESP_OK) {
    return esp_wifi_set_config(ESP_IF_WIFI_STA, &wifiSTAconfig);
  }
  if(esp_wifi_start() != ESP_OK) {
    return esp_wifi_start();
  }

  EventBits_t bits = xEventGroupWaitBits(wifiEventGroup, WIFI_CONNECTED_BIT | WIFI_FAIL_BIT, pdFALSE, pdFALSE, portMAX_DELAY);

  if(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id) != ESP_OK) {
    return ESP_FAIL;
  }
  if(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip) != ESP_OK) {
    return ESP_FAIL;
  }

  vEventGroupDelete(wifiEventGroup);

  if (bits & WIFI_CONNECTED_BIT) {
    return ESP_OK;
  } else if (bits & WIFI_FAIL_BIT) {
    esp_wifi_stop();
    return ESP_FAIL;
  } else {
    esp_wifi_stop();
    return ESP_FAIL;
  }
}
```
```
/* Event AP Handler */
static void eventAPHandler(void * arg, esp_event_base_t event_base, int32_t event_id, void * event_data) {
  if (event_id == WIFI_EVENT_AP_STACONNECTED) {
    wifi_event_ap_staconnected_t * event = (wifi_event_ap_staconnected_t*) event_data;
    char buff[30];
    snprintf(buff, sizeof(buff), "%02X:%02X:%02X:%02X:%02X:%02X", MAC2STR(event->mac));
    sendMCUchar("SYS", "StationJoin", buff);
  } else if (event_id == WIFI_EVENT_AP_STADISCONNECTED) {
    wifi_event_ap_stadisconnected_t * event = (wifi_event_ap_stadisconnected_t *) event_data;
    char buff[30];
    snprintf(buff, sizeof(buff), "%02X:%02X:%02X:%02X:%02X:%02X", MAC2STR(event->mac));
    sendMCUchar("SYS", "StationLeave", buff);
  }
}

// Init AP Mode
esp_err_t initAP() {

	esp_wifi_stop();
	esp_wifi_deinit();

  if(esp_event_loop_create_default() != ESP_OK) {
    return ESP_FAIL;
  }
  esp_netif_create_default_wifi_ap();

  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  if(esp_wifi_init(&cfg) != ESP_OK) {
    return ESP_FAIL;
  }

  if(esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &eventAPHandler, NULL, NULL) != ESP_OK) {
    return ESP_FAIL;
  }

  wifi_config_t wifiAPConfig = {
    .ap = {
      .ssid = CONFIG_WIFI_SSID_AP,
      .ssid_len = strlen(CONFIG_WIFI_SSID_AP),
      .channel = CONFIG_WIFI_CHANNEL_AP,
      .password = CONFIG_WIFI_PASSWORD_AP,
      .max_connection = 4,
      .authmode = WIFI_AUTH_WPA_WPA2_PSK
    },
  };
  if (strlen(CONFIG_WIFI_PASSWORD_AP) == 0) {
      wifiAPConfig.ap.authmode = WIFI_AUTH_OPEN;
  }

  if(esp_wifi_set_mode(WIFI_MODE_AP) != ESP_OK) {
    return ESP_FAIL;
  }
  if(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifiAPConfig) != ESP_OK) {
    return esp_wifi_set_config(ESP_IF_WIFI_AP, &wifiAPConfig);
  }
  if(esp_wifi_start() != ESP_OK) {
    return esp_wifi_start();
  }

  return ESP_OK;
}
```

YouDONG-ESP, 2020-12-17:
@Oleg-Perevyshin You don't have to deinit or stop wifi to switch mode, this code will give you an idea:
```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

static const char *TAG = "wifi ";
static esp_netif_t *netif_ap = NULL;
static esp_netif_t *netif_sta = NULL;
static EventGroupHandle_t wifi_event_group;

static void wifi_ap_mode(void)
{
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "wifi1234",
            .ssid_len = 0,
            .max_connection = 4,
            .password = "12345678",
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
}

static void wifi_sta_mode(void)
{
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "wifi111",
            .password = ""
        }
     };

    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_connect() );
}

static void sta_connect_handler(void* arg, esp_event_base_t event_base,
                               int32_t event_id, void* event_data)
{
    printf("sta connected, switch to ap mode\n");
    vTaskDelay(5000/portTICK_PERIOD_MS);
    wifi_ap_mode();
}

static void ap_connect_handler(void* arg, esp_event_base_t event_base,
                               int32_t event_id, void* event_data)
{
    printf("ap connected, switch to sta mode\n");
    vTaskDelay(5000/portTICK_PERIOD_MS);
    wifi_sta_mode();
}

void initialise_wifi(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_create_default() );
    netif_ap = esp_netif_create_default_wifi_ap();
    assert(netif_ap);
    netif_sta = esp_netif_create_default_wifi_sta();
    assert(netif_sta);
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        WIFI_EVENT_AP_STACONNECTED,
                                                        &ap_connect_handler,
                                                        NULL,
                                                        NULL));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        WIFI_EVENT_STA_CONNECTED,
                                                        &sta_connect_handler,
                                                        NULL,
                                                        NULL));                                                        

    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    wifi_ap_mode();
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    initialise_wifi();
}
```

Oleg-Perevyshin, 2020-12-17:
Thanks for the example!
There would be more such examples that are used in real problems.

closed 2020-12-17 by Oleg-Perevyshin