espressif/esp-idf #6279: Heap memory is reduced (40 bytes) every time malloc(), free() is repeated.

ansee7221, 2020-12-16:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    // 2.2.0-28-g9fc5ba6
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
Memory leakage occurs when data is stored periodically in the NVS area.
After malloc(), save to NVS area, free() then heap memory reduced approximately 40 bytes.
  
//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <pthread.h>
#include <stdio.h>
#include <inttypes.h>
#include <signal.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "nvs_flash.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
//#include "driver/gpio.h"
//#include "debug_print.h"
#include "esp_system.h"
#include "port/oc_clock.h"

#define EXAMPLE_WIFI_SSID CONFIG_WIFI_SSID
#define EXAMPLE_WIFI_PASS CONFIG_WIFI_PASSWORD
//#define BLINK_GPIO CONFIG_BLINK_GPIO

static EventGroupHandle_t wifi_event_group;

static const int IPV4_CONNECTED_BIT = BIT0;
static const int IPV6_CONNECTED_BIT = BIT1;

static const char *TAG = "iotivity server";

#define SVR_TAG_MAX (32)
#define STORE_PATH_SIZE 64
#define MAX_APP_DATA_SIZE (2048L)		

static char storePath[STORE_PATH_SIZE];
static int storePath_len;
static bool pathSet = false;
[... 355 lines cut ...]

testCount:2
storage_write(), ret=0, 12345678901234567890
**free heap size:150196**
minimum free heap size:136388

testCount:3
storage_write(), ret=0, 12345678901234567890
**free heap size:150160**
minimum free heap size:136388

testCount:4
storage_write(), ret=0, 12345678901234567890
**free heap size:150120**
minimum free heap size:136388
```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-12-16:
Thanks for the detailed report and letting us know, we will look into.

igrr, 2020-12-16:
@ansee7221 Could you check `storage_write` function, it seems it calls nvs_open every time, but doesn't call nvs_close in case of success.
For debugging heap memory leaks, please see the instructions in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-tracing.

0xjakob, 2020-12-16:
Hi @ansee7221,
Thanks for trying out current code!

Could you try to turn on [stand alone heap tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html?highlight=memory#heap-tracing) in menuconfig?

There should be some configuration options regarding stack depth, etc., if I remember correctly. You'll get trace output which provide valuable information where (de-)allocations occur. Let us know how it goes.

Thanks,
Jakob

boarchuz, 2020-12-16:
Your storage_write is missing nvs_close

ansee7221, 2020-12-17:
@boarchuz, Thanks.
I missed nvs_close.

closed 2020-12-17 by ansee7221