espressif/esp-idf #6281: How to add .cpp library files to esp idf

GunashekarG, 2020-12-16:
OS: Linux(Ubuntu Mate)
ESP-IDF Version: v4.2
Chip: ESP WROOM 32

I am trying to use Arduino host shield with ESP32 in order to convert USB to SPI. I have tested the working with Arduino UNO and it's working perfectly fine. Now, I want to add the library files used in Arduino in order to initiate the USB port and conversion of data. while I try to add them, I am getting the following error.

[image]

Someone please help me fixing this issue. Thanks in advance.

Alvin1Zhang, 2020-12-18:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

Kiznoh, 2020-12-23:
Hi GunashekarG,
using the arduino library [USB_Host_Shield_2.0](https://github.com/felis/USB_Host_Shield_2.0) with ESP-IDF is not supported. If you want to use this library with the ESP32, you'll need to use [arduino-esp32](https://github.com/espressif/arduino-esp32/) with the Arduino IDE.

GunashekarG, 2020-12-23:
@Kiznoh Thanks for the reply. Is there any other USB host module that I can work with ESP WROOM 32 using ESP-IDF?

Kiznoh, 2020-12-23:
It may be possible to use the arduino library by using arduino-esp32 as a component of ESP-IDF by following this procedure: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
However, you may encounter some weird issues and may not get support for them.

Abhay-2412, 2023-11-07:
hello im trying to port a wifi client library from Arduino framework to esp idf framework i want to use functions like `client.connect()` or` client.write()` since iam not able find APIs like this in esp idf framework i dont want to add arduino as component since its consumes more memory and i dont want all libraries i just want to use Wificlient.cpp functions only so help me out