espressif/esp-idf #6282: wifi coredump with AP suddenly restarting

sebirdman, 2020-12-16:
## Environment

- IDF version: 4.0.2
- Build System:  Cmake
- Compiler version: esp-2020r3

## Problem Description

While the wifi AP was down, one of my devices crashed in the wifi task causing a coredump

### Expected Behavior

not crashing

### Actual Behavior

crashing

### Steps to reproduce

Exact steps unknown, but wifi AP was power cycled during this crash period.

## Debug Logs

```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffd3d44, name: 'wifi', GDB name: 'process 1073560900'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x10
epc7           0x3ffd1ca0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x3ffd3b70	0x3ffd3b70
lbeg           0x40084948	1074284872
lend           0x0	0
lcount         0x801a7465	2149217381
sar            0x1	1
ps             0x2	2
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40084a74	1074285172
a1             0x0	0
[... 305 lines cut ...]
.coredump.tasks.data 0x3ffd1890 0x248 RW 
.coredump.tasks.data 0x3ffdfc60 0x170 RW 
.coredump.tasks.data 0x3ffcfb60 0x22c RW 
.coredump.tasks.data 0x3ffb74cc 0x170 RW 
.coredump.tasks.data 0x3ffb72e0 0x1e4 RW 
.coredump.tasks.data 0x3ffd94e4 0x170 RW 
.coredump.tasks.data 0x3ffd9260 0x27c RW 
.coredump.rom.text 0x40000000 0x170 R X
.coredump.rom.text 0x40000180 0x8 R X
.coredump.rom.text 0x400001c0 0x8 R X
.coredump.rom.text 0x40000200 0x8 R X
.coredump.rom.text 0x40000240 0x8 R X
.coredump.rom.text 0x40000280 0xc R X
.coredump.rom.text 0x400002c0 0x4 R X
.coredump.rom.text 0x40000300 0x8 R X
.coredump.rom.text 0x40000340 0x18 R X
.coredump.rom.text 0x400003c0 0x8 R X
.coredump.rom.text 0x40000400 0x160 R X
.coredump.rom.text 0x40000560 0xd10c R X
.coredump.rom.text 0x40010000 0x54ef4 R X

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

Alvin1Zhang, 2020-12-17:
Thanks for reporting and providing the logs, we will look into.

HarveyRong-Esp, 2020-12-17:
Hi @sebirdman, can you provide the minimal reproduction code and elf file?

sebirdman, 2020-12-17:
@HarveyRong-Esp Unfortunately I cannot share the code and i'm not sure what the minimal repro would be.

sebirdman, 2020-12-19:
I'm going to close this as i think i had some memory corruption. will reopen if i get this again.

closed 2020-12-19 by sebirdman

HarveyRong-Esp, 2020-12-21:
@sebirdman, yeah..., `assert(res == coreID || res == portMUX_FREE_VAL)`should be caused by memory corruption