espressif/esp-idf #6283: LWIP sys_sem_signal semaphore locking on NULL semaphore

sebirdman, 2020-12-16:
## Environment

- IDF version: 4.0.2
- Build System:  Cmake
- Compiler version: esp-2020r3

## Problem Description

Over the past few days, i've many of my devices in the field have crashed from attempting to lock on (what appears to be) a NULL semaphore

Am i reading this right?:

```
#2  0x40096dd4 in xQueueGenericSend (xQueue=0x3f80b0c4, pvItemToQueue=0x0, xTicksToWait=<optimized out>, xCopyPosition=0) at ../dependencies/espressif/esp-idf/components/freertos/queue.c:720
```

where pvItemToQueue=0x0 means we're trying to queue a NULL pointer? Why would this happen?

### Expected Behavior

not crashing

### Actual Behavior

crashing

### Steps to reproduce

Unknown, just seeing this on some devices in the field 

## Debug Logs

```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbcdb8, name: 'tiT', GDB name: 'process 1073466808'

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x1d
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x400945bf	0x400945bf <invoke_abort+27>
lbeg           0x40091854	1074337876
lend           0x40091870	1074337904
lcount         0xffffffff	4294967295
sar            0x5	5
ps             0x60020	393248
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
[... 313 lines cut ...]
.coredump.tasks.data 0x3ffb5630 0x1f4 RW 
.coredump.tasks.data 0x3ffb74cc 0x170 RW 
.coredump.tasks.data 0x3ffb72e0 0x1e4 RW 
.coredump.tasks.data 0x3ffd3e70 0x170 RW 
.coredump.tasks.data 0x3ffd3c70 0x1f8 RW 
.coredump.tasks.data 0x3ffd8c14 0x170 RW 
.coredump.tasks.data 0x3ffd8990 0x27c RW 
.coredump.rom.text 0x40000000 0x170 R X
.coredump.rom.text 0x40000180 0x8 R X
.coredump.rom.text 0x400001c0 0x8 R X
.coredump.rom.text 0x40000200 0x8 R X
.coredump.rom.text 0x40000240 0x8 R X
.coredump.rom.text 0x40000280 0xc R X
.coredump.rom.text 0x400002c0 0x4 R X
.coredump.rom.text 0x40000300 0x8 R X
.coredump.rom.text 0x40000340 0x18 R X
.coredump.rom.text 0x400003c0 0x8 R X
.coredump.rom.text 0x40000400 0x160 R X
.coredump.rom.text 0x40000560 0xd10c R X
.coredump.rom.text 0x40010000 0x54ef4 R X

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

Alvin1Zhang, 2020-12-17:
Thanks for reporting and sorry for the inconvenience, we will look into.

david-cermak, 2020-12-17:
@sebirdman Thanks for the clean and detailed report.
It's not trying to lock a nullptr semaphore:
```
#1  0x40094962 in abort () at ../dependencies/espressif/esp-idf/components/esp32/panic.c:175
#2  0x40096dd4 in xQueueGenericSend (xQueue=0x3f80b0c4, pvItemToQueue=0x0, xTicksToWait=<optimized out>, xCopyPosition=0) at ../dependencies/espressif/esp-idf/components/freertos/queue.c:720
#3  0x40166601 in sys_sem_signal (sem=0x3f80a694) at ../dependencies/espressif/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:133
```
* `sys_sem_signal()` uses FreeRTOS semaphores which in turn use [generic queues](https://github.com/espressif/esp-idf/blob/e5f754b26cb2e77ba1ad0423b83536788c14cfc5/components/freertos/include/freertos/semphr.h#L474) with null items
* queue consistency check uses asserts to validate the API is in line with internal structures
* one of the checks fails in [here](https://github.com/espressif/esp-idf/blob/e5f754b26cb2e77ba1ad0423b83536788c14cfc5/components/freertos/queue.c#L720) testing the queue used as a semaphore has zero itemsize, apparently hitting the panic

The only reason I could think of, could be a memory corruption of some kind. Fortunately, here we know which memory location gets corrupted. The easiest method to check on this issue is to connect a debugger and set a watchpoint to this field https://github.com/espressif/esp-idf/blob/e5f754b26cb2e77ba1ad0423b83536788c14cfc5/components/freertos/queue.c#L153. But since the issue is happening in the field, randomly, might be more convenient to use the `esp_set_watchpoint()` api:
* https://github.com/espressif/esp-idf/blob/e5f754b26cb2e77ba1ad0423b83536788c14cfc5/components/xtensa/include/esp_debug_helpers.h#L70
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html?highlight=esp_set_watchpoint#finding-heap-corruption

sebirdman, 2020-12-19:
Going to close, i think i found the memory corruption, thanks for the help @david-cermak !

closed 2020-12-19 by sebirdman