espressif/esp-idf #6285: CMake Build System Ignoring -DPYTHON

AndruePeters, 2020-12-16:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-2136-gb0150615d
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux] (WSL on Windows 10)
- Using an IDE?: [Yes] Using CLion on Windows, but with a WSL profile
- Power Supply:         [USB]

## Problem Description

According to the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#setting-up-the-python-interpreter), specifying -DPYTHON=/path/to/python should change the Python interpreter used. However, this seems to have no effect. 

### Expected Behavior
The python interpretor used should be /path/to/python

### Actual Behavior
'python' is chosen instead of '/path/to/python'

### Steps to reproduce

1. Use a CMake project  that uses IDF as a library. [Here](https://github.com/AndruePeters/gaggia_classic_pro_pid) is my current setup.
2. Pass -DPYTHON=~/.espressif/python_env/idf4.3_py3.8_env/bin/python3
3. Build the project using CMake. 
4. Notice that it chooses /usr/bin/python as the interpreter 

This bottom image illustrates a couple things. First notice that it chooses /usr/bin/python for the interpreter still. I modified _build.cmake_ to the value of the python variables. Notice the difference. The lowercase **python** defaults to "python", but the uppercase **PYTHON** is the correct value that is passed into CMake.  So, the build system isn't properly choosing the correct interpreter. 
[image]

### Code to reproduce this issue
```
#
# Check if the Python interpreter used for the build has all the required modules.
#
function(__build_check_python)
    idf_build_get_property(check __CHECK_PYTHON)
    if(check)
        idf_build_get_property(python PYTHON)
        idf_build_get_property(idf_path IDF_PATH)
        message(STATUS "Checking Python dependencies...")
        message(STATUS "python: ${python}")
        message(STATUS "PYTHON: ${PYTHON}")
        execute_process(COMMAND "${python}" "${idf_path}/tools/check_python_dependencies.py"
            RESULT_VARIABLE result)
        if(NOT result EQUAL 0)
            message(FATAL_ERROR "Some Python dependencies must be installed. Check above message for details.")
        endif()
    endif()
endfunction()
```

AndruePeters, 2020-12-16:
The only reason it says that the python requirements are satisfied is because I ran ```/usr/bin/python -m pip install -r ~/Development/sdk/esp-idf/requirements.txt``` which installed it for the global python.

projectgus, 2021-03-31:
Hi @AndruePeters,

Thanks for the very clear report and for being patient while someone got back to you (which took much longer than it should have, sorry!)

I have a fix for this internally now. It may take a little while to merge and then backport to the release/v4.3 branch.

In the meantime, a workaround that works with ESP-IDF before and after the fix is to either pass both `python` and `PYTHON` on the command line (clunky), or alternatively add a workaround in the top of your CMakeLists.txt file:

```cmake
# workaround for https://github.com/espressif/esp-idf/issues/6285
if(PYTHON)
    set(python "${PYTHON}")
endif()

project(...)
```

(This can be removed after the IDF version has the upcoming fix, but will continue to work  with either version.)

Angus

PS Great project! :coffee:

closed 2021-04-15 by espressif-bot