espressif/esp-idf #6286: unlikely error from i2c_cmd_link_append()

mzimmers, 2020-12-16:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // v4.1-520-gc3324a82a
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (please give details)] Qt Creator
- Power Supply:         [USB]

## Problem Description

Frequent error on I2C reads/writes. Problem is not 100% reproducible. The error message comes from i2c_cmd_link_append(), when it gets an error from one of these two calls:
```cpp
        cmd_desc->head = (i2c_cmd_link_t*) heap_caps_calloc(1, sizeof(i2c_cmd_link_t), MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
        cmd_desc->cur->next = (i2c_cmd_link_t*) heap_caps_calloc(1, sizeof(i2c_cmd_link_t), MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
```
### Expected Behavior

All calls to i2c_master_* return successfully.

### Actual Behavior

Various calls to i2c_master_* return error:

E (2006015) i2c: i2c command link malloc error
I (2006015) memreport: memReport(): there are 4336 MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL bytes free.
E (2006015) Max77818: readReg(): error "ESP_FAIL" calling i2c_master_write_byte (2) on Charger, register 0xb3.

(The last two lines are my own error reporting.)

CONFIG_SPIRAM_USE_MALLOC is enabled.
### Steps to reproduce

### Code to reproduce this issue

This is a snippet from my register read routine. I get the error from various i2c_master_* calls here.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5705446/sdkconfig.txt)

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
        // build the i2c transfer.
        rc = i2c_master_start(m_i2c_cmd);
        if (rc != ESP_OK)
        {
			memReport();
			s = getDevName(devAddr);
            ESP_LOGE(TAG, "readReg(): error \"%s\" calling i2c_master_start (1) on %s, register 0x%x.",
                     esp_err_to_name(rc),
                     s.c_str(),
                     regAddr);
            continue;
        }
        rc = i2c_master_write_byte(m_i2c_cmd, addrWrite, true);
        if (rc != ESP_OK)
        {
			memReport();
			s = getDevName(devAddr);
            ESP_LOGE(TAG, "readReg(): error \"%s\" calling i2c_master_write_byte (1) on %s, register 0x%x.",
                     esp_err_to_name(rc),
                     s.c_str(),
                     regAddr);
            continue;
        }
        rc = i2c_master_write_byte(m_i2c_cmd, regAddr, true);
        if (rc != ESP_OK)
        {
			memReport();
			s = getDevName(devAddr);
            ESP_LOGE(TAG, "readReg(): error \"%s\" calling i2c_master_write_byte (2) on %s, register 0x%x.",
                     esp_err_to_name(rc),
                     s.c_str(),
                     regAddr);
            continue;
        }
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-12-17:
Thanks for reporting, we will look into.

mythbuster5, 2020-12-17:
Ok. Can you show your `i2c_config_t` structure? BTW, the debug log maybe not show accurately, could you give me a new debug log? thanks~

mzimmers, 2020-12-17:
```cpp
    i2c_config_t m_i2c;
    m_i2c.mode = I2C_MODE_MASTER;
    m_i2c.sda_io_num = GPIO_I2C_SDA;
    m_i2c.sda_pullup_en = GPIO_PULLUP_ENABLE;
    m_i2c.scl_io_num = GPIO_I2C_SCL;
    m_i2c.scl_pullup_en = GPIO_PULLUP_ENABLE;
    m_i2c.master.clk_speed = CONFIG_I2C_MASTER_FREQUENCY;

    ESP_ERROR_CHECK(i2c_param_config(I2C_PORT_NBR, &m_i2c));
    ESP_ERROR_CHECK(i2c_driver_install(I2C_PORT_NBR, I2C_MODE_MASTER, 16, 16, 0));
```
I don't really have a debug log, but I was getting this error:
    Mon Dec 14 18:03:44 2020 E (614193) i2c: i2c command link malloc error
This was happening at several steps during the i2c_master_* sequence.

mythbuster5, 2020-12-17:
@mzimmers 

```
i2c_config_t m_i2c;
    m_i2c.mode = I2C_MODE_MASTER;
    m_i2c.sda_io_num = GPIO_I2C_SDA;
    m_i2c.sda_pullup_en = GPIO_PULLUP_ENABLE;
    m_i2c.scl_io_num = GPIO_I2C_SCL;
    m_i2c.scl_pullup_en = GPIO_PULLUP_ENABLE;
    m_i2c.master.clk_speed = CONFIG_I2C_MASTER_FREQUENCY;
```
You ignored `clk_flags` in `i2c_config_t`, you'd better set it zero(0) on esp32, the related documentation is in progress.

BTW, for initializing C structure, we strongly recommand to use a more formal way, like：
```
struct_name_t xxx = {
    .content = value;
    ....
};
```
for `i2c_config_t` please imitate what https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/examples/peripherals/i2c/i2c_self_test/main/i2c_example_main.c#L147 does

mzimmers, 2020-12-17:
my i2c_config_t struct doesn't have a clk_flags member.
```cpp
typedef struct{
    i2c_mode_t mode;     /*!< I2C mode */
    int sda_io_num;      /*!< GPIO number for I2C sda signal */
    int scl_io_num;      /*!< GPIO number for I2C scl signal */
    bool sda_pullup_en;  /*!< Internal GPIO pull mode for I2C sda signal*/
    bool scl_pullup_en;  /*!< Internal GPIO pull mode for I2C scl signal*/

    union {
        struct {
            uint32_t clk_speed;     /*!< I2C clock frequency for master mode, (no higher than 1MHz for now) */
        } master;
        struct {
            uint8_t addr_10bit_en;  /*!< I2C 10bit address mode enable for slave mode */
            uint16_t slave_addr;    /*!< I2C address for slave mode */
        } slave;
    };
} i2c_config_t;
```
I updated my code above with the values for my constants.

mythbuster5, 2020-12-17:
`ESP_ERROR_CHECK(i2c_driver_install(I2C_PORT_NBR, I2C_MODE_MASTER, 16, 16, 0));`
little wierd for your this line, i2c only needs `slv_rx_bug_len` and `slv_tx_buf_len` when it works on slave mode. On master mode, it should be 0

mzimmers, 2020-12-17:
Thank you, mythbuster5. I've made that change. I'm not sure how that will help the problem, but it can't hurt.

So, what happened to the clk_flags member of the i2c_config_t struct? Was that deprecated? Is there a different call I should be making, and if so, what should those flags be?

ginkgm, 2020-12-22:
Hi @mzimmers @mythbuster5 
It ia s general rule that all the structure used by IDF drivers (and maybe other parts) should be properly initialized, and unused fields should be set to 0 before use.

Suggested initialization is one of the following
```
xxx_config_t xxx_config={}; //all set to 0

xxx_config_t xxx_config={
.member1 = 1,
}; //all set to 0, but member1 set to 1

memset(&xxx_config, 0, sizeof(xxx_config)); // all set to 0

xxx_context_t = (xxx_context_t*)calloc(sizeof(xxx_context_t)); //malloc and all set to 0.
```

To support new features, we will continuously add new members into existing structures, and the ways as above can keep your code compatible. A local structure on the stack is supposed to be has random value, and without explicit initialization, some members may left uninitialized.

We will not suddenly remove any fields **for one chip that is already supported**, so don't worry about this. Sorry about the misleading in our example to use `i2c_config_t i2c_config;`.

ginkgm, 2020-12-22:
If we use new flags field to add new feature support, we will make sure the default value (0) is back-compatible. In this case, if clk_flags is set to 0, it will still work with the default clock source (APB)

mzimmers, 2020-12-22:
OK, thanks guys. I wasn't aware of that convention. I'll use it going forward.

Since implementing the suggestions made above, I'm no longer seeing my errors, so I think I can close this issue. Thanks again!

closed 2020-12-22 by mzimmers