espressif/esp-idf #6287: ESP_TLS_ERR_SSL_WANT_READ undeclared

smokezp, 2020-12-16:
Hi, I have issue. I'm trying to install sh2lib.h.
I followed instructions: [https://techtutorialsx.com/2018/11/23/esp32-http-2-get-request-with-continuous-stream-response/](url)
Also I found almost the same instruction on youtube.

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
I get an error:

libraries/sh2lib/sh2lib.c: In function 'callback_send_inner':
libraries/sh2lib/sh2lib.c:42:19: error: 'ESP_TLS_ERR_SSL_WANT_READ' undeclared (first use in this function)
         if (rv == ESP_TLS_ERR_SSL_WANT_READ || rv == ESP_TLS_ERR_SSL_WANT_WRITE) {
         
It looks like const ESP_TLS_ERR_SSL_WANT_READ is not declared. From where it should come? what am I going wrong?   

### Steps to reproduce

1. Added folder sh2lib with files sh2lib.c and sh2lib.h
2. When I trying to upload the code, it says:

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.

extern "C" {
#include <sh2lib.h>
}

mahavirj, 2020-12-17:
@smokezp 

This is defined at below:
https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/components/esp-tls/esp_tls.h#L69

Can you please try with latest IDF release?

mahavirj, 2020-12-23:
@smokezp Closing the issue, please feel free to re-open if you are still facing issue with latest IDF release

closed 2020-12-23 by mahavirj

raysconcept, 2021-04-13:
Hi Mahavirj, I am having the exact samen problem. Can you explain your solution? How should I try the latest IDF release? Thx in advance!

mahavirj, 2021-04-14:
@raysconcept 

Please see following pointers:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-2-get-esp-idf
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html

raysconcept, 2021-04-14:
Thank you very much! I am installing that now. 
Though..I have already arduinoIDE, VisualStudio and Eclipse installed. How does IDF relate to any of those?

raysconcept, 2021-04-14:
[image]

AL5Volgar, 2021-11-26:
Hi,

I want to use the sh2lib library for my project.
However i have the same error as the first comment : 
[image]

[image]

I'm currently using the Arduino IDE, and I have already installed ESP IDF 4.3

Can you help me ?