espressif/esp-idf #6288: E (243) psram: PSRAM ID read error: 0xffffffff E (243) cpu_start: Failed to init external RAM!

kaibin1111, 2020-12-17:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6976
ho 0 tail 12 room 4
load:0x40078000,len:14756
load:0x40080400,len:3712
0x40080400: _init at ??:?

entry 0x40080678
I (30) boot: ESP-IDF v4.3-dev-2136-gb0150615d-dirty 2nd stage bootloader
I (31) boot: compile time 11:33:23
I (32) boot: chip revision: 1
I (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0aea4h ( 44708) map
I (134) esp_image: segment 1: paddr=0001aecc vaddr=3ffb0000 size=028dch ( 10460) load
I (139) esp_image: segment 2: paddr=0001d7b0 vaddr=40080000 size=00404h (  1028) load
I (140) esp_image: segment 3: paddr=0001dbbc vaddr=40080404 size=0245ch (  9308) load
I (153) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=18690h ( 99984) map
I (198) esp_image: segment 5: paddr=000386b8 vaddr=40082860 size=0bd14h ( 48404) load
I (230) boot: Loaded app from partition at offset 0x10000
I (230) boot: Disabling RNG early entropy source...
I (242) psram: This chip is ESP32-D0WD
E (243) psram: PSRAM ID read error: 0xffffffff
E (243) cpu_start: Failed to init external RAM!
Re-enable cpu cache.

abort() was called at PC 0x4008124a on core 0
0x4008124a: call_start_cpu0 at C:/Users/make/esp/test/esp-idf/components/esp_system/port/cpu_start.c:348 (discriminator 1)

Backtrace:0x40088196:0x3ffe3ba0 0x40088909:0x3ffe3bc0 0x4008d5ae:0x3ffe3be0 0x4008124a:0x3ffe3c50 0x40079201:0x3ffe3c80 |<-CORRUPTED
0x40088196: panic_abort at C:/Users/make/esp/test/esp-idf/components/esp_system/panic.c:367

0x40088909: esp_system_abort at C:/Users/make/esp/test/esp-idf/components/esp_system/system_api.c:108

0x4008d5ae: abort at C:/Users/make/esp/test/esp-idf/components/newlib/abort.c:46

0x4008124a: call_start_cpu0 at C:/Users/make/esp/test/esp-idf/components/esp_system/port/cpu_start.c:348 (discriminator 1)

ELF file SHA256: 0d323453274f18a3

Rebooting...

Spritetm, 2020-12-17:
What hardware are you using? If there's a module on the board, what module is it? How is your GPIO (especially GPIO16/17) connected?

kaibin1111, 2020-12-17:
> 您正在使用什么硬件？如果板上有一个模块，它是什么模块？GPIO（尤其是GPIO16 / 17）如何连接？

> What hardware are you using? If there's a module on the board, what module is it? How is your GPIO (especially GPIO16/17) connected?

ESP32-CAM,  How do I set it

Spritetm, 2020-12-17:
Hm, if nothing else is connected to your ESP32-CAM, you may have a broken one. That board is supposed to come with PSRAM.

Alvin1Zhang, 2021-01-04:
@kaibin1111 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

JohnMacrae, 2021-04-28:
Hello @Alvin1Zhang, I too have seen this error develop on a very recent wroom 32UE (Mouser - last week). Initially it ran a simple I2C scanner utility just fine (and continues to do so) but I noticed after uploading a few more utilities to test the board that the PSRAM error came up on boot.

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (131) psram: PSRAM ID read error: 0xffffffff

thomas725, 2021-05-10:
One of my ESP32-Cam boards has this issue too.
Is the most likely explanation hardware failure of the chip?

I'm not sure if it worked before or if I just didn't see the issue before..

chegewara, 2021-05-10:
Last time i got similar issue was caused by bootstrap pin. I had some voltage on flash voltage control pin (IO12)
> If driven High, flash voltage (VDD_SDIO) is 1.8V not default 3.3V. Has internal pull-down, so unconnected = Low = 3.3V. May prevent flashing and/or booting if 3.3V flash is used and this pin is pulled high, causing the flash to brownout. See the ESP32 datasheet for more details.

It was not easy to find it, because flash started but psram failed to init.

heyspacebuck, 2021-06-05:
I am experiencing this error as well. I was able to connect to the ESP32 once without issue and upload a program. However, when I did that I inadvertently used the "PSRAM enabled" setting. Now when I plug in my board to USB, I see a steady stream of serial output errors as described by the others above.

I am no longer able to connect to the board in order to re-program it. Has anyone found a way to connect and reprogram a board that is experiencing this boot error loop?

hasibkyau, 2022-01-27:
I'm having same kind of error. But can't figure out how to solve it.

02:39:15.212 -> ets Jun  8 2016 00:22:57
02:39:15.212 -> 
02:39:15.212 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:39:15.212 -> configsip: 0, SPIWP:0xee
02:39:15.212 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
02:39:15.212 -> mode:DIO, clock div:1
02:39:15.212 -> load:0x3fff0018,len:4
02:39:15.260 -> load:0x3fff001c,len:1216
02:39:15.260 -> ho 0 tail 12 room 4
02:39:15.260 -> load:0x40078000,len:10944
02:39:15.260 -> load:0x40080400,len:6388
02:39:15.260 -> entry 0x400806b4
02:39:15.496 -> E (70) psram: PSRAM ID read error: 0xffffffff
02:39:15.543 -> 
02:39:15.591 -> [E][camera.c:1113] camera_probe(): Detected camera not supported.
02:39:15.591 -> [E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004

BigGene, 2022-02-19:
Same problem here as well. Any ideas on how to fix it?

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (79) psram: PSRAM ID read error: 0xffffffff

[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004

Tony-Project, 2022-04-10:
Faced a similar problem.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
E (188) psram: PSRAM ID read error: 0xffffffff
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!

trying to build a Qr code scanner...

bil7317, 2022-07-26:
I ma fa

> Same problem here as well. Any ideas on how to fix it?
>
> ets Jul 29 2019 12:21:46
>
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee [...]

I am facing same error while using ILI9488 Touch display with ESP32. Kindly help!!

chegewara, 2022-08-29:
I wanted to open new issue, but we have very old one here, so i will join.
New S3 devkit:
https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-S3-DevKitC-1-N32R8V?qs=Li%252BoUPsLEnvTvWIWLPCZ4g%3D%3D

sdkconfig.defaults:
```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) Project Minimal Configuration
#
CONFIG_IDF_TARGET="esp32s3"
CONFIG_ESPTOOLPY_FLASHSIZE_32MB=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
CONFIG_ESP32_WIFI_RX_BA_WIN=6
```
logs:
```
E (178) psram: PSRAM ID read error: 0x00ffffff
E (178) spiram: SPI RAM enabled but initialization failed. Bailing out.
```
esp-idf:
- v4.4.2-155-g23d5a582cb
- v5.1-dev-303-g2b1bee166d

Offtopic: devkit is with 32MB flash, but esp-idf still does not support more than 16MB
```
Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
```

Thanks

chegewara, 2022-09-09:
Ok, i solved the problem. It was flash and PSRAM type mismatch setup in menuconfig.
The devkit i have is supporting OPI, and i think the default sdkconfig values were causing conflict.

TotiTolvukall, 2023-01-15:
> Hi, anyone managed to solve the PSRAM issue on an (ali) ESP32-CAM board? I the pin connections (GPIO16/17) are ok, but no matter how I set the esp-idf, the APS6404L-3SQR is not detected (even tried grounding IO12)

I too would like to know if this is a solvable issue on the AliExpress ESP32-CAM board.

zackfuchtel, 2023-01-17:
As mentioned in the code, select ESP32 Wrover Module. Was fixing it for me.

```
//
// WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
//            or another board which has PSRAM enabled
//            -->   ESP32 Board V1.0.6 is needed   <--
//
```

Tilie, 2023-01-20:
hey guys, I solved this problem if you bought this module name YD-ESP32-S3 from aliexpress, they used the `Octal Mode PSRAM` instead of `Quad Mode PSRAM`, so the solution is change the SPI RAM config from the `Quad Mode PSRAM` to `Octal Mode PSRAM` in SDK Configuration editor

sukesh-ak, 2023-03-14:
In my case, the error was because the PSRAM was OPI (Octal Mode PSRAM) and not QSPI (Quad Mode PSRAM).
You can change from `menuconfig`
`Component config > ESP PSRAM`

micheldufresne, 2023-06-26:
Hello, I know it's an old problem ... The problem had never appeared with this module (ESP32 WROOM 32) that I was already using to connect to a wifi network. The 1st time I used MQTT the crashes appeared. I had to test lots of libraries, lots of options in Vscode. So far I find that with this setting of the platformio.ini the problem magically disappeared...

[env:esp-32]
platform = espressif32
board =  esp32dev
framework = arduino