espressif/esp-idf #6289: Esp32: Sometimes Wifi AP never assign IP to stations

FrGrQuim, 2020-12-17:
I use the esp32-wrover-b as a Wifi AP. Sometimes when I try to connect to it with a device, this one never receive IP address but it is considered as connected (SYSTEM_EVENT_AP_STACONNECTED event received but SYSTEM_EVENT_AP_STAIPASSIGNED never received). When this problem occurs once, it continues until I restart the esp (I never have tried to only restart the AP). That means that if I try to connect other devices, it also doesn't received IP address if I have not restart the esp.

Is someone already had this problem or has an idea to make my AP more robust?

Thanks in advance.

AxelLin, 2020-12-17:
I observed the same symptom. (But not easy to reproduce).
https://github.com/espressif/esp-idf/issues/6108

FrGrQuim, 2020-12-17:
Exactly, I suppose that it is the same root cause. Thanks

YouDONG-ESP, 2020-12-17:
Hi @FrGrQuim, thanks for report, could you provid this infomation for us to debug?
1. IDF version
2. Steps to reproduce
3. Code to reproduce this issue

PilotAware, 2020-12-18:
@YouDONG-ESP 
I have created a testcase to reproduce the issue
I have seen this same problem reported in numerous issue reports but in different descriptions, it seems it is quite endemic
It would appear to be some kind of race condition.

In my application, I need 4 stations to connect, and listen on a UDP port
the esp32 needs to know the IP Addresses of the stations in order to perform unicast UDP transmissions to each of the stations. 
This can only be done if the IP address of the station is known. On a reboot of the esp32 the stations often come back with an IP address of 0.0.0.0

I have added a test here
https://github.com/PilotAware/esp-idf-bugs

a full description here
https://github.com/PilotAware/esp-idf-bugs/blob/main/softAP/README.bug.txt

inlined here
```
# to build
$ make build flash

# flash and monitor
$ make monitor

# connect a station to esp32 SSID TestSSID
# reported as connected in the monitor window
I (29283) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (29283) wifi:station: 00:0f:00:e1:d4:8f join, AID=1, bgn, 40U
I (29283) wifi softAP: station 00:0f:00:e1:d4:8f join, AID=1
I (29283) wifi softAP: WIFI Listing:
I (29293) wifi softAP:     [0]: 000F00E1D48F 0.0.0.0
I (29523) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (30763) wifi softAP: WIFI Listing:
I (30763) wifi softAP:     [0]: 000F00E1D48F 192.168.4.2

# A wifi listing is repeated every 10 seconds
#
# reboot esp32 device 2 methods described
# method 1
#     type CTRL']' in monitor
#     now reconnect monitor
$ make monitor

# lots of errors of the form 

W (3223) wifi:m f deauth

E (3223) wifi:esf_buf: t=3 l=28 max:32, alloc:32 no eb, TXQ_BLOCK=4000
W (3233) wifi:alloc eb len=28 type=3 fail, heap:226836

# ulitimately the IP address is not determined and listed as 0.0.0.0
I (11823) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (11823) wifi:station: 00:0f:00:e1:d4:8f join, AID=1, bgn, 40U
I (11823) wifi softAP: station 00:0f:00:e1:d4:8f join, AID=1
I (11833) wifi softAP: WIFI Listing:
I (11833) wifi softAP:     [0]: 000F00E1D48F 0.0.0.0
I (20773) wifi softAP: WIFI Listing:
I (20773) wifi softAP:     [0]: 000F00E1D48F 0.0.0.0

I have noticed this is specific to some station types
I have connected 
- Linux station    000F00E1D48F
- iPhone station   5611A01BBB66
- iPad station     1C9E46CB0740
- Android station  74EB80B26E12

The real problem is that the station initially handed an IP Address before the reboot,
thinks it still has that address, thus causing a conflict when this same address is handed out
to another station 

I tried connecting the 4 stations listed above as stations, after the reset I get the following

I (9883) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (10293) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (10293) wifi:station: 00:0f:00:e1:d4:8f join, AID=3, bgn, 40U
I (10293) wifi softAP: station 00:0f:00:e1:d4:8f join, AID=3
I (10303) wifi softAP: WIFI Listing:
I (10303) wifi softAP:     [0]: 5611A01BBB66 0.0.0.0
I (10313) wifi softAP:     [1]: 1C9E46CB0740 192.168.4.2
I (10313) wifi softAP:     [2]: 74EB80B26E12 0.0.0.0
I (10323) wifi softAP:     [3]: 000F00E1D48F 0.0.0.0

the behavior is random - indicating a race condition.

a solution would be to be able to disconnect a station and force authentication 
- is that possible ?

There is no other method to ensure unicast UDP can be implemented
```

eroom1966, 2020-12-18:
@YouDONG-ESP 
The problem most definitely exists, and is easily detectable, programmatically by iterating over the station list, and interrogating their IP Address
a workaround at the application level is to be able to force a station whose IP address is reported as 0.0.0.0 to re-authenticate

is there a method to force re-authentication ?

eroom1966, 2020-12-19:
@YouDONG-ESP 

I have seen this problem reported multiple times on multiple threads, but I do not see anyone providing a solution.
I have taken the time to produce a simple testcase which is definitive, please take the time to provide either a solution or workaround 
Thx

PilotAware, 2020-12-19:
I allowed the above example to run continuously the IP eventually gets resolved after a very long time
```
(2281294) wifi softAP:     [0]: 000F00E1D48F 0.0.0.0
(2291294) wifi softAP: WIFI Listing:
(2291294) wifi softAP:     [0]: 000F00E1D48F 0.0.0.0
(2301294) wifi softAP: WIFI Listing:
(2301294) wifi softAP:     [0]: 000F00E1D48F 0.0.0.0
(2311294) wifi softAP: WIFI Listing:
(2311294) wifi softAP:     [0]: 000F00E1D48F 192.168.1.2
(2321294) wifi softAP: WIFI Listing:
(2321294) wifi softAP:     [0]: 000F00E1D48F 192.168.1.2
(2331294) wifi softAP: WIFI Listing:
(2331294) wifi softAP:     [0]: 000F00E1D48F 192.168.1.2
```

negativekelvin, 2020-12-19:
> a solution would be to be able to disconnect a station and force authentication 
> - is that possible ?

Have you tried https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/components/esp_wifi/include/esp_wifi.h#L369-L381

Is sounds like a dhcp client issue, client is not trying to renew dhcp lease on reconnection.  You could turn on dhcp debugging to verify.  Question is how to trigger the client to invalidate its own dhcp lease.

eroom1966, 2020-12-19:
@negativekelvin 
I did try the deathenticate method, it did not solve the issue
Also, I have seen the same problem from iphone, ipad, Android and linuxpc
The linuxpc is worst, I really suspect a race condition 
Thx
Lee

YouDONG-ESP, 2020-12-23:
Hi guys, the root cause of your issue has been found @PilotAware. It's a little bit complicated, I will explain to you:  

Rebooting a router normally takes over 30 seconds. Stations won't receive any beacon frame during this time which will lead the sta to a disconnected state because it thinks the router is power down. (For esp32 station mode, it will happen if no beacon has been received for the last 6 seconds). **The problem is that rebooting esp32's softap only takes 2, 3 seconds. It will not pass most of the device's disconnected threshold. Station will enter a beacon loss process, different device has different processing strategy:**

1. For the esp32 station mode, it will only send a probe request frame. So after the softap recovered, ap will reply a probe respond frame. At this moment, the station think the connection is still etablished but the ap think there is no connection. It is not a problem because when our ap recive data frame from an unconnected device, it will send deauth frame to kick it off.

2. For A linux device, the strategy is send probe request at first, send authentication frame after receiving reply and send association frame at the end. The only thing it won't do is DHCP process. So at the AP side, the wifi connection has been etablished, but no DHCP required has been recieved, so it will consider the station got IP 0.0.0.0. From the station side, it think the disconnection has never happened, so it will keep the previous IP. 

If your issue happen when the softap has been reboot, it should be the cause @FrGrQuim @eroom1966 @negativekelvin @AxelLin.

As you can see, it's a station's issue, I suggest you to use @negativekelvin 's method. For example, Call esp_wifi_deauth_sta() to kick off the station who has IP 0.0.0.0 for a a period of time to let them reconnect.

PilotAware, 2020-12-23:
@YouDONG-ESP 
Many thanks for investigating.
I am thinking what could be done as a workaround to the issue you describe ...

From what you are saying I think that the Linux client needs to be disconnected from the IP layer for a considerable amount of time, from experiments monitoring the Linux driver code I think that 30 seconds is sufficient

Is it possible (from the application layer) to blacklist the known MAC address which has not been assigned an IP address for a period of time ?
in which case the the device will need to renegotiate at some point.
As this is a race condition, is it possible to make the AP appear unavailable to the problem MAC address for 30 seconds ?

I think the problem is that this is deeply embedded in the library code, but if I could monitor the fact that for a given MAC Address, an IP has not been requested, then I want to blacklist that MAC for 30 seconds, then allow it to try to connect again - in which case it will request an IP Address.

YouDONG-ESP, 2020-12-24:
If we do this, it's gonna be a new feature with low priority. But I don't think we are going to add this feature recently. Seem's like there is no good solution for this issue, If the ap is rebooted, the best thing you can do is restart the sta manually.  Or you have to stop AP for ~30s to let the IP layer disconnect.

negativekelvin, 2020-12-24:
Blacklist would be useful for other reasons too

Stefano-Orsolini, 2020-12-24:
I had the same problem (ESP32 in SoftAP doesn't assign IP to clients) when using BLUFI to create the SoftAP.

Since I combined the wifi_init_softap routine from [idf softap example](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c) into my blufi code, I'm able to use blufi as usual and client obtains IP immediately (not the cleanest workaround though).

I believe wifi_init_softap routine from the example does something to properly start DHCP which its missing from other implementations.

AxelLin, 2020-12-28:
> 2. For A linux device, the strategy is send probe request at first, send authentication frame after receiving reply and send association frame at the end. The only thing it won't do is DHCP process. So at the AP side, the wifi connection has been etablished, but no DHCP required has been recieved, so it will consider the station got IP 0.0.0.0. From the station side, it think the disconnection has never happened, so it will keep the previous IP.
> 
> If your issue happens right after the softap reboot, it should be the cause @FrGrQuim @eroom1966 @negativekelvin @AxelLin.
> 

Observe IP conflict with below symptom:

My Linux Join the esp32 softAP.
The esp32 softAP reboot.
After esp32 softAP reboot, it looks my linux still has the same IP (192.168.4.2)
but the esp32 does not got IP_EVENT_AP_STAIPASSIGNED event.

So if another device try to join the esp32 softAP, it will get 192.168.4.2 then causes IP conflict.

YouDONG-ESP, 2021-01-21:
HI guys, a solution of this issue is to delay ~30s before ap init to imitate a regular router's reboot and force sta's ip layer disconnect. You may add it in your application if this issue really bothers you.
```    
vTaskDelay(30000/portTICK_RATE_MS);
wifi_init_softap();
```
We will probably not add blacklist recently, maybe in v5.0.

FrGrQuim, 2021-02-24:
We have an idea to fix it and I want to share it with you (feel free to give your opinion)

If we reduce the DHCPS lease time it force the client to update more often its IP and then the dhcp server can get it at the next update

jack0c, 2022-05-20:
@FrGrQuim any updates?

closed 2024-02-19 by jack0c