espressif/esp-idf #6291: MSI installer on windows 10 not working

sheldon123z, 2020-12-17:
## Environment Windows 10

- Development Kit:      [|ESP32-DevKitC|]
- IDF version (run ``git describe --tags`` to find it):
  v4.1

## Problem Description
I encountered this problem when install idf tools on a new windows 10 computer 

```
Setting up Python environment
fatal: not a git repository (or any of the parent directories): .git
WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128.
Creating a new Python environment in C:\Users\zheng\.espressif\python_env\idf4.1_py3.7_env
FileNotFoundError: [Errno 2] No such file or directory: 'D:\\Applications\\anacoda\\Lib\\venv\\scripts\\nt\\python.exe'
Traceback (most recent call last):
  File "C:\Users\zheng\esp\esp-idf\tools\idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "C:\Users\zheng\esp\esp-idf\tools\idf_tools.py", line 1383, in main
    action_func(args)
  File "C:\Users\zheng\esp\esp-idf\tools\idf_tools.py", line 1189, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "D:\Applications\anacoda\lib\subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['D:\\Applications\\anacoda\\python.exe', '-m', 'virtualenv', 'C:\\Users\\zheng\\.espressif\\python_env\\idf4.1_py3.7_env']' returned non-zero exit status 1.
```
//Detailed problem description goes here.
I tried uninstall and newer version python,git, but none of these worked. It seems that something happened when the idf tool is trying to create python envrionment, but used git? what should I do >?

Alvin1Zhang, 2020-12-18:
Thanks for reporting, we will look into.

georgik, 2020-12-18:
@sheldon123z Thank you for reporting the problem.
Please install Python 3.7 from python.org. This version works with all major releases of IDF.

Python from Anaconda distribution does not work well with IDF, because Anaconda contains modified version of Python which behaves differently than distribution from Python.org.

sheldon123z, 2020-12-20:
Hi @georgik,

Thank you for your instructions. After two days of struggles, I finally found out the real problem is that I had my vpn proxy opened while using the installation program. When I closed the vpn proxy, everything worked well.

closed 2020-12-20 by sheldon123z

georgik, 2021-02-02:
Update. New installers are available. The solution can co-exist with installed Anaconda Python.
- Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

The solution is based on Embedded Python 3.8.7 and prepared package of Python Wheels.