espressif/esp-idf #6293: I2C failure in C++ after add std::vector to code

Metmacher, 2020-12-17:
## Environment

- Development Kit:      ESP32-DevKitC 
- Kit version (for WroverKit/PicoKit/DevKitC): unknow
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-2136-gb0150615d
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  8.4.0
- Operating System:     Linux
- Using an IDE?:  VSCode
- Power Supply:    USB

## Problem Description

I setup an program for communication with I2C in C++. After add an `std::vector`, the I2C communication failed, here an mini example for reproduce:  

```cpp
#include "driver/gpio.h"
#include "driver/i2c.h"
#include "freertos/task.h"
#include "sdkconfig.h" // generated by "make menuconfig"
#include <vector>

extern "C"
{
	void app_main();
}

void init_I2C()
{
	i2c_config_t i2c_config;
	i2c_config.mode = I2C_MODE_MASTER;
	i2c_config.sda_io_num = GPIO_NUM_17;
	i2c_config.scl_io_num = GPIO_NUM_16;
	i2c_config.sda_pullup_en = GPIO_PULLUP_ENABLE;
	i2c_config.scl_pullup_en = GPIO_PULLUP_ENABLE;
	i2c_config.master.clk_speed = 100000;
	i2c_param_config(I2C_NUM_0, &i2c_config);
	i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);
}

void app_main(void)
{
	init_I2C();
	std::vector<std::vector<uint8_t>> tmpData; // this brings trouble
}
```

The `monitor` output:  
```bash
I (307) spi_flash: flash io: dio
I (312) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (322) i2c: i2c_param_config(644): i2c clock choice is invalid, please check flag and frequency
```

If the `vector` is removed, the error didn't occur.  

If I also run my full program, and try communication on this interface this error will also prompted:  
```bash
E (1344) i2c: i2c_set_pin(825): scl and sda gpio numbers are the same
```

Alvin1Zhang, 2020-12-18:
Thanks for reporting and letting us know, we will look into.

boarchuz, 2020-12-18:
4.3 adds 'clk_flags' field to i2c_config_t. You need to zero i2c_config_t or set clk_flags.

Metmacher, 2020-12-18:
> You need to zero i2c_config_t or set clk_flags.

Solved it!  
Solution: add `i2c_config.clk_flags = 0;` to the config section.  

> 4.3 adds 'clk_flags' field to i2c_config_t. 

I checked out version 4.2 and there the initial code is working.

@boarchuz  thank you

closed 2020-12-18 by Metmacher