espressif/esp-idf #6295: LED is not getting on in esp-idf/tree/master/examples/get-started/blink

DipaliDh, 2020-12-18:
Hi,
         I am trying to blink an led on esp32s2 using example code in espidf. I programmed blink.c but the led is not blinking or even turn on. I have configured GPIO5 in menuconfig for LED. 

Please let me know any configuration I am missing

Thanks
Dipali

felmue, 2020-12-18:
Hello @DipaliDh 

- which board are you using?
- have you tried another GPIO? Maybe GPIO5 is already used for something else on your board?
- are you sure the code is running? Do you get the correct output on the serial console? E.g.
```
Turning on the LED
Turning off the LED
...
```

Thanks
Felix

DipaliDh, 2020-12-18:
Hi, @felmue 
1. I am using ESP32-S2-Saola-1_v1.2.
2.  I also tried using another gpio. but it did not work. 
3. The code is running and I am getting output on the serial monitor, i.e. Turning on the LED 
                                                                                                                     Turning off the LED
    But led is not getting on in the hardware

felmue, 2020-12-18:
Hello @DipaliDh 

hmm, that is strange. Maybe the LED is defective? Can you tell me how you've connected the LED?

Thanks
Felix

DipaliDh, 2020-12-23:
Hi, 470ohms resister is connected to the anode, and it's connected on gpio. A cathode is grounded. 

Thanks
Dipali

felmue, 2020-12-23:
Hello @DipaliDh 

and I assume the LED lights up when you connect it to 3.3v and ground?

Thanks
Felix

DipaliDh, 2020-12-30:
Hi @felmue 
 I will test the LED with 3.3V... Now, doing another task so couldn't test it ...thank you for your help

Kiznoh, 2021-01-14:
Hi @DipaliDh,
are you still experiencing the issue?
If so, can you measure the voltage on the pin with nothing connected?
You should see a changing voltage between 0V and 3.3V.
This would rule out a possible overload and ensure you are connecting the LED to the correct pin.

If the above doesn't help you fix the issue would you mind posting a picture of your setup?

Thank you
Jan

dobairoland, 2021-02-08:
This doesn't seem to an IDF bug and there is no recent activity discussing it. Therefore, I'm closing it.

closed 2021-02-08 by dobairoland