espressif/esp-idf #6296: USB Serial Driver hangs seemingly when writing data too fast

JustinOng, 2020-12-18:
## Environment

- Development Kit:      [nanoESP32-S2](https://github.com/wuxx/nanoESP32-S2/blob/master/schematic/nanoESP32S2-v1.2.pdf)
- Module or chip used:  ESP32-S2-WROOM
- IDF version: `v4.3-dev-2136-gb0150615d`
- Build System:         `idf.py`
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0`
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

USB Serial Device hangs seemingly when writing data too fast

### Expected Behavior

Writing multiple characters eg `h`, `e`, `l`, `l`, `o` one at a time should succeed

### Actual Behavior

USB Serial Device hangs after writing `h` and `e`

### Steps to reproduce

1. Flash the [`tusb_serial_device example`](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/tusb_serial_device)
2. Connect the native USB port to the computere
3. Send data to the native USB port:
``` python
import time
import serial

ser = serial.Serial("COM3", 115200)

for c in b'hello':
    print(c)
    ser.write(bytes([c]))
    # time.sleep(0.01)

ser.close()
```

The Serial Device hanging is observed when `time.sleep(0.01)` is commented out. The same code has been tested with a CP2102-based USB-UART adapter and sends fine.

## Debug Logs

Failure:
```
V (11904) TUSB:DCD: Transfer <-> EP0, HOST->DEV (out), pkgs: 1, bytes: 7
V (11904) TUSB:DCD: dcd_int_handler - rx!
V (11904) TUSB:DCD: TUSB IRQ - RX type : Out packet
V (11904) TUSB:DCD: USB - receive_packet
V (11904) TUSB:DCD: dcd_int_handler - rx!
V (11904) TUSB:DCD: TUSB IRQ - RX type : Out packet done
V (11904) TUSB:DCD: dcd_int_handler - OUT endpoint!
V (11904) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
V (11944) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 0
V (11944) TUSB:DCD: dcd_int_handler - IN endpoint!
V (11944) TUSB:DCD: TUSB IRQ - EP IN 0
V (11944) TUSB:DCD: TUSB IRQ - IN XFER complete!
V (11944) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
V (11944) TUSB:DCD: USB - transmit_packet
V (11944) TUSB:DCD: dcd_int_handler - rx!
V (11944) TUSB:DCD: TUSB IRQ - RX : Setup packet : 0x000021a1 0x00070000
V (11944) TUSB:DCD: dcd_int_handler - rx!
V (11944) TUSB:DCD: TUSB IRQ - RX : Setup packet done
V (11944) TUSB:DCD: dcd_int_handler - OUT endpoint!
V (11994) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 7
V (12004) TUSB:DCD: dcd_int_handler - IN endpoint!
V (12004) TUSB:DCD: TUSB IRQ - EP IN 0
V (12004) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
V (12004) TUSB:DCD: USB - transmit_packet
V (12014) TUSB:DCD: dcd_int_handler - IN endpoint!
V (12014) TUSB:DCD: TUSB IRQ - EP IN 0
V (12014) TUSB:DCD: TUSB IRQ - IN XFER complete!
V (12014) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
V (12014) TUSB:DCD: USB - transmit_packet
V (12044) TUSB:DCD: Transfer <-> EP0, HOST->DEV (out), pkgs: 1, bytes: 0
V (12044) TUSB:DCD: dcd_int_handler - rx!
V (12044) TUSB:DCD: TUSB IRQ - RX type : Out packet
V (12044) TUSB:DCD: USB - receive_packet
V (12044) TUSB:DCD: dcd_int_handler - rx!
V (12044) TUSB:DCD: TUSB IRQ - RX type : Out packet done
V (12044) TUSB:DCD: dcd_int_handler - rx!
V (12044) TUSB:DCD: TUSB IRQ - RX type : Out packet
V (12044) TUSB:DCD: USB - receive_packet
V (12044) TUSB:DCD: dcd_int_handler - OUT endpoint!
V (12044) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
V (12044) TUSB:DCD: dcd_int_handler - rx!
V (12044) TUSB:DCD: TUSB IRQ - RX type : Out packet done
V (12044) TUSB:DCD: dcd_int_handler - OUT endpoint!
V (12044) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
V (12114) TUSB:DCD: Transfer <-> EP2, HOST->DEV (out), pkgs: 1, bytes: 64
V (12124) TUSB:DCD: dcd_int_handler - rx!
V (12124) TUSB:DCD: TUSB IRQ - RX type : Out packet
V (12124) TUSB:DCD: USB - receive_packet
V (12124) TUSB:DCD: dcd_int_handler - rx!
V (12124) TUSB:DCD: TUSB IRQ - RX type : Out packet done
V (12124) TUSB:DCD: dcd_int_handler - OUT endpoint!
V (12124) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
V (12124) tusb_cdc_acm: Sent 1 bytes to the buffer
I (12164) example: Got data (1 bytes): h
V (12164) TUSB:DCD: Transfer <-> EP2, USB0.HOST (in), pkgs: 1, bytes: 1
V (12174) TUSB:DCD: dcd_int_handler - IN endpoint!
V (12174) TUSB:DCD: TUSB IRQ - EP IN 2
V (12174) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
V (12174) TUSB:DCD: USB - transmit_packet
V (12194) TUSB:DCD: dcd_int_handler - IN endpoint!
V (12194) TUSB:DCD: TUSB IRQ - EP IN 2
V (12194) TUSB:DCD: TUSB IRQ - IN XFER complete!
V (12194) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
V (12194) TUSB:DCD: USB - transmit_packet
V (12214) TUSB:DCD: Transfer <-> EP2, HOST->DEV (out), pkgs: 1, bytes: 64
V (12224) TUSB:DCD: dcd_int_handler - rx!
V (12224) TUSB:DCD: TUSB IRQ - RX type : Out packet
V (12224) TUSB:DCD: USB - receive_packet
V (12224) TUSB:DCD: dcd_int_handler - rx!
V (12224) TUSB:DCD: TUSB IRQ - RX type : Out packet done
V (12224) TUSB:DCD: dcd_int_handler - OUT endpoint!
V (12224) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
V (12224) tusb_cdc_acm: Sent 1 bytes to the buffer
I (12254) example: Got data (1 bytes): e
<device hangs here, nothing else is logged (or happens) even when native USB port is unplugged>
```

## Other items if possible
- [pcaps.zip](https://github.com/espressif/esp-idf/files/5714283/pcaps.zip): packet capture of sending data with (`succeed_hello.pcapng` and without the delay (`fail_hello.pcapng`), filter for `usb.addr == "2.4.2"` for the relevant packets.

JustinOng, 2020-12-18:
I ran into the above issue while diagnosing something else:
[`tinyusb_cdcacm_read`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb.html#_CPPv419tinyusb_cdcacm_read20tinyusb_cdcacm_itf_tP7uint8_t6size_tP6size_t) is essentially performing a [read from a ringbuffer](https://github.com/espressif/esp-idf/blob/master/components/tinyusb/additions/src/tusb_cdc_acm.c#L229). In the event that data written into the ringbuffer has wrapped around to the head of the buffer (eg 32 bytes of data was written to the ringbuffer, 16 bytes at the end, 16 bytes wrapped around to the head), two calls to `tinyusb_cdcacm_read` are actually necessary to read out all the data in the ringbuffer. 

Not sure if this should be handled by the user application, or would it be appropriate to perform two reads in `tinyusb_cdcacm_read` so long as the first call to `xRingbufferReceiveUpTo` received less than `out_buf_sz` bytes?

igrr, 2021-01-08:
@JustinOng You are right, indeed the driver should handle this situation. We have a fix in review for this issue right now.

Alvin1Zhang, 2021-08-05:
Thanks for reporting, and sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/66ac736a8c01c2853d7688535ee79a4f1b8f8785, feel free to reopen.

closed 2021-08-05 by Alvin1Zhang