espressif/esp-idf #6297: esp_restart() crash on rtc_clk_cpu_freq_to_xtal since 6b0a5af73

lorenzomerici, 2020-12-18:
## Environment

- Development Kit:      ESP32-Wroom-32D rev1 on custom board, with ethernet
- IDF version (run ``git describe --tags`` to find it):
    problem present since commit 6b0a5af73, still present on master
- Build System:       make
- Operating System:    Linux
- Using an IDE?: atom

## Problem Description

On esp_restart(), it goes into bootloop. 

### Expected Behavior
Reboot, not to bootloop
### Actual Behavior

bootloop

`GPIO[39] intr, val: 0
W (33166) main-task: RESET NETWORK CONFIG and reboot
I (33766) wifi:flush txq
I (33766) wifi:stop sw txq
I (33766) wifi:lmac stop hw txq
W (33766) main-task: esp_restart called, reboot
Re-enable cpu cache.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4010fa8c: 284aacb9 a502ad01 0acd30db
0x4010fa8c: rtc_clk_cpu_freq_to_xtal at /home/lorenzo/esp/esp-idf/components/esp_hw_support/port/esp32/rtc_clk.c:474

Core  1 register dump:
PC      : 0x4010fa92  PS      : 0x00060330  A0      : 0x8008145c  A1      : 0x3ffcdcc0  
0x4010fa92: rtc_clk_cpu_freq_set_xtal at /home/lorenzo/esp/esp-idf/components/esp_hw_support/port/esp32/rtc_clk.c:545

A2      : 0x00000000  A3      : 0x3ff000c4  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x10624dd3  A7      : 0x00000001  A8      : 0x80084c33  A9      : 0x3ffcdca0  
A10     : 0x3801ff80  A11     : 0x3801ff80  A12     : 0x000249f0  A13     : 0x3801fb80  
A14     : 0x3f400ab8  A15     : 0x3ffcd5e0  SAR     : 0x0000001e  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace:0x4010fa8f:0x3ffcdcc0 0x40081459:0x3ffcdd00 0x400d3b4d:0x3ffcdd20 0x4008bfdd:0x3ffcdd50
0x4010fa8f: rtc_clk_cpu_freq_to_xtal at ??:?

0x40081459: esp_restart at /home/lorenzo/esp/esp-idf/components/esp_system/system_api.c:83

0x400d3b4d: reset_button_task at /home/lorenzo/Desktop/lavoro/Progetti/ElektraConnect/doModem/main/domodem_main.c:71

0x4008bfdd: vPortTaskWrapper at /home/lorenzo/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 7998f07ef4d4c8fe

Rebooting...                           
Re-enable cpu cache.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40116f27  PS      : 0x00060a33  A0      : 0xf5a5fe6a  A1      : 0x3ffcdae0  
0x40116f27: hostap_input at ??:?
`

end so on: 

`ELF file SHA256: 7998f07ef4d4c8fe                                                                                                                             
                                                                                                                                                              
Rebooting...
Re-enable cpu cache.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40116f27  PS      : 0x00060633  A0      : 0xf5a5fe6a  A1      : 0x3ffc4570  
0x40116f27: hostap_input at ??:?

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000009  A5      : 0x00000009  
A6      : 0x20100000  A7      : 0x60000000  A8      : 0x8010faa0  A9      : 0x3ffc4540  
A10     : 0x3f44ccf4  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffc4550  
A14     : 0x00000018  A15     : 0x3ffb0040  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8010faa0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x40116f24:0x3ffc4570 0x75a5fe67:0x3ffc4590 |<-CORRUPTED
0x40116f24: hostap_input at ??:?

ELF file SHA256: 7998f07ef4d4c8fe

Rebooting...
Re-enable cpu cache.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40116f27  PS      : 0x00060f33  A0      : 0xf5a5fe6a  A1      : 0x3ffc4390  
0x40116f27: hostap_input at ??:?

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000009  A5      : 0x00000009  
A6      : 0x20100000  A7      : 0x60000000  A8      : 0x8010faa0  A9      : 0x3ffc4360  
A10     : 0x3f44ccf4  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffc4370  
A14     : 0x00000018  A15     : 0x3ffb0040  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8010faa0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x40116f24:0x3ffc4390 0x75a5fe67:0x3ffc43b0 |<-CORRUPTED
0x40116f24: hostap_input at ??:?
`
### Steps to reproduce

I don't have another shield with ethernet (I don't know why but I think ethrnet is kind of related with this problem, maybe for the clock on gpio0) to try if it happents also there. 

I'm pretty sure that the problem is present since this commit: 6b0a5af73 

I don't want to keep the idf not updated after the latest working commit, so I would like to fix this. How can I help you? I can provide all the informations you need. 

I want to point out that on older esp-idf commits, I have no problem and it reboot. Do you think this may also be related with esp chip revision?

Thank you

Alvin1Zhang, 2020-12-21:
Thanks for reporting and sorry for inconvenience, we will look into.

renzbagaporo, 2021-01-11:
Hi @lorenzomerici, 

Sorry for the late reply. Can you check if adding `COMPONENT_ADD_LDFRAGMENTS := linker.lf` to `$IDF_PATH/components/esp_hw_support/component.mk` fixes the issue?

lorenzomerici, 2021-01-11:
Yes you were right! This fix the issue. I should keep it like this or you will update the idf soon?

AxelLin, 2021-01-11:
May I know what kind of condition will triger this issue?
Or which configuration causes this issue?
Just curious why my device using master tree does not hit this crash.

lorenzomerici, 2021-01-11:
Maybe it's related with the hardware rev? I'm current on REV1. Let's wait if @renzbagaporo can give us more info

renzbagaporo, 2021-01-13:
This was a regression related to code placement. The functions in question were placed in flash instead of RAM.

closed 2021-03-12 by espressif-bot