espressif/esp-idf #6298: [v4.2] Ble/WiFi coexistence issue, order of starting

redradist, 2020-12-19:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

In last release v4.2 Ble/WiFi coexistence is working, but it depends on order of starting ...

If I first start Ble and then WiFi then only WiFi works, but if first I start WiFi and then Ble - both Ble and WiFI works, it is strange issue

### Expected Behavior
The order of starting Ble or WiFi should not depend on each other

### Actual Behavior
To have proper coexistence pf Ble/WiFi you need to start WiFi first and then Ble

### Steps to reproduce

1. Run Ble based on NimBle framework
2. Run WiFi
3. Run MQTT Client and connect to some broker to check real connection over WiFi
4. MQTT and WiFi connection works, but Ble is not working properly

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
[attachments.zip](https://github.com/espressif/esp-idf/files/5719261/attachments.zip)

Alvin1Zhang, 2020-12-21:
Thanks for reporting and sorry for inconvenience, we will look into.

TimXia, 2020-12-22:
@redradist What do you mean by "run Ble"? What does Ble do? And what do you mean by "Run WiFi"?

redradist, 2020-12-22:
> 
> 
> @redradist What do you mean by "run Ble"? What does Ble do? And what do you mean by "Run WiFi"?

@TimXia Hi

Run Ble - start as peripheral device with some attributes
Run WiFi - connect to some Access Point as station

Ble is just peripheral device with some attributes started as GATT Server (attribute provider) and it sends update for particular attributes and waits external commands

redradist, 2021-01-01:
@Alvin1Zhang @TimXia 
I would help with this issue, but I did not find the `wifi_start()` function definition ...

davidpurser, 2021-01-26:
@redradist What advertising interval have you selected for the BLE? I am investigating a similar issue where I have seen Wi-Fi connection failures and TCP timeouts with too frequent advertising interval (15ms). The issue seemed to disappear after changing to 100ms advertising interval.

daniel-grabner, 2021-01-28:
Just upgraded from ESP-IDF v4.1 to v4.2 and observing the same behaviour (ESP32-WROOM-32D, custom HW, Bluedroid):

1) Starting BLE; advertising interval is 250-500ms
2) Starting WIFI (for me this sequence is a must, because WIFI is configured by BLE characteristics via smartphone app)
3) After WIFI has been started, it looks like BLE advertising is not stable - sometimes the device can be found from the app, sometimes not. Also some BLE packets might be discarded, which means that e.g. ESP_GATTS_DISCONNECT_EVT events are not going through.

In the same application I do not have any problems with BLE/WIFI coexistence with ESP-IDF v4.1.

xiongweichao, 2021-02-03:
Hi @redradist 

Could you please provide more information? An executable demo could be very helpful.

Thanks

xiongweichao, 2021-02-03:
Hi @daniel-grabner 

What's the probability that the APP won't find the device?  Will the ESP_GATTS_DISCONNECT_EVT event never going through?

Thanks

xiongweichao, 2021-02-03:
Hi @daniel-grabner and @redradist 

Sorry, I can't  reproduce the issue. So could you please provide a demo.

Thanks

redradist, 2021-02-06:
> Hi @daniel-grabner and @redradist
> 
> Sorry, I can't reproduce the issue. So could you please provide a demo.
> 
> Thanks

@xiongweichao I assume you tested with BleDroid Framework, this issue I have seen on NimBle Framework

xiongweichao, 2021-02-07:
@redradist I've tested it with both the Bledroid Framework and the Nimble Framework, but neither has this issue.

xiongweichao, 2021-03-19:
@redradist Any update?

xiongweichao, 2021-04-06:
Hi @daniel-grabner

Sorry, I can't reproduce the issue. So could you please provide **a demo** or please provide me with the **commit id** of the idf you are currently using. 

Thanks.

daniel-grabner, 2021-04-07:
> Hi @daniel-grabner
> 
> Sorry, I can't reproduce the issue. So could you please provide **a demo** or please provide me with the **commit id** of the idf you are currently using.
> 
> Thanks.

@xiongweichao I am using c40f2590bf759ff60ef122afa79b4ec04e7633d2 (tag v4.2).

Unfortunately I cannot provide you with my full application and can't find the time this week to extract a running demo. Instead I provide you with my sdkconfig file, I hope that helps in order to reproduce the issue: [sdkconfig.txt](https://github.com/espressif/esp-idf/files/6269361/sdkconfig.txt)

If you still can't reproduce please let me know, then I will provide you with a demo. Should be possible by next week.

xiongweichao, 2021-08-05:
Hi, @daniel-grabner 

Sorry for the late reply.  I still can't reproduce this issue and I hope you can provide a demo.

Thanks

jack0c, 2023-07-20:
Since EOL of  v4.2 and no update for this issue for long, will close this issue. Feel free to reopen it if necessary, @daniel-grabner

closed 2023-07-20 by jack0c