espressif/esp-idf #6300: CSI - number of carrier

GeneralFox, 2020-12-19:
I'm working on a project that use CSI data collection with ESP32.

Here below a trace collected with csv header and a sample data:
type,role,mac,rssi,rate,sig_mode,mcs,bandwidth,smoothing,not_sounding,aggregation,stbc,fec_coding,sgi,noise_floor,ampdu_cnt,channel,secondary_channel,local_timestamp,ant,sig_len,rx_state,real_time_set,real_timestamp,len,CSI_DATA,timestamp
CSI_DATA,AP,24:0A:C4:60:0F:08,-52,11,1,4,1,1,1,1,0,0,0,-96,1,8,1,47262566,0,84,0,0,47.464719,384,[84 -64 4 0 0 0 0 0 0 0 0 0 4 -7 4 -7 3 -8 3 -7 4 -7 4 -7 4 -8 4 -8 4 -8 4 -8 4 -9 4 -8 4 -9 4 -9 5 -9 4 -9 5 -10 4 -9 5 -9 5 -10 5 -9 5 -9 6 -9 6 -10 6 -10 6 -10 0 0 6 -10 6 -10 6 -10 5 -10 6 -10 6 -10 6 -10 6 -10 6 -9 6 -10 6 -10 7 -10 6 -10 6 -10 6 -9 6 -10 6 -9 6 -10 6 -9 5 -9 6 -9 6 -8 5 -9 5 -9 5 -9 5 -8 0 0 0 0 0 0 0 0 0 0 ],1607774756.965976
CSI_DATA,AP,24:0A:C4:60:0F:08,-51,11,1,3,1,1,1,1,0,0,0,-96,2,8,1,47270894,0,84,0,0,47.473106,384,[84 -64 4 0 0 0 0 0 0 0 0 0 7 11 7 11 7 10 7 11 8 10 7 11 7 11 8 11 8 11 8 11 9 12 9 11 9 12 9 12 9 12 10 12 10 13 9 12 9 13 9 13 10 13 9 13 9 14 10 14 10 14 10 14 0 0 9 14 10 14 10 15 10 13 9 14 10 15 9 15 9 14 9 14 9 14 10 14 9 14 9 14 9 14 9 14 10 14 10 13 10 13 9 13 9 12 9 13 8 12 8 12 9 12 8 12 8 11 0 0 0 0 0 0 0 0 0 0 ],1607774756.970717
CSI_DATA,AP,24:0A:C4:60:0F:08,-51,11,1,3,1,1,1,1,0,0,0,-96,2,8,1,47270894,0,84,0,0,47.475871,384,[84 -64 4 0 0 0 0 0 0 0 0 0 7 11 7 11 7 10 7 11 8 10 7 11 7 11 8 11 8 11 8 11 9 12 9 11 9 12 9 12 9 12 10 12 10 13 9 12 9 13 9 13 10 13 9 13 9 14 10 14 10 14 10 14 0 0 9 14 10 14 10 15 10 13 9 14 10 15 9 15 9 14 9 14 9 14 10 14 9 14 9 14 9 14 9 14 10 14 10 13 10 13 9 13 9 12 9 13 8 12 8 12 9 12 8 12 8 11 0 0 0 0 0 0 0 0 0 0 ],1607774756.973452

I'm in the HT-LTF case with no STBC and so carrier index should be 0~63, -64~-1 according to the docs.
What I'm observing is that not all the 64 carrier have information so if we think at CSI as a vector of 128 elements (64 real value and 64 imaginary), I have to map this for extraction:

csi_start1=12
csi_end1=64
csi_start2=66
csi_end2=118
csi_size=104

# Create list of imaginary and real numbers from CSI
        for i in range(csi_start1,csi_end1):
            if i % 2 == 0:
                    imaginary.append(csi_raw[i])
            else:
                    real.append(csi_raw[i])
        for i in range(csi_start2,csi_end2):
            if i % 2 == 0:
                    imaginary.append(csi_raw[i])
            else:
                    real.append(csi_raw[i])
        for i in range(int(csi_size / 2)):
                amplitudes.append(sqrt(imaginary[i] ** 2 + real[i] ** 2))
                phases.append(atan2(imaginary[i], real[i]))

you can easily spot the pattern looking at the data between [ ] in the output above.
I'm just wondering if this is an expected behaviour maybe due to fft and the need of 2^n sample or there is something that I'm missing ?
In case I'm right maybe it's better to report this in the doc.

YouDONG-ESP, 2020-12-21:
@GeneralFox That's the exact reason, thanks for reporting, going to update it in the doc.

TimXia, 2021-03-05:
@GeneralFox  It is an expected behaviour according to the IEEE80211 standard. Please refer to it.

NiyamPoudel, 2021-07-23:
Hello @TimXia and @GeneralFox . I am quite new to this whole thing. Please help me understand the CSI_DATA, [112 -128 6 0 0 0 0 0 0 0 0 0 3 9 2 10 1 12 0 14 1 14 1 16 2 16 3 16 3 16 4 15 5 15 5 14 5 13 4 12 2 11 1 11 0 11 -1 11 -2 12 -3 13 -3 14 -3 15 -2 16 -1 17 0 17 1 17 0 0 2 15 2 14 2 14 1 12 0 12 -1 12 -2 12 -4 12 -4 13 -4 15 -4 16 -4 17 -3 18 -3 19 -1 19 0 19 1 18 2 17 1 16 2 15 0 14 -1 14 -2 13 -3 14 -4 15 -6 16 0 0 0 0 0 0 0 0 0 0 ]

What I understand so far is that this is a 1 X 128 Matrix because we have 1 antenna in the receiving end and 2 antennas in the transmitting end. We have 64 sub carriers. I understand the even ones are imaginary and odd ones are real. 
[image]

Next, How do I interpret this chart to make sense of the CSI Data. Thanks

MaxwellAlan, 2022-03-29:
Hi @NiyamPoudel 
For more details, you can refer to this link: http://dhalperi.github.io/linux-80211n-csitool

mhdong, 2022-06-09:
hi @NiyamPoudel  @GeneralFox  Please refer to the latest documentation https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information.

closed 2022-06-09 by Alvin1Zhang