espressif/esp-idf #6301: error building using the 4.3

elo1957, 2020-12-19:
Re cloned the git esp-idf again and then when building
src/bootloader_random_esp32.c:16:
C:/Users/osc/Desktop/esp-idf/components/soc/include/soc/rtc_io_periph.h:111:44: error: 'GPIO_PIN_COUNT' undeclared here (not in a function); did you mean 'SOC_GPIO_PIN_COUNT'?
 extern const rtc_gpio_desc_t rtc_gpio_desc[GPIO_PIN_COUNT];
                                            ^~~~~~~~~~~~~~
                                            SOC_GPIO_PIN_COUNT 

this is due to a missing  file

Alvin1Zhang, 2020-12-21:
@elo1957 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

elo1957, 2020-12-21:
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig
Loading defaults file I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults...
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:4 CONFIG_OPTIMIZATION_LEVEL_DEBUG was replaced with CONFIG_COMPILER_OPTIMIZATION_DEFAULT
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:5 CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED was replaced with CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:6 CONFIG_STACK_CHECK_ALL was replaced with CONFIG_COMPILER_STACK_CHECK_MODE_ALL
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:7 CONFIG_STACK_CHECK was replaced with CONFIG_COMPILER_STACK_CHECK
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:8 CONFIG_WARN_WRITE_STRINGS was replaced with CONFIG_COMPILER_WARN_WRITE_STRINGS
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:14 CONFIG_SPIRAM_SUPPORT was replaced with CONFIG_ESP32_SPIRAM_SUPPORT
I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/sdkconfig.defaults:16 CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST was replaced with CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP  
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "esp32_cam_http_jpg" version: 96a742a
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script I:/ESP/projects/esp32-cam-ai-thinker/examples/http_jpg/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/osc/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning 
wpa_supplicant xtensa
-- Component paths: C:/Users/osc/Desktop/esp-idf/components/app_trace C:/Users/osc/Desktop/esp-idf/components/app_update C:/Users/osc/Desktop/esp-idf/components/asio C:/Users/osc/Desktop/esp-idf/components/bootloader C:/Users/osc/Desktop/esp-idf/components/bootloader_support C:/Users/osc/Desktop/esp-idf/components/bt C:/Users/osc/Desktop/esp-idf/components/cbor C:/Users/osc/Desktop/esp-idf/components/cmock C:/Users/osc/Desktop/esp-idf/components/coap C:/Users/osc/Desktop/esp-idf/components/console C:/Users/osc/Desktop/esp-idf/components/cxx C:/Users/osc/Desktop/esp-idf/components/driver C:/Users/osc/Desktop/esp-idf/components/efuse C:/Users/osc/Desktop/esp-idf/components/esp-tls C:/Users/osc/Desktop/esp-idf/components/esp32 C:/Users/osc/Desktop/esp-idf/components/esp_adc_cal C:/Users/osc/Desktop/esp-idf/components/esp_common C:/Users/osc/Desktop/esp-idf/components/esp_eth C:/Users/osc/Desktop/esp-idf/components/esp_event C:/Users/osc/Desktop/esp-idf/components/esp_gdbstub C:/Users/osc/Desktop/esp-idf/components/esp_hid C:/Users/osc/Desktop/esp-idf/components/esp_http_client C:/Users/osc/Desktop/esp-idf/components/esp_http_server C:/Users/osc/Desktop/esp-idf/components/esp_https_ota C:/Users/osc/Desktop/esp-idf/components/esp_https_server C:/Users/osc/Desktop/esp-idf/components/esp_hw_support C:/Users/osc/Desktop/esp-idf/components/esp_ipc C:/Users/osc/Desktop/esp-idf/components/esp_local_ctrl C:/Users/osc/Desktop/esp-idf/components/esp_netif C:/Users/osc/Desktop/esp-idf/components/esp_pm C:/Users/osc/Desktop/esp-idf/components/esp_ringbuf C:/Users/osc/Desktop/esp-idf/components/esp_rom C:/Users/osc/Desktop/esp-idf/components/esp_serial_slave_link C:/Users/osc/Desktop/esp-idf/components/esp_system C:/Users/osc/Desktop/esp-idf/components/esp_timer C:/Users/osc/Desktop/esp-idf/components/esp_websocket_client C:/Users/osc/Desktop/esp-idf/components/esp_wifi C:/Users/osc/Desktop/esp-idf/components/espcoredump C:/Users/osc/Desktop/esp-idf/components/esptool_py C:/Users/osc/Desktop/esp-idf/components/expat C:/Users/osc/Desktop/esp-idf/components/fatfs C:/Users/osc/Desktop/esp-idf/components/freemodbus C:/Users/osc/Desktop/esp-idf/components/freertos C:/Users/osc/Desktop/esp-idf/components/hal C:/Users/osc/Desktop/esp-idf/components/heap C:/Users/osc/Desktop/esp-idf/components/idf_test C:/Users/osc/Desktop/esp-idf/components/jsmn C:/Users/osc/Desktop/esp-idf/components/json C:/Users/osc/Desktop/esp-idf/components/libsodium C [... cut ...]
C:/Users/osc/Desktop/esp-idf/components/soc/include/hal/gpio_types.h:18:10: fatal error: soc/gpio_caps.h: No such file or directory
 #include "soc/gpio_caps.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
[177/980] Building C object esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/src/smartconfig_ack.c.obj
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

elo1957, 2020-12-22:
The components directory structure for the latest git repository has changes and the files in soc/include has moved to C:\Users\osc\Desktop\esp-idf\components\soc\esp32c3\include\soc  - therefore the compiler can not find them - how to update the cmakelist.txt so the new includes can be found - or make an update on git so it fits!  

ind the ver4.1
C:\Users\osc\Downloads\esp-idf-v4.1\esp-idf-v4.1\components\soc\esp32\include\soc

mahavirj, 2020-12-23:
@elo1957 For first issue, fix is available at https://github.com/espressif/esp-idf/pull/6029. This has been merged internally as well, it will soon reflect on github as well.