espressif/esp-idf #6302: ESP32 ble scan whitelist, not work

JerryZheng89, 2020-12-20:
## Environment

- Development Kit:      [NodeMCU-32 v1.2]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version  :v4.2-47-g2532ddd9f
- Build System:         [idf.py]
- Compiler version : (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ ESP Command Prompt ].
- Using an IDE?: [Yes(VS code)]
- Power Supply:         [USB]

## Problem Description
when I change scan policy to `BLE_SCAN_FILTER_ALLOW_ONLY_WLST`, flow 3steps change in gatt_client example:
1. update whitelist with my ble periperal address;
2. set ble scan parameters
3. start scan 30 seconds
or
1. set ble scan paramters
2. update whitelist with my ble periperal address;
3. start scan 30 seconds

### Expected Behavior
**event list**
1. ESP_GAP_BLE_SCAN_START_COMPLETE_EVT,  start scan success
2. ESP_GAP_BLE_SCAN_RESULT_EVT->ESP_GAP_SEARCH_INQ_RES_EVT,  searched the device, which is in whitelist
3. ESP_GAP_BLE_SCAN_RESULT_EVT->ESP_GAP_SEARCH_INQ_CMPL_EVT,  search finished.
### Actual Behavior
**event list**
1. ESP_GAP_BLE_SCAN_START_COMPLETE_EVT,  start scan success
2. ESP_GAP_BLE_SCAN_RESULT_EVT->ESP_GAP_SEARCH_INQ_CMPL_EVT,  search finished.

 
### Code to reproduce this issue

```cpp
uint8_t whitelist_addr[6] = {0xc9, 0xa7, 0xab, 0x5e, 0x2d, 0xbe};

static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ONLY_WLST,
    .scan_interval          = 0x50,
    .scan_window            = 0x30,
    .scan_duplicate         = BLE_SCAN_DUPLICATE_DISABLE
};
```
- in esp_gap_cb callback function
```cpp
    case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT: {
        //the unit of the duration is second
        uint32_t duration = 30;
        esp_ble_gap_start_scanning(duration);
        break;
    }
    case ESP_GAP_BLE_UPDATE_WHITELIST_COMPLETE_EVT:{
        esp_err_t scan_ret = esp_ble_gap_set_scan_params(&ble_scan_params);
        if (scan_ret){
            ESP_LOGE(GATTC_TAG, "set scan params error, error code = %x", scan_ret);
        }
        break;
    }
```
- in gattc_profile_event handler
```cpp
    case ESP_GATTC_REG_EVT:
        ESP_LOGI(GATTC_TAG, "REG_EVT");
        esp_ble_gap_update_whitelist(true, whitelist_addr, BLE_WL_ADDR_TYPE_PUBLIC);
        break;
```

## Debug Logs

```
I (418) boot: Disabling RNG early entropy source...
I (419) cpu_start: Pro cpu up.
I (422) cpu_start: Application information:
I (427) cpu_start: Project name:     gatt_client_demo
I (433) cpu_start: App version:      1
I (437) cpu_start: Compile time:     Dec 19 2020 23:48:58
I (443) cpu_start: ELF file SHA256:  b16071c68d527579...
I (449) cpu_start: ESP-IDF:          v4.2-47-g2532ddd9f
I (455) cpu_start: Starting app cpu, entry point is 0x40081854
......
I (0) cpu_start: Starting scheduler on APP CPU.
I (620) BTDM_INIT: BT controller compile version [3723d5b]
I (620) system_api: Base MAC address is not set
I (620) system_api: read default base MAC address from EFUSE
I (720) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (1090) GATTC_DEMO: REG_EVT
I (1100) GATTC_DEMO: scan start success
```

##  sniffer
Although, the scan type, which I set is BLE_SCAN_TYPE_ACTIVE, I don't have seen any scan or respone sacn package in Wireshark within the 30 seconds.

Alvin1Zhang, 2020-12-21:
Thanks for reporting, we will look into.

JerryZheng89, 2020-12-23:
Is threre any progress about this  problems,  or any  suggestion to let me move forward?

xiewenxiang, 2020-12-24:
The `uint8_t whitelist_addr[6] = {0xc9, 0xa7, 0xab, 0x5e, 0x2d, 0xbe};` is Static  Random Device Address.

Maybe you need to modify it this way `esp_ble_gap_update_whitelist(true, whitelist_addr, BLE_WL_ADDR_TYPE_PUBLIC);` => `esp_ble_gap_update_whitelist(true, whitelist_addr, BLE_WL_ADDR_TYPE_RANDOM);`

JerryZheng89, 2020-12-24:
> The `uint8_t whitelist_addr[6] = {0xc9, 0xa7, 0xab, 0x5e, 0x2d, 0xbe};` is Static  Random Device Address.
>
> Maybe you need to modify it this way `esp_ble_gap_update_whitelist(true, whitelist_addr, [...]

Is threre any  method to add public address into white list? 
I will try your suggestion tomorrow. 
@xiewenxiang thanks

JerryZheng89, 2020-12-25:
The issue is solved, @xiewenxiang @Alvin1Zhang 
follow these steps:
1. update whitelist with my ble periperal address;
2. set ble scan parameters
3. start scan 30 seconds
change, whitelist update function addr type parameters:
```c
esp_ble_gap_update_whitelist(true, whitelist_addr, BLE_WL_ADDR_TYPE_RANDOM);
```

And it works well!!!

but I really don't konw why the wl_addr_type is random,   i really think it is the public address, vevry confusing me!

closed 2020-12-25 by JerryZheng89