espressif/esp-idf #6303: When using WroverKit PSRAM and Bluetooth, "memory allocation failed" occurs.

ansee7221, 2020-12-21:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-1521-g0289d1c-dirty
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
When using WROVER PSRAM and Bluetooth, "memory allocation fail" occurs.
If "bluetooth" is not used through menuconfig, it works normally.

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void heap_dbg(void)
{
	printf("free heap size:%d\n", esp_get_free_heap_size());
	printf("minimum free heap size:%d\n", esp_get_minimum_free_heap_size());
	
	#ifdef EXTERNAL_PSRAM_USE
	printf("free internal memory: %u lifetime minimum: %u\n",
		heap_caps_get_free_size(MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT),
		heap_caps_get_minimum_free_size(MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT));
	printf("integrity check: %d\n", heap_caps_check_integrity(MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT, true));	
	#endif
}

void server_main (void *arg) 
{    
	int init = 0;	
	long ret = 0;
	int size = 0;
	size_t device= 0;

	tcpip_adapter_ip_info_t ip4_info = {0};
	struct ip6_addr if_ipaddr_ip6 = {0};

	printf("iotivity server task started\r\n");
	
	// wait to fetch IPv4 && ipv6 address
	xEventGroupWaitBits(wifi_event_group, IPV4_CONNECTED_BIT | IPV6_CONNECTED_BIT, false, true, portMAX_DELAY);
	xEventGroupWaitBits(wifi_event_group, IPV6_CONNECTED_BIT, false, true, portMAX_DELAY);

	if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &ip4_info) != ESP_OK) {
		printf("get IPv4 address failed\r\n");
	} else {
		printf("got IPv4 addr:%s\r\n", ip4addr_ntoa(&(ip4_info.ip)));
	}

[... 178 lines cut ...]
I (2057) wifi_init: tcp mss: 1440
I (2067) wifi_init: WiFi/LWIP prefer SPIRAM
I (2177) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (2187) wifi:mode : sta (ac:67:b2:71:61:7c)
I (2187) wifi:enable tsf
**I (2187) BTDM_INIT: BT controller compile version [b73c48e]

Memory allocation failed**

Backtrace:0x4009399e:0x3ffe5b90 0x400942c5:0x3ffe5bb0 0x400d46cb:0x3ffe5bd0 0x40083906:0x3ffe5bf0 0x40083921:0x3ffe5c20 0x40099cb3:0x3ffe5c40 0x40099cd4:0x3ffe5c60 0x401265f7:0x3ffe5c80 0x400feb48:0x3ffe5ce0 0x400ff0e4:0x3ffe5d00 0x40125a69:0x3ffe5d30 0x40127baf:0x3ffe5d50

ELF file SHA256: 002574f1670f654b

Rebooting...

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

chegewara, 2020-12-21:
Please check in menuconfig option that enables `xTaskCreateStatic` to use PSRAM, its been introduced recently.

ansee7221, 2020-12-21:
@chegewara, If I set the option to "heap_caps_malloc" in menuconfig, there is no "Allow external memory as an argument to xTaskCreateStatic" item.  
Also, as a result of testing with external PSRAM disabled in menuconfig, a "Memory allocation failed" error occurs. The error does not seem to be related to the use of external PSRAM.

chegewara, 2020-12-28:
This is new option in v4.3 you have to enable:
```
Name: SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
Prompt: Allow external memory as an argument to xTaskCreateStatic
Type: bool
Value: n

Help:

  Because some bits of the ESP32 code environment cannot be recompiled with the cache workaround,
  normally tasks cannot be safely run with their stack residing in external memory; for this reason
  xTaskCreate and friends always allocate stack in internal memory and xTaskCreateStatic will check if
  the memory passed to it is in internal memory. If you have a task that needs a large amount of stack
  and does not call on ROM code in any way (no direct calls, but also no Bluetooth/WiFi), you can try to
  disable this and use xTaskCreateStatic to create the tasks stack in external memory.
```

boblane1, 2020-12-29:
Hi @ansee7221,

I guess your code is developed based on the example of blufi, and the demo can work without any problem. According to the backtrace and .elf file which you gave, it seems that the memory allocation failed when retrieving the configuration in the nvs flash. I suggest you to erase the whole flash first and then download your application code to see if the problem still exists.

Thanks!

ansee7221, 2020-12-31:
> This is new option in v4.3 you have to enable:
>
> ```
> Name: SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
> Prompt: Allow external memory as an argument to xTaskCreateStatic
> Type: bool
> Value: n
>
> Help:
>
>   Because [...]

ansee7221, 2020-12-31:
> This is new option in v4.3 you have to enable:
> 
> ```
> Name: SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
> Prompt: Allow external memory as an argument to xTaskCreateStatic
> Type: bool
> Value: n
> 

The following modifications have been made, but the same symptoms will occur.

#
# SPIRAM cache workaround debugging
#
CONFIG_SPIRAM_CACHE_WORKAROUND_STRATEGY_MEMW=y
# CONFIG_SPIRAM_CACHE_WORKAROUND_STRATEGY_DUPLDST is not set
# CONFIG_SPIRAM_CACHE_WORKAROUND_STRATEGY_NOPS is not set
# end of SPIRAM cache workaround debugging

CONFIG_SPIRAM_BANKSWITCH_ENABLE=y
CONFIG_SPIRAM_BANKSWITCH_RESERVE=8
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y

I (1926) wifi_init: rx ba win: 6
I (1926) wifi_init: tcpip mbox: 32
I (1936) wifi_init: udp mbox: 6
I (1936) wifi_init: tcp mbox: 6
I (1936) wifi_init: tcp tx win: 5744
I (1946) wifi_init: tcp rx win: 5744
I (1946) wifi_init: tcp mss: 1440
I (2056) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (2066) wifi:mode : sta (ac:67:b2:71:61:7c)
I (2066) wifi:enable tsf
I (2066) BTDM_INIT: BT controller compile version [b73c48e]

Memory allocation failed

Backtrace:0x400938d2:0x3fff74a0 0x40094211:0x3fff74c0 0x400d4753:0x3fff74e0 0x4008395e:0x3fff7500 0x40083987:0x3fff7530 0x40099c8f:0x3fff7550 0x40099cb0:0x3fff7570 0x4012693f:0x3fff7590 0x4011846a:0x3fff75b0 0x40113155:0x3fff7600 0x401133db:0x3fff7620 0x40127577:0x3fff7640

ansee7221, 2021-01-05:
> Hi @ansee7221,
>
> I guess your code is developed based on the example of blufi, and the demo can work without any problem. According to the backtrace and .elf file which you gave, it seems that the [...]

Hi @boblane1, 
I erased the whole flash and then downloaded application code, but "Memory allocation failed" error still occur.

boblane1, 2021-01-06:
Hi @ansee7221, 

The attached file is my configuration about PSRAM for blufi demo, you can refer it and have a try. Please pay attention to `CONFIG_SPIRAM_USE_MALLOC`.

Thanks!
[Uploading sdkconfig.txt…]()

ansee7221, 2021-01-06:
> Hi @ansee7221,
> 
> The attached file is my configuration about PSRAM for blufi demo, you can refer it and have a try. Please pay attention to `CONFIG_SPIRAM_USE_MALLOC`.
> 
> Thanks!
> [sdkconfig.txt](https://github.com/espressif/esp-idf/files/5773943/sdkconfig.txt)

Thank you!
I solved problem by setting it as below.
component config > Application Level Tracing > Data Destination (None)  : Yes -> None

Alvin1Zhang, 2021-01-06:
Thanks for reporting, glad the issue resolved, feel free to reopen. Thanks.

closed 2021-01-06 by Alvin1Zhang