espressif/esp-idf #6304: Passing parameters to configure WiFi

Oleg-Perevyshin, 2020-12-21:
I have a structure with settings.
I am initializing a structure from a file. There is data before WiFi setup.
```
/* Структура настроек */
struct Config {
  char apSSID[33];  // "TestAP"
  char apCh[3];     // "5"
  char apIP[16];    // "192.168.108.1"
  //...
};
extern struct Config configDev;
```
I am trying to use parameters from a **configDev** to configure WiFi.
```
/* Настройка параметров WiFi */
wifi_config_t wifiAPConfig = {
    .ap = {
      .ssid = configDev.apSSID,
      .ssid_len = strlen(configDev.apSSID)
      .channel = configDev.apCh,
      //...
    },
  };
//...
netif_ap = esp_netif_create_default_wifi_ap();
esp_netif_ip_info_t ipInfo;
IP4_ADDR(&ipInfo.ip, configDev.apIP);
//...
```
But I can't do this, the compiler throws an error.
How to set the string settings and IP address correctly?

david-cermak, 2020-12-21:
@Oleg-Perevyshin 

``` c
    esp_netif_ip_info_t ipInfo;
    ip_info.ip.addr = ipaddr_addr(configDev.apIP);
```

Oleg-Perevyshin, 2020-12-21:
Thanks!
IPs It works!
How do you make string parameters (SSID, password...)?

david-cermak, 2020-12-21:
> How do you make string parameters (SSID, password...)?

Both ssid and password in the AP config structure are character arrays:
https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi_types.h#L217
so you might need to copy the data with `str(n)cpy`/`memcpy()` (please check idf examples for more details)

Alvin1Zhang, 2020-12-22:
Thanks for reporting and for "How" questions, you could create a post on our forum at https://esp32.com. Thanks.

closed 2020-12-22 by Alvin1Zhang

Oleg-Perevyshin, 2020-12-22:
I just can't set the channel number.
The type is the same as Password or SSID.
I tried different options.
```
uint8_t ch = atoi(configDev.apCh);
strcpy(wifiAPConfig.ap.channel, ch);

memcpy( wifiAPConfig.ap.channel, atoi(configDev.apCh), sizeof( atoi(configDev.apCh)) );
memcpy( wifiAPConfig.ap.channel, configDev.apCh, sizeof(configDev.apCh) );
```
By the way, for the password and SSID, only the option worked:
```memcpy(wifiAPConfig.ap.password, configDev.apPSK, sizeof(configDev.apPSK));```

`str(n)cpy` why doesn't it work ...