espressif/esp-idf #6305: Failed to resolve component (CMake Build System)

benjs, 2020-12-21:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-2136-gb0150615d
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I try to setup a new project with one custom component and get the following error when trying to build:
`Failed to resolve component 'example_component'.`

### Steps to reproduce

1. `. $IDF_PATH/export.sh`
2. `idf.py create-project example_project --path example_project`
3. `idf.py create-component example_component`
4. Add to `main/CMakeLists.txt` inside `idf_component_register:` `REQUIRES example_component`
5. `idf.py build`

## Debug Logs

<details>
<summary>Compiler log</summary>
```
[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/home/username/Documents/projects/esp/example_project/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
/usr/bin/cmake -S/home/username/Documents/projects/esp/example_project -B/home/username/Documents/projects/esp/example_project/build
Executing action: all (aliases: build)
Running ninja in directory /home/username/Documents/projects/esp/example_project/build
Executing "ninja all"...

```
</details>

boarchuz, 2020-12-21:
`idf.py -C example_project/components create-component example_component`

benjs, 2020-12-21:
Thanks, that worked! 

However, both file trees are equal. What is different with extra parameter `-C components`?

boarchuz, 2020-12-21:
The build system will look in IDF_PATH/components and PROJECT_DIR/components by default.

You created `/example_component` (at the same depth as `/example_project`); you needed `/example_project/components/example_component`

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#example-project

closed 2020-12-21 by benjs

egameezakira14, 2024-10-19:
CMake Error at C:/Users/User/.platformio/packages/framework-espidf/tools/cmake/component.cmake:440 (message):
  Called idf_component_register from a non-component directory.
Call Stack (most recent call first):
  CMakeLists.txt:5 (idf_component_register)

AlexanderSkvortsov, 2025-03-13:
Somthing like this : REQUIRES "example_component" maybe?