espressif/esp-idf #6306: Full path of source files into application binaries

alezancomelit, 2020-12-21:
## Environment

- Development Kit:      Homemade board
- Module or chip used:  ESP32-WROOM-32E
- IDF version: 			v4.1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows 10
- Environment type: ESP Command Prompt.
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

The application binary include some strings of full path of source files in esp-idf repository or compiled project.

I understand that these paths are used by error, assert and other functions that include `__FILE__` macro.
But, is it possible to remove full path and use a relative path or only file name?

It's very important for our applications to be portable: same project, at same commit, built on different machines by different users, need to create the same binary!! This permit a long support and debug of our products.

For now the unique solution that I found is:
- define NDEBUG adding "add_compile_definitions(NDEBUG)" in application CMakeLists.txt
- define CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE in menuconfig
- define CONFIG_FREERTOS_ASSERT_DISABLE in menuconfig

But it's a big limit. And introduce a lot of compilation warnings, for components that not manage correctly assert call. An example:
> C:/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_conn.c:376:9: warning: variable 'rc' set but not used [-Wunused-but-set-variable]
>     int rc;
>         ^~

Alvin1Zhang, 2020-12-22:
Thanks for reporting.

projectgus, 2021-01-10:
Hi @alezancomelit,

Thanks for being patient while someone got back to you.

The limitation of having the assert macro pass `__FILE__` is a significant one, as this is part of the standard newlib assert mechanism.

We have a [configuration option for silent assertions](https://docs.espressif.com/projects/esp-idf/en/v4.1/api-reference/kconfig.html#config-compiler-optimization-assertion-level) instead. However unfortunately there are currently some issues with this as well, asserts in LWIP and FreeRTOS still include the file name. This is noted in #5873 as well.

> * define NDEBUG adding "add_compile_definitions(NDEBUG)" in application CMakeLists.txt
> * define CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE in menuconfig
> * define CONFIG_FREERTOS_ASSERT_DISABLE in menuconfig

It should only be necessary to set the second two options. Do you recall a case where explicitly setting NDEBUG in the CMakeLists.txt file was required? If you've found one then this is a bug.

FWIW, we plan to remove the standalone option for FreeRTOS asserts in ESP-IDF V5.0 and have this controlled by the main assert setting (as this is a breaking change, we can't do it in a minor update.)

> And introduce a lot of compilation warnings, for components that not manage correctly assert call. An example:

Thanks for pointing this out, will fix as well.

> But, is it possible to remove full path and use a relative path or only file name?

We'd really like to support this, but unfortunately it's quite hard to do with gcc and/or CMake. There's no good solution we're aware of, although some people have hacked this in for other projects it's not pretty.

> It's very important for our applications to be portable: same project, at same commit, built on different machines by different users, need to create the same binary!! This permit a long support and debug of our products.

Reproducible builds are something we would like to support as well, but as you've found there are a few blockers to this. #5873 mentions an additional one.

alezancomelit, 2021-01-11:
> It should only be necessary to set the second two options. Do you recall a case where explicitly setting NDEBUG in the CMakeLists.txt file was required? If you've found one then this is a bug.

You are right! I retry to compile with and without NDEBUG and it's same.

But I found another problem that I have not seen before.

If define only CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE  and CONFIG_FREERTOS_ASSERT_DISABLE, into binary found this file full-path:
`C:/esp/esp-idf/components/driver/rtc_io.c`

Into rtc_io.c, I found a macro definition with this:
`ESP_LOGE(RTCIO_TAG,"%s:%d (%s):%s", __FILE__, __LINE__, __FUNCTION__, str)`

So, to remove the full path I need to remove log output.

projectgus, 2021-02-12:
Hi @alezancomelit,

Thanks for the heads-up about the logging definition. I thought we removed all these some time ago, but seems like they've been added again.

I'm working on this now, should hopefully have fixes in master soon. I'm unsure how much we'll be able to backport to earlier versions, hopefully enough to remove the most obvious places paths are injected by asserts. Will update once the work is done.

closed 2021-03-12 by espressif-bot