espressif/esp-idf #6308: UDP Server | Answer length no more than 16 characters

Oleg-Perevyshin, 2020-12-22:
I have put together an example of a [UDP server from Example](https://github.com/espressif/esp-idf/tree/0b71a0a46d67cce681ec55973b020d950d8596bd/examples/protocols/sockets/udp_server).

I am trying to send data back.
But the packet length is no more than 16 characters. I tried to increase the buffer, it didn't work.
[On line 100](
), I write my data to the buffer (rx_buffer).
```
/* Положить в rx_buffer полезные данные и вернуть в ответ */
// cJSON * root = cJSON_CreateObject();
// cJSON_AddStringToObject(root, "Key", "Value");
// cJSON_AddStringToObject(root, "Key", "Post a reply with useful information about the device. JSON format!");
// const char * jsonString = cJSON_PrintUnformatted(root);
const char * jsonString = "Post a reply with useful information about the device. JSON format!";
memcpy(rx_buffer, jsonString, strlen(jsonString));
send_uart("UDP", "Send Data", rx_buffer);
//cJSON_Delete(root);
```
I tried to immediately write data there and via cJSON.
How do I send larger data?

david-cermak, 2020-12-22:
@Oleg-Perevyshin 
Do i understand correctly, the line 100 of the udp-server example prevents you from sending more data than 16 bytes to the socket? Or do you actually experiment with uart or cJSON?

https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/examples/protocols/sockets/udp_server/main/udp_server.c#L101

``` c
int err = sendto(sock, rx_buffer, len, 0, (struct sockaddr *)&source_addr, sizeof(source_addr));
```
Have you tried to check the `len` parameter? Didn't see this being modified anywhere in your code pasted.

Oleg-Perevyshin, 2020-12-22:
It's okay, I was wrong!

closed 2020-12-22 by Oleg-Perevyshin