espressif/esp-idf #6309: esp32 stuck in loop of "xEventGroupWaitBits"

MarkEvens, 2020-12-22:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.0.2-206-g5630b17e1
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: [VS code]
- Power Supply:         [USB]

## Problem Description

I am using esp32 in sta+ap mode both mode start running but my code not go forward after "xEventGroupWaitBits"
function called and stuck in there.

### Code to reproduce this issue

```cpp
wifi_config
wifi_config.sta.ssid = "xxxx" 
wifi_config.sta.password = "xxxxxxxxxx";
wifi_config.sta.pmf_cfg.capable = true;
wifi_config.sta.pmf_cfg.required = false;

wifi_config_t wifi_ap_config = {};
wifi_ap_config.ap.ssid = "ESP-32"
wifi_config.ap.ssid_len = 32;
wifi_ap_config.ap.password = "123456789"
wifi_ap_config.ap.max_connection = wifiCred.maxConnection;
wifi_ap_config.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;

if (strlen(wifiCred.apPWD) == 0)
{
    wifi_ap_config.ap.authmode = WIFI_AUTH_OPEN;
}

ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_ap_config));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
ESP_ERROR_CHECK(esp_wifi_start());

ESP_LOGI(TAG, "wifi_init_sta finished.");
start_webserver();
    /* Waiting until either the connection is established (WIFI_CONNECTED_BIT) or connection failed for the maximum
        * number of re-tries (WIFI_FAIL_BIT). The bits are set by event_handler() (see above) */
bits = xEventGroupWaitBits(s_wifi_event_group,
                                        WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
                                        pdFALSE,
                                        pdFALSE,
                                        portMAX_DELAY);

/* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
        * happened. */
    ESP_LOGI(TAG, "\nfinished");
    if (bits & WIFI_CONNECTED_BIT)
    {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                    wifiCred.routerSSID, wifiCred.routerPWD);
}
else if (bits & WIFI_FAIL_BIT)
{
    ESP_LOGI(TAG, "Failed to connect to SSID:%s, password:%s",
                wifiCred.routerSSID, wifiCred.routerPWD);
}
else
{
    ESP_LOGE(TAG, "UNEXPECTED EVENT");
}
```

 "wifi_init_sta finished" this message in printing on but terminal but after that next mesaage of "finished" in not.

MarkEvens, 2020-12-23:
this is my debug log
```
I (1191) wifi:wifi firmware version: 6e1f9e4
I (1191) wifi:config NVS flash: enabled
I (1191) wifi:config nano formating: disabled
I (1191) wifi:Init data frame dynamic rx buffer num: 32
I (1201) wifi:Init management frame dynamic rx buffer num: 32
I (1201) wifi:Init management short buffer num: 32
I (1211) wifi:Init dynamic tx buffer num: 32
I (1211) wifi:Init static rx buffer size: 1600
I (1211) wifi:Init static rx buffer num: 10
I (1221) wifi:Init dynamic rx buffer num: 32
[0;32mI (1321) phy: phy_version: 4390, 6b3c1f2, Sep 10 2020, 15:09:07, 0, 0[0m
I (1331) wifi:mode : sta (4c:11:ae:7b:b9:88) + softAP (4c:11:ae:7b:b9:89)
I (1331) wifi:Total power save buffer number: 16
I (1331) wifi:Init max length of beacon: 752/752
I (1341) wifi:Init max length of beacon: 752/752
[0;32mI (1341) Wifi: wifi_init_sta finished.[0m
I (3791) wifi:ap channel adjust o:1,1 n:7,2
I (3791) wifi:new:<7,2>, old:<1,1>, ap:<7,2>, sta:<7,2>, prof:1
I (4841) wifi:state: init -> auth (b0)
I (4851) wifi:state: auth -> assoc (0)
I (4861) wifi:state: assoc -> run (10)
I (4871) wifi:connected with TP-LINK_1032, aid = 1, channel 7, 40D, bssid = f4:f2:6d:8f:10:32
I (4871) wifi:security: WPA2-PSK, phy: bgn, rssi: -39
I (4881) wifi:pm start, type: 1

I (4931) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (13741) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (15691) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
I (18761) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (19681) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
I (60741) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (61671) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
I (67711) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (69761) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
I (89721) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (90751) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
I (129761) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (130691) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
I (167751) wifi:new:<7,0>, old:<7,2>, ap:<7,2>, sta:<7,0>, prof:1
I (168671) wifi:new:<7,2>, old:<7,0>, ap:<7,2>, sta:<7,2>, prof:1
```
as we can see esp never got ip

MarkEvens, 2020-12-23:
I am tried example to change in c++ code and run it 

```
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char *TAG = "wifi station";

static int s_retry_num = 0;

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
    }
}

void wifi_init_sta()
{
[... 124 lines cut ...]
I (613) wifi:Init management frame dynamic rx buffer num: 32
I (623) wifi:Init management short buffer num: 32
I (623) wifi:Init dynamic tx buffer num: 32
I (633) wifi:Init static rx buffer size: 1600
I (633) wifi:Init static rx buffer num: 10
I (643) wifi:Init dynamic rx buffer num: 32
[0;32mI (733) phy: phy_version: 4390, 6b3c1f2, Sep 10 2020, 15:09:07, 0, 0[0m
I (733) wifi:mode : sta (4c:11:ae:7b:b9:88)
[0;32mI (743) wifi station: wifi_init_sta finished.[0m
I (863) wifi:new:<7,2>, old:<1,0>, ap:<255,255>, sta:<7,2>, prof:1
I (863) wifi:state: init -> auth (b0)
I (873) wifi:state: auth -> assoc (0)
I (873) wifi:state: assoc -> run (10)
I (893) wifi:connected with TP-LINK_1032, aid = 1, channel 7, 40D, bssid = f4:f2:6d:8f:10:32
I (893) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (893) wifi:pm start, type: 1

I (973) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (96513) wifi:new:<7,0>, old:<7,2>, ap:<255,255>, sta:<7,0>, prof:1
I (97533) wifi:new:<7,2>, old:<7,0>, ap:<255,255>, sta:<7,2>, prof:1
I (107473) wifi:new:<7,0>, old:<7,2>, ap:<255,255>, sta:<7,0>, prof:1
I (121503) wifi:new:<7,2>, old:<7,0>, ap:<255,255>, sta:<7,2>, prof:1
I (134503) wifi:new:<7,0>, old:<7,2>, ap:<255,255>, sta:<7,0>, prof:1
I (136453) wifi:new:<7,2>, old:<7,0>, ap:<255,255>, sta:<7,2>, prof:1
```

YouDONG-ESP, 2020-12-24:
Can you fix the channel of your router? Seem like your  router keep switching channel. 
 ```
I (96513) wifi:new:<7,0>, old:<7,2>, ap:<255,255>, sta:<7,0>, prof:1
I (97533) wifi:new:<7,2>, old:<7,0>, ap:<255,255>, sta:<7,2>, prof:1
```
I run your second code and there is no problem for getting IP address.(except i run it in C and it need to remove line  `wifi_config.sta.pmf_cfg = {};`for the compiler), dose this issue happen with others router?

MarkEvens, 2020-12-24:
In 'C' it also work with my setup but in "C++" not

MarkEvens, 2020-12-24:
my TP-LINK router has some issue but worked with other router second code is working.

MarkEvens, 2020-12-24:
need help my esp32 is in AP+STA  mode, and esp32 is not connected with any router, my mobile connect to my esp32 AP now i want to scan wifi network, without losing my mobile connection.

give me something it took my lot's of time.

YouDONG-ESP, 2020-12-24:
Could you show me the complete code for the AP+STA mode?

MarkEvens, 2020-12-25:
[sample_code.zip](https://github.com/espressif/esp-idf/files/5741392/sample_code.zip)
this my code in which i have two rest api
1. scanSSIDs
2. getSSIDs

first i call scanSSIDs which start scanning fill ssid list to get_sstd_list variable
my mobile disconnect from esp32 hotspot
i want to do this without disconnect

MarkEvens, 2020-12-25:
only first time I have above issue after that mobile is not disconnecting 

but don't how esp32 is watch dog resting.
```
[0;31mE (145695) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (145695) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (145695) task_wdt: Tasks currently running:[0m
[0;31mE (145695) task_wdt: CPU 0: tiT[0m
[0;31mE (145695) task_wdt: CPU 1: wifi_app_task[0m
[0;31mE (152695) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (152695) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (152695) task_wdt: Tasks currently running:[0m
[0;31mE (152695) task_wdt: CPU 0: tiT[0m
[0;31mE (152695) task_wdt: CPU 1: wifi_app_task[0m
```

YouDONG-ESP, 2020-12-25:
@MarkEvens  in funciton wifi_app_task(),` vTaskDelay(1/portTICK_PERIOD_MS)`; means delay 1ms which is too short that may lead task watchdog. Set it too `1000/portTICK_PERIOD_MS`(1 second) will be ok.

MarkEvens, 2020-12-30:
@YouDONG-ESP  Thanks i got my answer.

closed 2020-12-30 by MarkEvens